ryanbinns/ttwatch #7: Unable to download activities nor upgrade firmware

rory, 2015-01-11:
Hi, I have a TomTom Runner, software verison 1.2.0, serial number starts with "HC145...", and when I try to use `ttwatch` I get these errors, and nothing in the `~/gps/tomtomwatch` directory. Do you have any ideas?

    $ ttwatch -a -s ~/gps/tomtomwatch/
    Unable to read file data: 0x00850000
    Unable to set watch time
    Unable to read file data: 0x00910000
    Unable to read file data: 0x00910001
    Unable to read file data: 0x00910002
    Unable to read file data: 0x00910004
    Unable to read file data: 0x00910005
    Unable to read file data: 0x00910003
    Unable to read file data: 0x00910006
    Unable to read file data: 0x00910007
    Unable to read file data: 0x00910008
    Unable to read file data: 0x00910009
    Unable to read file data: 0x0091000a
    Downloading GPSQuickFix data file...
    Writing file to watch...
    Unable to read file data: 0x00010100
    GPSQuickFix data update failed
    Sorry, BLE firmware updating not supported yet

ryanbinns, 2015-01-12:
Hi Rory,

Have you tried running it as root (using sudo)? If it works, it's almost certainly a permissions issue with accessing the usb devices. There is information in the README file about how to grant access to non-root users.

If it doesn't work as root, then it may be something to do with being a Runner watch vs a Multisport. I have only tested it with my Multisport and the only feedback I have received is from Multisport users. There is a "--packets" option which logs all communications with the watch to standard output. If you run that and send me the resulting data, I'll have a look and see what's going on.

Hope this helps,
Ryan

rory, 2015-01-12:
Hi. Yes I get the same results when running as root. (I have also added my self the right group and relogged in). I'll try the `--packets` thing later. Thanks. :)

rory, 2015-01-14:
Here's the output of running that command with `--packets`: http://paste.ubuntu.com/9751586/

(It's 160KB, hence the pastebin)

ryanbinns, 2015-01-15:
> Here's the output of running that command with --packets: http://paste.ubuntu.com/9751586/ [1]

Thanks for that. It looks like your watch is returning different data to
every watch I've seen before, but as I said, I've only seen data from
Multisports before. To be able to see exactly what the difference is,
I'd need to get some raw data from the communcations between the TomTom
Connect software and your watch. It's a bit tricky, but if you would
like to help out (and you have a Windows machine available), I'd
appreciate it. What you'd need to do is this: 

1. Download and install USBPcap from http://desowin.org/usbpcap/ [3] 

2. Run the USBPcapCMD program that was installed. There is no shortcut
in the start menu, so browse to the folder it installed into. It should
run as Administrator, but if not, you'll need to run it explicitly as
Administrator. 

3. Make sure the watch is _NOT_ docked. 

4. Screenshot or otherwise take note of the list of devices that are
listed by the program. 

5. Exit the program. 

6. Dock the watch. 

7. Run USBPCapCMD again and find the device appears that didn't appear
in the last list. Note which filter number it is listed under (this is
important). If the watch is listed under the same filter number as
another HID Input Device (mouse or keyboard), I would prefer if you move
it to another USB port as the program will collect data from _ALL_ the
devices under the selected filter number, and the fewer there are, the
easier it is to understand the data. 

8. Undock the watch. 

9. Create a dummy activity on the watch. The quickest way is to go to
treadmill mode and just shake the watch for 20 seconds or so. 

10. In the USBPcap screen, enter the filter number noted earlier and
press enter. 

11. Enter a file name to save to and press enter. No paths are allowed,
and it should have the .pcap extension. It will be stored in the USBPcap
program directory. 

12. Dock the watch and wait for the TomTom Connect software to do its
thing. If the mouse or keyboard are on the same filter number as the
watch, DO NOT touch them until the processing is complete. 

13. When the TomTom Connect software has completed, wait a few seconds
then undock the watch. 

14. Switch back to the USBPCapCMD window and hit Ctrl+C to finish the
capture. 

15. Send me the file that was captured. 

Hopefully this will give me enough data to understand what your watch is
doing and why it is different to the Multisport watches. 

Thanks for your help, 

Ryan 

 

Links:
------
[1] http://paste.ubuntu.com/9751586/
[2] https://github.com/ryanbinns/ttwatch/issues/7#issuecomment-69976989
[3] http://desowin.org/usbpcap/

adnz, 2015-01-15:
I suggest you try to update via the Windows software first. I have TomTom
runner 1.8.25 and it works. Sometimes there are errors but I just run it
again and it is usually OK.

Ad

ryanbinns, 2015-01-16:
Adnz, yes good point, I missed how old the firmware version was. That is almost definitely the problem. BTW, what sorts of errors do you get? I haven't seen any errors for a long time. I go for a run 2-3 times a week, so I'm using it reasonably often (although I've got it set up as a daemon, so I don't actually start the program all the time).

Rory, if you do use the Windows software to update to the latest firmware, I would appreciate it if you could use the USB packet capturing method I outlined, as there are some aspects of firmware updating I haven't been able to capture and analyse yet. There are two firmware files on the device, and I have captured an update of one of them, but not the other, as my watch has not needed to update the second one yet. Your watch needs to update both.

Cheers,
Ryan

ryanbinns, 2015-02-10:
Rory, do you have any update on this? If not I'll close the issue

closed 2015-04-12 by ryanbinns