ryanbinns/ttwatch #15: Downloading data from tomtom watch

dbrb2, 2015-04-11:
Hi,
This could be user error...
I have a tomtom cardio runner watch running latest firmware
Your software built fine, and some queries work. For instance: 

sudo ttwatch --list-history
Running:
1: 2015/04/09 18:29:03,  210s,   121.98m,    7 calories

But trying to download an activity gives:

sudo ttwatch -d 0 --get-activities -s ~/Desktop
*** Error in `ttwatch': free(): invalid pointer: 0x00007fff955d1923 ***

Getting firmware version returns, but with an error: 

sudo ttwatch -d 0 -v
Product ID: 0xeb030000
BLE Version: 24
Firmware Version: 1.8.34
Watch Name: benWatch
Serial Number: HE2434G01487
*** Error in `ttwatch': free(): invalid pointer: 0x00007fff6450d969 **

Doing the same with a packet dump gives: 

sudo ttwatch --packets -d 0 -v
09 06 00 06 00 F2 00 00 
01 16 00 06 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 06 01 05 00 F2 00 00 
01 16 01 05 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 02 EE 00 00 00 00 
09 0A 02 07 00 F2 00 00 00 00 00 32 
01 3C 02 09 00 F2 00 00 00 00 00 32 3C 3F 78 6D 6C 20 76 65 72 73 69 6F 6E 3D 22 31 2E 30 22 20 65 6E 63 6F 64 69 6E 67 3D 22 55 54 46 2D 38 22 3F 3E 0A 3C 70 72 65 66 65 72 65 6E 63 65 
09 0A 03 07 00 F2 00 00 00 00 00 32 
01 3C 03 09 00 F2 00 00 00 00 00 32 73 20 76 65 72 73 69 6F 6E 3D 22 31 22 20 6D 6F 64 69 66 69 65 64 3D 22 46 72 69 20 31 30 2E 20 41 70 72 20 31 30 3A 30 35 3A 35 39 20 32 30 31 35 22 
09 0A 04 07 00 F2 00 00 00 00 00 32 
01 3C 04 09 00 F2 00 00 00 00 00 32 3E 0A 20 20 20 20 3C 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 30 3C 2F 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 0A 20 20 20 
09 0A 05 07 00 F2 00 00 00 00 00 32 
01 3C 05 09 00 F2 00 00 00 00 00 32 20 3C 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 31 3C 2F 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 0A 20 20 20 20 3C 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 
09 0A 06 07 00 F2 00 00 00 00 00 32 
01 3C 06 09 00 F2 00 00 00 00 00 32 44 61 74 61 3E 30 3C 2F 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 44 61 74 61 3E 0A 20 20 20 20 3C 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 
09 0A 07 07 00 F2 00 00 00 00 00 32 
01 3C 07 09 00 F2 00 00 00 00 00 32 65 64 3E 31 3C 2F 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 65 64 3E 0A 20 20 20 20 3C 77 61 74 63 68 4E 61 6D 65 3E 62 65 6E 57 61 74 63 
09 0A 08 07 00 F2 00 00 00 00 00 32 
01 3C 08 09 00 F2 00 00 00 00 00 32 68 3C 2F 77 61 74 63 68 4E 61 6D 65 3E 0A 20 20 20 20 3C 43 6F 6E 66 69 67 55 52 4C 3E 68 74 74 70 73 3A 2F 2F 6D 79 73 70 6F 72 74 73 2E 74 6F 6D 74 
09 0A 09 07 00 F2 00 00 00 00 00 32 
01 3C 09 09 00 F2 00 00 00 00 00 32 6F 6D 2E 63 6F 6D 2F 73 65 72 76 69 63 65 2F 63 6F 6E 66 69 67 2F 63 6F 6E 66 69 67 2E 6A 73 6F 6E 3C 2F 43 6F 6E 66 69 67 55 52 4C 3E 0A 20 20 20 20 
09 0A 0A 07 00 F2 00 00 00 00 00 32 
01 3C 0A 09 00 F2 00 00 00 00 00 32 3C 65 78 70 6F 72 74 65 72 73 3E 0A 20 20 20 20 20 20 20 20 3C 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 65 78 70 6F 72 74 20 69 
09 0A 0B 07 00 F2 00 00 00 00 00 32 
01 3C 0B 09 00 F2 00 00 00 00 00 32 64 3D 22 4D 79 53 70 6F 72 74 73 22 20 61 75 74 6F 4F 70 65 6E 3D 22 31 22 2F 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 41 
09 0A 0C 07 00 F2 00 00 00 00 00 32 
01 3C 0C 09 00 F2 00 00 00 00 00 32 75 74 68 54 6F 6B 65 6E 3E 30 2E 70 7C 40 5E 15 5C 5A 60 79 75 26 67 74 3B 00 2F 01 12 55 30 0D 24 56 26 61 6D 70 3B 08 0E 5D 1B 06 18 57 3C 2F 4D 79 
09 0A 0D 07 00 F2 00 00 00 00 00 32 
01 3C 0D 09 00 F2 00 00 00 00 00 32 53 70 6F 72 74 73 41 75 74 68 54 6F 6B 65 6E 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 74 3E 
09 0A 0E 07 00 F2 00 00 00 00 00 32 
01 3C 0E 09 00 F2 00 00 00 00 00 32 03 01 74 64 67 43 1E 47 67 07 7C 71 21 7B 11 79 19 07 27 00 01 02 13 51 07 51 02 06 1D 06 3C 2F 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 
09 0A 0F 07 00 F2 00 00 00 00 00 32 
01 3C 0F 09 00 F2 00 00 00 00 00 32 74 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 30 3C 2F 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 0A 
09 0A 10 07 00 F2 00 00 00 00 00 32 
01 3C 10 09 00 F2 00 00 00 00 00 32 20 20 20 20 20 20 20 20 3C 2F 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 3C 2F 65 78 70 6F 72 74 65 72 73 3E 0A 3C 2F 70 72 65 66 65 72 65 6E 63 65 73 3E 0A 
09 06 11 0C 00 F2 00 00 
01 16 11 0C 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 02 12 0D 
01 16 12 0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 02 13 0A 
01 02 13 0A 
09 02 14 22 
01 03 14 22 00 
09 02 15 22 
[... 143 lines cut ...]
09 0A 28 07 00 F2 00 00 00 00 00 32 
01 3C 28 09 00 F2 00 00 00 00 00 32 3C 65 78 70 6F 72 74 65 72 73 3E 0A 20 20 20 20 20 20 20 20 3C 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 65 78 70 6F 72 74 20 69 
09 0A 29 07 00 F2 00 00 00 00 00 32 
01 3C 29 09 00 F2 00 00 00 00 00 32 64 3D 22 4D 79 53 70 6F 72 74 73 22 20 61 75 74 6F 4F 70 65 6E 3D 22 31 22 2F 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 41 
09 0A 2A 07 00 F2 00 00 00 00 00 32 
01 3C 2A 09 00 F2 00 00 00 00 00 32 75 74 68 54 6F 6B 65 6E 3E 30 2E 70 7C 40 5E 15 5C 5A 60 79 75 26 67 74 3B 00 2F 01 12 55 30 0D 24 56 26 61 6D 70 3B 08 0E 5D 1B 06 18 57 3C 2F 4D 79 
09 0A 2B 07 00 F2 00 00 00 00 00 32 
01 3C 2B 09 00 F2 00 00 00 00 00 32 53 70 6F 72 74 73 41 75 74 68 54 6F 6B 65 6E 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 74 3E 
09 0A 2C 07 00 F2 00 00 00 00 00 32 
01 3C 2C 09 00 F2 00 00 00 00 00 32 03 01 74 64 67 43 1E 47 67 07 7C 71 21 7B 11 79 19 07 27 00 01 02 13 51 07 51 02 06 1D 06 3C 2F 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 
09 0A 2D 07 00 F2 00 00 00 00 00 32 
01 3C 2D 09 00 F2 00 00 00 00 00 32 74 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 30 3C 2F 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 0A 
09 0A 2E 07 00 F2 00 00 00 00 00 32 
01 3C 2E 09 00 F2 00 00 00 00 00 32 20 20 20 20 20 20 20 20 3C 2F 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 3C 2F 65 78 70 6F 72 74 65 72 73 3E 0A 3C 2F 70 72 65 66 65 72 65 6E 63 65 73 3E 0A 
09 06 2F 0C 00 F2 00 00 
01 16 2F 0C 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Watch Name: benWatch
Serial Number: HE2434G01487
*** Error in `ttwatch': munmap_chunk(): invalid pointer: 0x00007fff0152f95f ***

Any thoughts? 

Cheers,

Ben

dbrb2, 2015-04-12:
Another dump, if relevant:

$ ttwatch --packets --get-formats
09 06 00 06 00 F2 00 00 
01 16 00 06 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 06 01 05 00 F2 00 00 
01 16 01 05 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 02 EE 00 00 00 00 
09 0A 02 07 00 F2 00 00 00 00 00 32 
01 3C 02 09 00 F2 00 00 00 00 00 32 3C 3F 78 6D 6C 20 76 65 72 73 69 6F 6E 3D 22 31 2E 30 22 20 65 6E 63 6F 64 69 6E 67 3D 22 55 54 46 2D 38 22 3F 3E 0A 3C 70 72 65 66 65 72 65 6E 63 65 
09 0A 03 07 00 F2 00 00 00 00 00 32 
01 3C 03 09 00 F2 00 00 00 00 00 32 73 20 76 65 72 73 69 6F 6E 3D 22 31 22 20 6D 6F 64 69 66 69 65 64 3D 22 46 72 69 20 31 30 2E 20 41 70 72 20 31 30 3A 30 35 3A 35 39 20 32 30 31 35 22 
09 0A 04 07 00 F2 00 00 00 00 00 32 
01 3C 04 09 00 F2 00 00 00 00 00 32 3E 0A 20 20 20 20 3C 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 30 3C 2F 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 0A 20 20 20 
09 0A 05 07 00 F2 00 00 00 00 00 32 
01 3C 05 09 00 F2 00 00 00 00 00 32 20 3C 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 31 3C 2F 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 0A 20 20 20 20 3C 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 
09 0A 06 07 00 F2 00 00 00 00 00 32 
01 3C 06 09 00 F2 00 00 00 00 00 32 44 61 74 61 3E 30 3C 2F 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 44 61 74 61 3E 0A 20 20 20 20 3C 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 
09 0A 07 07 00 F2 00 00 00 00 00 32 
01 3C 07 09 00 F2 00 00 00 00 00 32 65 64 3E 31 3C 2F 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 65 64 3E 0A 20 20 20 20 3C 77 61 74 63 68 4E 61 6D 65 3E 62 65 6E 57 61 74 63 
09 0A 08 07 00 F2 00 00 00 00 00 32 
01 3C 08 09 00 F2 00 00 00 00 00 32 68 3C 2F 77 61 74 63 68 4E 61 6D 65 3E 0A 20 20 20 20 3C 43 6F 6E 66 69 67 55 52 4C 3E 68 74 74 70 73 3A 2F 2F 6D 79 73 70 6F 72 74 73 2E 74 6F 6D 74 
09 0A 09 07 00 F2 00 00 00 00 00 32 
01 3C 09 09 00 F2 00 00 00 00 00 32 6F 6D 2E 63 6F 6D 2F 73 65 72 76 69 63 65 2F 63 6F 6E 66 69 67 2F 63 6F 6E 66 69 67 2E 6A 73 6F 6E 3C 2F 43 6F 6E 66 69 67 55 52 4C 3E 0A 20 20 20 20 
09 0A 0A 07 00 F2 00 00 00 00 00 32 
01 3C 0A 09 00 F2 00 00 00 00 00 32 3C 65 78 70 6F 72 74 65 72 73 3E 0A 20 20 20 20 20 20 20 20 3C 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 65 78 70 6F 72 74 20 69 
09 0A 0B 07 00 F2 00 00 00 00 00 32 
01 3C 0B 09 00 F2 00 00 00 00 00 32 64 3D 22 4D 79 53 70 6F 72 74 73 22 20 61 75 74 6F 4F 70 65 6E 3D 22 31 22 2F 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 41 
09 0A 0C 07 00 F2 00 00 00 00 00 32 
01 3C 0C 09 00 F2 00 00 00 00 00 32 75 74 68 54 6F 6B 65 6E 3E 30 2E 70 7C 40 5E 15 5C 5A 60 79 75 26 67 74 3B 00 2F 01 12 55 30 0D 24 56 26 61 6D 70 3B 08 0E 5D 1B 06 18 57 3C 2F 4D 79 
09 0A 0D 07 00 F2 00 00 00 00 00 32 
01 3C 0D 09 00 F2 00 00 00 00 00 32 53 70 6F 72 74 73 41 75 74 68 54 6F 6B 65 6E 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 74 3E 
09 0A 0E 07 00 F2 00 00 00 00 00 32 
01 3C 0E 09 00 F2 00 00 00 00 00 32 03 01 74 64 67 43 1E 47 67 07 7C 71 21 7B 11 79 19 07 27 00 01 02 13 51 07 51 02 06 1D 06 3C 2F 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 
09 0A 0F 07 00 F2 00 00 00 00 00 32 
01 3C 0F 09 00 F2 00 00 00 00 00 32 74 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 30 3C 2F 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 0A 
09 0A 10 07 00 F2 00 00 00 00 00 32 
01 3C 10 09 00 F2 00 00 00 00 00 32 20 20 20 20 20 20 20 20 3C 2F 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 3C 2F 65 78 70 6F 72 74 65 72 73 3E 0A 3C 2F 70 72 65 66 65 72 65 6E 63 65 73 3E 0A 
09 06 11 0C 00 F2 00 00 
01 16 11 0C 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 02 12 0D 
01 16 12 0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 02 13 0A 
01 02 13 0A 
09 02 14 22 
01 03 14 22 00 
09 02 15 22 
01 03 15 22 00 
09 02 16 20 
01 06 16 20 EB 03 00 00 
09 02 17 0A 
01 02 17 0A 
09 02 18 28 
01 06 18 28 00 00 00 18 
09 02 19 28 
01 06 19 28 00 00 00 18 
09 02 1A 1F 
01 06 1A 1F 00 00 00 00 
09 06 1B 06 00 F2 00 00 
01 16 1B 06 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 06 1C 05 00 F2 00 00 
01 16 1C 05 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 02 EE 00 00 00 00 
09 0A 1D 07 00 F2 00 00 00 00 00 32 
01 3C 1D 09 00 F2 00 00 00 00 00 32 3C 3F 78 6D 6C 20 76 65 72 73 69 6F 6E 3D 22 31 2E 30 22 20 65 6E 63 6F 64 69 6E 67 3D 22 55 54 46 2D 38 22 3F 3E 0A 3C 70 72 65 66 65 72 65 6E 63 65 
09 0A 1E 07 00 F2 00 00 00 00 00 32 
01 3C 1E 09 00 F2 00 00 00 00 00 32 73 20 76 65 72 73 69 6F 6E 3D 22 31 22 20 6D 6F 64 69 66 69 65 64 3D 22 46 72 69 20 31 30 2E 20 41 70 72 20 31 30 3A 30 35 3A 35 39 20 32 30 31 35 22 
09 0A 1F 07 00 F2 00 00 00 00 00 32 
01 3C 1F 09 00 F2 00 00 00 00 00 32 3E 0A 20 20 20 20 3C 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 30 3C 2F 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 0A 20 20 20 
09 0A 20 07 00 F2 00 00 00 00 00 32 
01 3C 20 09 00 F2 00 00 00 00 00 32 20 3C 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 31 3C 2F 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 0A 20 20 20 20 3C 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 
09 0A 21 07 00 F2 00 00 00 00 00 32 
01 3C 21 09 00 F2 00 00 00 00 00 32 44 61 74 61 3E 30 3C 2F 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 44 61 74 61 3E 0A 20 20 20 20 3C 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 
09 0A 22 07 00 F2 00 00 00 00 00 32 
[... cut ...]
01 3C 2A 09 00 F2 00 00 00 00 00 32 74 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 30 3C 2F 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 0A 
09 0A 2B 07 00 F2 00 00 00 00 00 32 
01 3C 2B 09 00 F2 00 00 00 00 00 32 20 20 20 20 20 20 20 20 3C 2F 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 3C 2F 65 78 70 6F 72 74 65 72 73 3E 0A 3C 2F 70 72 65 66 65 72 65 6E 63 65 73 3E 0A 
09 06 2C 0C 00 F2 00 00 
01 16 2C 0C 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Unable to read watch preferences

ryanbinns, 2015-04-12:
Ok. Give me a few days as I've just got back from holidays and I'll have a look at it.

Cheers,
Ryan

dbrb2, 2015-04-12:
Cheers. I could hardly complain about delays on an open source project with
one maintainer :-)

Whether or not you are able to fix it, thanks very much for the work done
to date

Ben

ryanbinns, 2015-04-12:
It was easier than I expected. Have another go now.

Cheers,
Ryan

dbrb2, 2015-04-12:
That has certainly made a difference: 

ttwatch --get-formats

No longer throws an error, but it does not seem to return anything either

$ ttwatch --packets --get-formats
09 06 00 06 00 F2 00 00 
01 16 00 06 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 06 01 05 00 F2 00 00 
01 16 01 05 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 02 EE 00 00 00 00 
09 0A 02 07 00 F2 00 00 00 00 00 32 
01 3C 02 09 00 F2 00 00 00 00 00 32 3C 3F 78 6D 6C 20 76 65 72 73 69 6F 6E 3D 22 31 2E 30 22 20 65 6E 63 6F 64 69 6E 67 3D 22 55 54 46 2D 38 22 3F 3E 0A 3C 70 72 65 66 65 72 65 6E 63 65 
09 0A 03 07 00 F2 00 00 00 00 00 32 
01 3C 03 09 00 F2 00 00 00 00 00 32 73 20 76 65 72 73 69 6F 6E 3D 22 31 22 20 6D 6F 64 69 66 69 65 64 3D 22 46 72 69 20 31 30 2E 20 41 70 72 20 31 30 3A 30 35 3A 35 39 20 32 30 31 35 22 
09 0A 04 07 00 F2 00 00 00 00 00 32 
01 3C 04 09 00 F2 00 00 00 00 00 32 3E 0A 20 20 20 20 3C 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 30 3C 2F 65 70 68 65 6D 65 72 69 73 4D 6F 64 69 66 69 65 64 3E 0A 20 20 20 
09 0A 05 07 00 F2 00 00 00 00 00 32 
01 3C 05 09 00 F2 00 00 00 00 00 32 20 3C 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 31 3C 2F 53 79 6E 63 54 69 6D 65 54 6F 50 43 3E 0A 20 20 20 20 3C 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 
09 0A 06 07 00 F2 00 00 00 00 00 32 
01 3C 06 09 00 F2 00 00 00 00 00 32 44 61 74 61 3E 30 3C 2F 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73 44 61 74 61 3E 0A 20 20 20 20 3C 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 
09 0A 07 07 00 F2 00 00 00 00 00 32 
01 3C 07 09 00 F2 00 00 00 00 00 32 65 64 3E 31 3C 2F 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 65 64 3E 0A 20 20 20 20 3C 77 61 74 63 68 4E 61 6D 65 3E 62 65 6E 57 61 74 63 
09 0A 08 07 00 F2 00 00 00 00 00 32 
01 3C 08 09 00 F2 00 00 00 00 00 32 68 3C 2F 77 61 74 63 68 4E 61 6D 65 3E 0A 20 20 20 20 3C 43 6F 6E 66 69 67 55 52 4C 3E 68 74 74 70 73 3A 2F 2F 6D 79 73 70 6F 72 74 73 2E 74 6F 6D 74 
09 0A 09 07 00 F2 00 00 00 00 00 32 
01 3C 09 09 00 F2 00 00 00 00 00 32 6F 6D 2E 63 6F 6D 2F 73 65 72 76 69 63 65 2F 63 6F 6E 66 69 67 2F 63 6F 6E 66 69 67 2E 6A 73 6F 6E 3C 2F 43 6F 6E 66 69 67 55 52 4C 3E 0A 20 20 20 20 
09 0A 0A 07 00 F2 00 00 00 00 00 32 
01 3C 0A 09 00 F2 00 00 00 00 00 32 3C 65 78 70 6F 72 74 65 72 73 3E 0A 20 20 20 20 20 20 20 20 3C 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 65 78 70 6F 72 74 20 69 
09 0A 0B 07 00 F2 00 00 00 00 00 32 
01 3C 0B 09 00 F2 00 00 00 00 00 32 64 3D 22 4D 79 53 70 6F 72 74 73 22 20 61 75 74 6F 4F 70 65 6E 3D 22 31 22 2F 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 41 
09 0A 0C 07 00 F2 00 00 00 00 00 32 
01 3C 0C 09 00 F2 00 00 00 00 00 32 75 74 68 54 6F 6B 65 6E 3E 30 2E 70 7C 40 5E 15 5C 5A 60 79 75 26 67 74 3B 00 2F 01 12 55 30 0D 24 56 26 61 6D 70 3B 08 0E 5D 1B 06 18 57 3C 2F 4D 79 
09 0A 0D 07 00 F2 00 00 00 00 00 32 
01 3C 0D 09 00 F2 00 00 00 00 00 32 53 70 6F 72 74 73 41 75 74 68 54 6F 6B 65 6E 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 74 3E 
09 0A 0E 07 00 F2 00 00 00 00 00 32 
01 3C 0E 09 00 F2 00 00 00 00 00 32 03 01 74 64 67 43 1E 47 67 07 7C 71 21 7B 11 79 19 07 27 00 01 02 13 51 07 51 02 06 1D 06 3C 2F 4D 79 53 70 6F 72 74 73 54 6F 6B 65 6E 53 65 63 72 65 
09 0A 0F 07 00 F2 00 00 00 00 00 32 
01 3C 0F 09 00 F2 00 00 00 00 00 32 74 3E 0A 20 20 20 20 20 20 20 20 20 20 20 20 3C 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 30 3C 2F 4D 79 53 70 6F 72 74 73 55 73 65 72 49 64 3E 0A 
09 0A 10 07 00 F2 00 00 00 00 00 32 
01 3C 10 09 00 F2 00 00 00 00 00 32 20 20 20 20 20 20 20 20 3C 2F 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 3C 2F 65 78 70 6F 72 74 65 72 73 3E 0A 3C 2F 70 72 65 66 65 72 65 6E 63 65 73 3E 0A 
09 06 11 0C 00 F2 00 00 
01 16 11 0C 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 02 12 0D 
01 16 12 0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 02 13 0A 
01 02 13 0A 
09 02 14 22 
01 03 14 22 00 
09 02 15 22 
01 03 15 22 00 
09 02 16 20 
01 06 16 20 EB 03 00 00 
09 02 17 0A 
01 02 17 0A 
09 02 18 28 
01 06 18 28 00 00 00 18 
09 02 19 28 
01 06 19 28 00 00 00 18 
09 02 1A 1F 
01 06 1A 1F 00 00 00 00 
09 06 1B 06 00 F2 00 00 
01 16 1B 06 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
09 06 1C 05 00 F2 00 00 
01 16 1C 05 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 02 EE 00 00 00 00 
09 0A 1D 07 00 F2 00 00 00 00 00 32 
01 3C 1D 09 00 F2 00 00 00 00 00 32 3C 3F 78 6D 6C 20 76 65 72 73 69 6F 6E 3D 22 31 2E 30 22 20 65 6E 63 6F 64 69 6E 67 3D 22 55 54 46 2D 38 22 3F 3E 0A 3C 70 72 65 66 65 72 65 6E 63 65 
09 0A 1E 07 00 F2 00 00 00 00 00 32 
01 3C 1E 09 00 F2 00 00 00 00 00 32 73 20 76 65 72 73 69 6F 6E 3D 22 31 22 20 6D 6F 64 69 66 69 65 64 3D 22 46 72 69 20 31 30 2E 20 41 70 72 20 31 30 3A 30 35 3A 35 39 20 32 30 31 35 22 
09 0A 1F 07 00 F2 00 00 00 00 00 32 
[... 5 lines cut ...]
09 0A 22 07 00 F2 00 00 00 00 00 32 
01 3C 22 09 00 F2 00 00 00 00 00 32 65 64 3E 31 3C 2F 57 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E 69 6D 69 7A 65 64 3E 0A 20 20 20 20 3C 77 61 74 63 68 4E 61 6D 65 3E 62 65 6E 57 61 74 63 
09 0A 23 07 00 F2 00 00 00 00 00 32 
01 3C 23 09 00 F2 00 00 00 00 00 32 68 3C 2F 77 61 74 63 68 4E 61 6D 65 3E 0A 20 20 20 20 3C 43 6F 6E 66 69 67 55 52 4C 3E 68 74 74 70 73 3A 2F 2F 6D 79 73 70 6F 72 74 73 2E 74 6F 6D 74 
09 0A 24 07 00 F2 00 00 00 00 00 32 
[... cut ...]
09 0A 2B 07 00 F2 00 00 00 00 00 32 
01 3C 2B 09 00 F2 00 00 00 00 00 32 20 20 20 20 20 20 20 20 3C 2F 6F 6E 6C 69 6E 65 3E 0A 20 20 20 20 3C 2F 65 78 70 6F 72 74 65 72 73 3E 0A 3C 2F 70 72 65 66 65 72 65 6E 63 65 73 3E 0A 
09 06 2C 0C 00 F2 00 00 
01 16 2C 0C 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

If it is behaving properly, but for whatever reason none of the output formats are recognized by the watch, would that explain why there is no data downloaded when I run: 
ttwatch -a -s ~/Desktop

ryanbinns, 2015-04-12:
I checked your preferences file (from the packet dump), and you don't currently have any offline exporters set (the program ignores the web-based ones as it doesn't do anything web-based). You only have a Mysports exporter set. This is why you don't see any output. Previously, the program was reporting an error in this case, but I removed the error message as it's not really the correct behaviour. You would need to add offline exporters using the --set-formats command or using the Windows client to see some output. Currently csv, tcx, pwx and kml formats are supported.

Yes, this will also stop any files being created when downloading activities. You should still see the .ttbin file saved in ~/Desktop, but there won't be any other files there. They should appear if you set some file formats and download another activity.

Cheers,
Ryan

dbrb2, 2015-04-12:
Ah Ok. Makes some sense now:

I added some output formats:

$ ttwatch --get-formats
csv kml pwx tcx

Querying the unit for stored activities returned the  stored activities (as
before):

ben@ben-desktop ~/Desktop $ ttwatch --list-history
Treadmill:
1: 2015/04/10 13:25:27,   31s,     0.00m,    0 calories
2: 2015/04/11 22:28:01,   38s,     0.00m,    0 calories
Running:
1: 2015/04/09 18:29:03,  210s,   121.98m,    7 calories
2: 2015/04/12 09:29:16, 2478s,  9954.14m,    0 calories

But these were all activities created prior to adding the output formats
above.
Initially, calling

ben@ben-desktop ~/Desktop $ ttwatch --get-activities -s ~/Desktop
gave no output as before (not even a ttbin file)

I then created a test activity, after having already added the outupt
formats, then ran:

ben@ben-desktop ~/Desktop $ ttwatch --get-activities -s ~/Desktop

This time a new folder was created in the output directory, benWatch,
containing a subfolder of todays date, and inside that was a ttbin file:

Treadmill_13-39-03.ttbin

No files of any other format were present though...

ryanbinns, 2015-04-12:
The activities are deleted from the watch when they are downloaded. The watch maintains a history list (that --list-history shows) which contains basic information about the activity, but the activity data itself is not stored, which means that you cannot re-download activities to create the extra file formats. You can, however, process them using the ttbincnv program to create the extra files.

For your treadmill activity, the kml, tcx and pwx files will not be created as they are dependent on GPS data, which is not available for treadmill files. Only the csv file will be created if it's selected. I'm not sure why the csv file would not be created. I'll have a look into it.

dbrb2, 2015-04-12:
That makes sense:

After using the watch, I used the bluetooth android app from tomtom to
retrieve the activity
That placed a series of ttbin files on my phone, but if it works the same
way your software does, it will also have deleted the copy on the watch
itself, which explains why I couldn't subsequently download it. I assume
this is done for housekeeping to prevent the watch memory clogging up over
time...I thought they were still there since they were still in the history
- but as you say that is just a summary of activities, that can be
explicitly cleared from your app if needed.

Am I right in thinking that although you store on the watch itself the
desired export formats, the only file actually retrieved from the phone is
the ttbin? That file is then parsed locally to produce whatever the desired
final format is?

Cheers,

Ben

dbrb2, 2015-04-12:
Final one for the day:

ttbincnv -a 00910001_20150412_101700.ttbin

Works fine, except for altitude information.

Unable to download elevation data: 56

I had assumed that the altitude was obtained from the GPS fix directly, but
perhaps instead it is a lookup based on known location...? Seems a little
strange if it is the latter, but maybe the designers think the lat/lng is
more accuate than the GPS altitude...

Any idea where this information does (or doesn't) come from...?

ryanbinns, 2015-04-13:
Yes, only the ttbin file is stored on the watch. The other file formats are created from the ttbin file after it is downloaded from the watch. The watch has an XML-format preferences file stored on it that contains (amongst other things) the list of file formats that should be created, Mysports login information (encrypted, don't worry I can't see your details) and watch name.

The watch does not record altitude data. I don't know why not, since GPS has to have altitude to get an accurate lat/long fix anyway. The altitude data is downloaded from TomTom's website. The lat/long points are posted to their web server and the server responds with altitude for each point. It's not the best, but mimics what TomTom's Windows client does. The altitude data is not great in some areas and doesn't contain any information for polar areas higher than 70 degrees (I think, from memory) latitude. What this means is that the program requires an internet connection to download altitude data. Error code 56 is a generic receive error - the request was sent, but there was an error receiving data. The program uses libcurl - the error list is here: http://curl.haxx.se/libcurl/c/libcurl-errors.html

dbrb2, 2015-04-13:
Ah cool
I'll run up wireshark with the myconnect software running to check the URL and post data used for elevation checking. I had a connection, but maybe something else wasn't right. 

I found out why my calorie count was zero  - my height and weight were zero on the watch, so any trip required zero energy :-) I assume that is info that is also stored in the XML, but in this case is not something that can be written back by your application?

ryanbinns, 2015-04-13:
This is where it gets complicated. Height and weight are not in the XML file. The XML file contains only settings used for the computer connection. There is a separate file on the watch which contains all the settings actually used on the watch. Unfortunately, it's binary. I reverse engineered most of the settings for the previous firmware version, but the most recent firmware version seems to have changed the layout of the file, so I'll have to go back through all the settings and try to work them out again. It's a time-consuming process unfortunately, as I have to change one single setting on the watch then download the settings file and see what changed. The previous version had about 170 different settings and most of the settings had a list of values that have numeric equivalents in the binary file so I had to choose *every* option of every setting and see what value was used in the file.

Also, I worked out why your files weren't being converted when they were downloaded from the watch. For some reason, I made it not do so unless the program was being run as a daemon. Anyway, I have changed it so the csv/tcx/gpx/kml files should be created when activities are downloaded.

Cheers,
Ryan

closed 2015-04-13 by ryanbinns

dbrb2, 2015-04-13:
Ouch - that's painful.

I wonder how the official client software dealt with this...Presumably
there was no need to update the client when the firmware was updated, but
the official client could still communicate with the watch both before and
afterwards....unless it noticed the f/w change and updated itself under the
bonnet. You's assume that the TomTom devs would want to avoid doing what
you are faced with now for their own releases :-)