ryanbinns/ttwatch #16: Do you think it would be feasible to flash the Runner watch with the Multi-Sport firmware?

holeyman, 2015-04-18:
I've kept wondering this since the day I got the watch. I know the two watches might have different internals or memory, but my gut tells me its just different firmware on same hardware...

ryanbinns, 2015-04-20:
Maybe. I wouldn't want to try it as it would definitely void the warranty if it doesn't work. I suspect that the current firmware will be able to tell that the new firmware file is not the correct one.

closed 2015-04-20 by ryanbinns

lgrangeia, 2015-06-11:
So you didn't try? I have a runner and I'm dying to try this. Most likely the worst that could happen is the upgrade fails (the current firmware rejects the upgrade).

Let me know if anyone attempted this already.

ryanbinns, 2015-06-12:
I haven't heard of anyone trying this. Obviously, neither the TomTom software or my ttwatch software will do this, so you'd need to use the 'write' command-line option to write the correct firmware file directly.

I'd strongly recommend looking at the ttwatch source code to see how it determines which firmware files to download, as there are more than 10 that need to be updated, and you'd need to do all of them. You may find some strange settings afterwards as well, since the MultiSport-specific settings in the manifest file won't have any valid data.

If you do decide to do this, let me know how it goes. I wouldn't want to try it, just in case, but you never know...

oleg-umnik, 2015-06-12:
If my memory serves me well, there is no difference between firmware for Running and Multisport. So watches decide itself which features should be enabled (for example, based on serial number). Maybe I need to check this again.

lgrangeia, 2015-06-12:
Thank you both!

I haven't tested your software before but I have used a usb sniffer to verify the upgrade procedure in windows. I also suspect the firmware is the same for both versions of the watch, but I will look at the firmware files to validate that, before trying anything risky :)

If I get to a conclusion this weekend I'll let you know.

dlenski, 2015-08-03:
@oleg-umnik is exactly right. The firmware is byte-for-byte identical across Runner and Multi-Sport (my [`fwgrabber` branch](//github.com/dlenski/ttwatch/tree/fwgrabber) has a tool to scrape all the FW versions).

The hardware appears to be identical between the two, so I think it's correct that the firmware soft-disable the cycling/swimming features on watches based on serial number or product IDs (`1001` for Runner, `1002` for Multi-Sport).

lgrangeia, 2015-08-23:
I looked at the firmware files a bit during the weekend. The main firmware file (0x000000F0) appears encrypted using some sort of block cipher using either ECB or counter block mode (not CBC / GCM). My support for this is that multiple versions of the same firmware file differ only by blocks/multiples of 16 bytes.

Unless someone finds the key there doesn't seem to be a way to customize the firmware. There might be a way by tinkering with the other files (the key could be there), but most likely the key is burned on the main SoC's ROM at the factory.

Guess I'll just go out for a run then :)

RoooNY87, 2016-05-31:
Any News? I really warnt tomtom use my Running dir cycling

ryanbinns, 2016-06-01:
No. There is no difference between the firmware for the Runner and Multisport watches. The watches themselves know which features should be enabled. If you want to be able to use the cycling features, you'll have to buy the Multisport, or another watch.