ryanbinns/ttwatch #22: Segmentation fault with --get-activities

lufik, 2015-04-20:
I have a problem to download my last running (Training = None).

Firmware 1.8.25 (I have no windows machine :( )
$ ttwatch --update-fw
Sorry, BLE firmware updating not supported yet

I tried to track it down and the problem is that parse_ttbin_data returns null and it is not tested (expected non-null value) because it's unable to find the right index (for p.record->tag == 255).

I modified sources only with more verbose output (parse_ttbin_data):
```c
    for (p.data = data; p.data < end; p.data += length)
    {
        unsigned index = 0;

        fprintf(stdout, "Next data %d: %p ? %p ....\n", count, p.data, end);
        /* find the length of this tag */
        while ((index < file_header->length_count) && (file_header->lengths[index].tag < p.record->tag))
        {
            fprintf(stdout, "index: %d/%d ; tag: %d ? %d\n", index, file_header->length_count, file_header->lengths[index].tag, p.record->tag);
            ++index;
        }
        if ((index < file_header->length_count) && (file_header->lengths[index].tag == p.record->tag))
            length = file_header->lengths[index].length;
        else
        {
            fprintf(stdout, "Wrong index\n");
            free_ttbin(file);
            return 0;
        }
```

This returns:
```sh
$ ttwatch --get-activities
Next data: 0x7f415b19c0cb ? 0x7f415b1d74de ....
index: 0/23 ; tag: 32 ? 48
index: 1/23 ; tag: 33 ? 48
index: 2/23 ; tag: 34 ? 48
index: 3/23 ; tag: 35 ? 48
index: 4/23 ; tag: 37 ? 48
index: 5/23 ; tag: 39 ? 48
index: 6/23 ; tag: 42 ? 48
index: 7/23 ; tag: 43 ? 48
index: 8/23 ; tag: 45 ? 48
index: 9/23 ; tag: 47 ? 48
Next data: 0x7f415b19c0ce ? 0x7f415b1d74de ....
index: 0/23 ; tag: 32 ? 33
Next data: 0x7f415b19c0d5 ? 0x7f415b1d74de ....
index: 0/23 ; tag: 32 ? 33
Next data: 0x7f415b19c0dc ? 0x7f415b1d74de ....
index: 0/23 ; tag: 32 ? 37
index: 1/23 ; tag: 33 ? 37
index: 2/23 ; tag: 34 ? 37
index: 3/23 ; tag: 35 ? 37
Next data: 0x7f415b19c0e3 ? 0x7f415b1d74de ....
index: 0/23 ; tag: 32 ? 34
index: 1/23 ; tag: 33 ? 34
Next data: 0x7f415b19c0ff ? 0x7f415b1d74de ....
index: 0/23 ; tag: 32 ? 35
index: 1/23 ; tag: 33 ? 35
index: 2/23 ; tag: 34 ? 35
Next data: 0x7f415b19c113 ? 0x7f415b1d74de ....
index: 0/23 ; tag: 32 ? 255
index: 1/23 ; tag: 33 ? 255
index: 2/23 ; tag: 34 ? 255
index: 3/23 ; tag: 35 ? 255
index: 4/23 ; tag: 37 ? 255
index: 5/23 ; tag: 39 ? 255
index: 6/23 ; tag: 42 ? 255
index: 7/23 ; tag: 43 ? 255
index: 8/23 ; tag: 45 ? 255
index: 9/23 ; tag: 47 ? 255
index: 10/23 ; tag: 48 ? 255
index: 11/23 ; tag: 49 ? 255
index: 12/23 ; tag: 50 ? 255
index: 13/23 ; tag: 52 ? 255
index: 14/23 ; tag: 53 ? 255
index: 15/23 ; tag: 55 ? 255
index: 16/23 ; tag: 57 ? 255
index: 17/23 ; tag: 58 ? 255
index: 18/23 ; tag: 59 ? 255
index: 19/23 ; tag: 60 ? 255
index: 20/23 ; tag: 61 ? 255
index: 21/23 ; tag: 62 ? 255
index: 22/23 ; tag: 63 ? 255
Wrong index
Segmentation fault (core dumped)
```

Any idea how to fix it? I don't want to loose this activity. I downloaded the activity file from watch if needed for deeper investigation.

Thanks,

Luf

ryanbinns, 2015-04-21:
That's not good. A tag of 255 is not a valid tag, so it looks like either the file is corrupt, or there is some other sort of error. Can you attach the activity file to this issue? I'll have a look at it when I get a moment.

lufik, 2015-04-21:
I see no usable way how to attach file :( Any suggestion?

ryanbinns, 2015-04-21:
Yep, that's hopelessly corrupted. It's a bug in the watch firmware; I've seen this exact error before. The file is randomly interspersed with 256 consecutive bytes of value 0xff. Unfortunately, it doesn't just insert the wrong bytes, it overwrites what should be there, so the real data is lost. The last 40% or so of the file looks correct, so the data could possibly be extracted by hand in a hex editor if you want to try.

I believe more recent firmware versions have fixed this bug. Unfortunately, the ttwatch can't update your watch because the BLE firmware has changed, and I haven't been able to capture an update of that firmware to work out how to do it yet. The main firmware updating works properly, but unfortunately your watch needs to update both firmware files.

I'll modify the ttwatch software so it checks for a corrupted file and doesn't attempt to do anything except write the file to disk.

ryanbinns, 2015-04-21:
The latest update will not attempt to use a corrupted file - it will simply write it to disk unmodified, then print a warning that the file is corrupted.

closed 2015-04-21 by ryanbinns