ryanbinns/ttwatch #23: Latest ttwatch segfault in --get-formats

lufik, 2015-04-21:
$ ./ttwatch/ttwatch --get-formats
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_S_construct null not valid
Aborted (core dumped)

It fails in libttwatch/libttwatch.cpp - ttwatch_enumerate_offline_formats because watch->preferences_file is NULL

```c
int ttwatch_enumerate_offline_formats(TTWATCH *watch, TTWATCH_FORMAT_ENUMERATOR enumerator, void *data)
{
    if (!watch || !enumerator)
        return TTWATCH_InvalidParameter;

    std::string file = watch->preferences_file;

```

ryanbinns, 2015-04-21:
I'll add the logic to fix this, but I'm surprised it happened. The only way this could have happened is if the program failed to read the watch name, which is does when the watch is opened. Does 'ttwatch -v' work successfully?

lufik, 2015-04-21:
I'm not able to get/set watch name. As I have no windows I changed a little bit the code to upgrade fw (from 1.7.xy to 1.8.25). As you wrote it's not ready so maybe that's the reason. I never connected my watch with TomTom original SW.

```sh
$ ttwatch -v
Product ID: 0xec030000
BLE Version: 19
Firmware Version: 1.8.25
Unable to read watch name
```

ryanbinns, 2015-04-21:
Oh I see, if you've never set the watch name, then it won't have one. Currently, the software will /change/ the name, but will never /add/ it if it doesn't exist. It should be fairly easy to modify to add the name though - it's only changing an XML file. I'll see what I can do.

ryanbinns, 2015-04-21:
I have uploaded a fix that will allow you to set a watch name. It will also return an empty string instead of an error if the watch name has never been set.

lufik, 2015-04-21:
I confirm I can get activities now but it seems it doesn't fix setting the name:

```sh
$ ./ttwatch/ttwatch --set-name="Luf - TomTom GPS watch"
Unable to write new watch name
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_S_construct null not valid
Aborted (core dumped)
```

I'm going to dive deeper.

lufik, 2015-04-21:
The root case of setting watch name is that I have no file "0x00f20000" in my watch.

ryanbinns, 2015-04-22:
Aaah I see. I didn't realise that the watch shipped without a preferences file at all; I assumed it would have a default one. I'll add code to add a default preferences file if one does not exist when the watch is opened. In the meantime, you can fix this by adding one manually. Copy the following to a file and write it to the watch using `ttwatch --write 0x00f20000 <filename>`:

```
<?xml version="1.0" encoding="UTF-8"?>
<preferences version="1" modified="Wed 22. Apr 10:00:00 2015">
    <watchName>Luf</watchName>
    <SyncTimeToPC>1</SyncTimeToPC>
    <SendAnonymousData>0</SendAnonymousData>
    <WatchWindowMinimized>0</WatchWindowMinimized>
    <ConfigURL>https://mysports.tomtom.com/service/config/config.json</ConfigURL>
    <exporters>
        <offline>
        </offline>
    </exporters>
</preferences>
```

Most of the settings in the file are not important if you're not using the Windows client at all, but you may as well keep them there just in case. Obviously the modified time can be any time in the past.

Cheers,
Ryan

lufik, 2015-04-22:
Great. Thank you very much for you support. I'll try it tomorrow and send you the feedback.

lufik, 2015-04-23:
I did what you wrote (just I don't include watchName line) and it failed in:
```sh
$ ttwatch -w 0x00f20000 0x00f20000 
*** Error in `ttwatch': free(): invalid pointer: 0x00007fff46d0355d ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7d1fd)[0x7f48d382b1fd]
ttwatch[0x40ab62]
ttwatch[0x408e33]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f48d37cfaf5]
ttwatch[0x402249]
======= Memory map: ========
00400000-0041d000 r-xp 00000000 fd:00 23225372                           /usr/local/bin/ttwatch
0061d000-0061e000 r--p 0001d000 fd:00 23225372                           /usr/local/bin/ttwatch
0061e000-0061f000 rw-p 0001e000 fd:00 23225372                           /usr/local/bin/ttwatch
01e63000-01e84000 rw-p 00000000 00:00 0                                  [heap]
7f48ce4a2000-7f48ce4a3000 ---p 00000000 00:00 0 
7f48ce4a3000-7f48ceca3000 rw-p 00000000 00:00 0 
7f48ceca3000-7f48ceca5000 r-xp 00000000 fd:00 23065                      /usr/lib64/libfreebl3.so
7f48ceca5000-7f48ceea4000 ---p 00002000 fd:00 23065                      /usr/lib64/libfreebl3.so
7f48ceea4000-7f48ceea5000 r--p 00001000 fd:00 23065                      /usr/lib64/libfreebl3.so
7f48ceea5000-7f48ceea6000 rw-p 00002000 fd:00 23065                      /usr/lib64/libfreebl3.so
7f48ceea6000-7f48ceeae000 r-xp 00000000 fd:00 4767                       /usr/lib64/libcrypt-2.17.so
7f48ceeae000-7f48cf0ad000 ---p 00008000 fd:00 4767                       /usr/lib64/libcrypt-2.17.so
7f48cf0ad000-7f48cf0ae000 r--p 00007000 fd:00 4767                       /usr/lib64/libcrypt-2.17.so
7f48cf0ae000-7f48cf0af000 rw-p 00008000 fd:00 4767                       /usr/lib64/libcrypt-2.17.so
7f48cf0af000-7f48cf0dd000 rw-p 00000000 00:00 0 
7f48cf0dd000-7f48cf101000 r-xp 00000000 fd:00 74889                      /usr/lib64/liblzma.so.5.0.99
7f48cf101000-7f48cf300000 ---p 00024000 fd:00 74889                      /usr/lib64/liblzma.so.5.0.99
7f48cf300000-7f48cf301000 r--p 00023000 fd:00 74889                      /usr/lib64/liblzma.so.5.0.99
7f48cf301000-7f48cf302000 rw-p 00024000 fd:00 74889                      /usr/lib64/liblzma.so.5.0.99
7f48cf302000-7f48cf361000 r-xp 00000000 fd:00 74891                      /usr/lib64/libpcre.so.1.2.0
7f48cf361000-7f48cf561000 ---p 0005f000 fd:00 74891                      /usr/lib64/libpcre.so.1.2.0
7f48cf561000-7f48cf562000 r--p 0005f000 fd:00 74891                      /usr/lib64/libpcre.so.1.2.0
7f48cf562000-7f48cf563000 rw-p 00060000 fd:00 74891                      /usr/lib64/libpcre.so.1.2.0
7f48cf563000-7f48cf57f000 r-xp 00000000 fd:00 76933                      /usr/lib64/libsasl2.so.3.0.0
7f48cf57f000-7f48cf77e000 ---p 0001c000 fd:00 76933                      /usr/lib64/libsasl2.so.3.0.0
7f48cf77e000-7f48cf77f000 r--p 0001b000 fd:00 76933                      /usr/lib64/libsasl2.so.3.0.0
7f48cf77f000-7f48cf780000 rw-p 0001c000 fd:00 76933                      /usr/lib64/libsasl2.so.3.0.0
7f48cf780000-7f48cf796000 r-xp 00000000 fd:00 1242645                    /usr/lib64/libresolv-2.17.so
7f48cf796000-7f48cf996000 ---p 00016000 fd:00 1242645                    /usr/lib64/libresolv-2.17.so
7f48cf996000-7f48cf997000 r--p 00016000 fd:00 1242645                    /usr/lib64/libresolv-2.17.so
7f48cf997000-7f48cf998000 rw-p 00017000 fd:00 1242645                    /usr/lib64/libresolv-2.17.so
7f48cf998000-7f48cf99a000 rw-p 00000000 00:00 0 
7f48cf99a000-7f48cf99d000 r-xp 00000000 fd:00 77084                      /usr/lib64/libkeyutils.so.1.5
7f48cf99d000-7f48cfb9c000 ---p 00003000 fd:00 77084                      /usr/lib64/libkeyutils.so.1.5
7f48cfb9c000-7f48cfb9d000 r--p 00002000 fd:00 77084                      /usr/lib64/libkeyutils.so.1.5
7f48cfb9d000-7f48cfb9e000 rw-p 00003000 fd:00 77084                      /usr/lib64/libkeyutils.so.1.5
7f48cfb9e000-7f48cfbab000 r-xp 00000000 fd:00 883544                     /usr/lib64/libkrb5support.so.0.1
7f48cfbab000-7f48cfdab000 ---p 0000d000 fd:00 883544                     /usr/lib64/libkrb5support.so.0.1
7f48cfdab000-7f48cfdac000 r--p 0000d000 fd:00 883544                     /usr/lib64/libkrb5support.so.0.1
7f48cfdac000-7f48cfdad000 rw-p 0000e000 fd:00 883544                     /usr/lib64/libkrb5support.so.0.1
7f48cfdad000-7f48cff6a000 r-xp 00000000 fd:00 444126                     /usr/lib64/libcrypto.so.1.0.1e
7f48cff6a000-7f48d016a000 ---p 001bd000 fd:00 444126                     /usr/lib64/libcrypto.so.1.0.1e
7f48d016a000-7f48d0184000 r--p 001bd000 fd:00 444126                     /usr/lib64/libcrypto.so.1.0.1e
7f48d0184000-7f48d0190000 rw-p 001d7000 fd:00 444126                     /usr/lib64/libcrypto.so.1.0.1e
7f48d0190000-7f48d0194000 rw-p 00000000 00:00 0 
7f48d0194000-7f48d01f6000 r-xp 00000000 fd:00 793                        /usr/lib64/libssl.so.1.0.1e
7f48d01f6000-7f48d03f6000 ---p 00062000 fd:00 793                        /usr/lib64/libssl.so.1.0.1e
7f48d03f6000-7f48d03fa000 r--p 00062000 fd:00 793                        /usr/lib64/libssl.so.1.0.1e
7f48d03fa000-7f48d0401000 rw-p 00066000 fd:00 793                        /usr/lib64/libssl.so.1.0.1e
7f48d0401000-7f48d0416000 r-xp 00000000 fd:00 113954                     /usr/lib64/libgcc_s-4.8.3-20140911.so.1
7f48d0416000-7f48d0615000 ---p 00015000 fd:00 113954                     /usr/lib64/libgcc_s-4.8.3-20140911.so.1
7f48d0615000-7f48d0616000 r--p 00014000 fd:00 113954                     /usr/lib64/libgcc_s-4.8.3-20140911.so.1
[... cut ...]
    fmt=fmt@entry=0x7f48d392c8c8 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/unix/sysv/linux/libc_fatal.c:196
#3  0x00007f48d382b1fd in malloc_printerr (ptr=<optimized out>, 
    str=0x7f48d3929fd4 "free(): invalid pointer", action=3) at malloc.c:4972
#4  _int_free (av=0x7f48d3b68760 <main_arena>, p=<optimized out>, have_lock=0)
    at malloc.c:3804
#5  0x000000000040ab62 in free_options (o=0x1e63030) at options.c:245
#6  0x0000000000408e33 in main (argc=4, argv=0x7fff46d03088) at ttwatch.c:2666
```

ryanbinns, 2015-04-23:
The latest version has that code in it. It automatically adds it when the watch is opened if it's not there. I also add a command-line option to delete files (`--delete <file id>`) so you can delete the file and test to make sure it is re-added if you want to.

closed 2015-04-23 by ryanbinns