ryanbinns/ttwatch #24: "Unable to read configured formats"

remimeier, 2015-05-03:
There seem to be NULL bytes in the preferences-XML that causes the file.find()s in ttwatch_enumerate_offline_formats to stop short and not find the "</exporters>".
In particular, my preferences file looks like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<preferences version="1" modified="Sun May 3 22:23:41 2015">
    <ephemerisModified>0</ephemerisModified>
    <SyncTimeToPC>1</SyncTimeToPC>
    <SendAnonymousData>0</SendAnonymousData>
    <WatchWindowMinimized>0</WatchWindowMinimized>
    <watchName>My-Watch</watchName>
    <ConfigURL>https://mysports.tomtom.com/service/config/config.json</ConfigURL>
    <exporters>
        <offline>
            <export id="TCX" autoOpen="0"/>
            <export id="FIT" autoOpen="0"/>
            <export id="GPX" autoOpen="0"/>
        </offline>
        <online>
            <export id="MySports" autoOpen="1"/>
            <MySportsAuthToken>XXX-some-NULL-byte</MySportsAuthToken>
            <MySportsTokenSecret>XXX-some-NULL-byte</MySportsTokenSecret>
        </online>
    </exporters>
</preferences>
```
If you need any more information, please let me know. And thanks for making my watch useful! :)

ryanbinns, 2015-05-04:
NULL bytes? Bizarre. The XML standard explicitly prohibits NULL bytes in XML files. That is going to make it much harder. I'll have a think about it. The internal processing uses std::string, so that would have to change.

remimeier, 2015-05-04:
My "fix" was replacing all NULL bytes with "?" after loading the file. Since it's not a proper fix, I didn't bother to attach a patch. But it seems to be the cause.

I don't know C++ very well, but it seems that std::string supports NULL characters in the string if you use the right constructor ```string (const char* s, size_t n);```. Maybe it's easy to fix after all?

ryanbinns, 2015-05-05:
You're right, I forgot about that. I've uploaded a patch that should fix this. Let me know if it works now.

Raemi, 2015-05-05:
Thank you! Seems to work fine now.

closed 2015-05-05 by Raemi

adnz, 2015-05-11:
The file had become corrupted on my watch. I manually wrote a correct file.
FYI The file returned was
./ttwatch/ttwatch --read=0x00f20000
```
<?xml version="1.0" encoding="UTF-8"?>
<preferences version="1" modified="Wed 6. May 18:33:33 2015">
    <ephemerisModified>0</ephemerisModified>
    <SyncTimeToPC>1</SyncTimeToPC>
    <SendAnonymousData>0</SendAnonymousData>
    <WatchWindowMinimized>1</WatchWindowMinimized>
    <watchName>Tom</watchName>
    <ConfigURL>https://mysports.tomtom.com/service/config/config.json</ConfigURL>
    <exporters>
        <online>
            <export id="MySports" autoOpen="1"/>
            <MySportsAuthToken>,&quot;
                                       WV!S
VZcQp/}
|wQ0|),
          wcpV</MySportsAuthToken>
qVaQ-</MySportsTokenSecret>ecret>{&quotwP3y~xX{t	2}S*[v
            <MySportsUserId></MySportsUserId>
        </online>
    <offline><export id="gpx" auoOpen="0"/></offline></exporters>
</preferences>
```