ryanbinns/ttwatch #25: GPSQuickFix data not updating

adnz, 2015-05-11:
Using the latest master code and watch firmware 1.8.35
It looks like the update is working OK
Downloading GPSQuickFix data file...
Writing file to watch...
GPSQuickFix data updated

But the watch shows the download icon and it does take several minutes to get  a fix so the data has not been applied.

Reading the file (0x00010100) from the watch and comparing it with the file from gpsquickfix.services.tomtom.com/fitness/sifgps.f2p3enc.ee shows it is being written to the watch correctly.

Time on the watch is correct.

Any thing else I can try?

Thanks for a great piece of code

Ad

ryanbinns, 2015-05-11:
I noticed that when I went for a run this afternoon too. Maybe something changed in the last firmware update. I'll look into it in the next few days when I get some time.

Cheers,
Ryan

ryanbinns, 2015-05-11:
Easy fix. Somehow in the last big update, the code to reset the GPS processor following the GPSQuickFix data being updated was lost. I have added it again, and it's working again.

closed 2015-05-11 by ryanbinns

adnz, 2015-05-13:
Yes all good. Thanks for the quick response