ryanbinns/ttwatch #26: Creating new races?

dbrb2, 2015-06-06:
My watch has somehow managed to end up with no pre-stored races - so "list-races" returns lots of cycling etc predefinced races, but none prefixed "r"

Trying to create one gives me: 

ttwatch --update-race "r1,10KM 40MIN,40:00,10000,10"
Unable to update race

Which makes sense I guess, since it is trying to update a race that doesn't exist...

Is there any way to create one?

ryanbinns, 2015-06-07:
It could be tricky. Each race is stored in a separate file on the watch, but there are some fields inside the file that I haven't managed to decipher yet. I update the files by just changing the parts I can understand, but I wouldn't know what to put in a new file for the parts I don't understand, because it seems to change between files. I could try putting in all zeros, or find what the most common values are, but I couldn't guarantee it would work, or that the watch would still function properly with such invalid data.

I'll have a look when I get a chance.

Ryan

dbrb2, 2015-06-07:
I'll try to find a windows machine to connect it to to see if the official app has any luck. If not, I wonder if there is a way to factory reset these units...

ryanbinns, 2015-06-08:
Yes, you can factory reset the watch using the Windows software. I haven't worked out exactly what it does so I can replicate it, mostly because I don't really want to factory reset my watch.

closed 2015-09-14 by dbrb2