ryanbinns/ttwatch #29: Firmware version not supported

weechatter, 2015-07-11:
with latest Firmware Version: 1.8.42

ttwatch --settings

weechatter, 2015-07-14:
here is the patch-file:

+++ ttwatch/ttwatch.c	2015-07-14 14:35:35.272866763 +0200
@@ -46,9 +46,10 @@
     { 0x00010819, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
     { 0x00010822, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
     { 0x00010823, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
+    { 0x0001082A, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
 };
 
-#define MANIFEST_DEFINITION_COUNT (3)
+#define MANIFEST_DEFINITION_COUNT (4)
 
 /*************************************************************************************************/

ryanbinns, 2015-07-16:
Sorry for the delay, and thanks for the patch. Have you tested the settings to make sure they work with the new firmware version? I try to check a substantial number of them before updating the code, just to make sure. It's a bit time consuming, but I don't want to start randomly changing settings on people's watches because the firmware version change has changed the setting locations. I don't know how the watch would respond to invalid values being put in some locations.

I've been busy (and sick) for the last week or so, and haven't had a chance to check this.

weechatter, 2015-07-16:
Hello ryanbinns,
i tested with following options (read/write), without any problems.

ttwatch --settings
ttwatch --setting

weechatter, 2015-07-18:
btw did TomTom contact you Ryan? I asked them to do that to give support for the Linux version :-)

ryanbinns, 2015-07-20:
No, they haven't contacted me recently. I spoke to a couple of people around the beginning of the year and they provided the specifications for the ttbin file format, but I haven't got anything else from them. I don't expect to, since they are working on their own Linux software.

ryanbinns, 2015-07-20:
I have patched the source to include your fix here. I made a small change to the definition of the number of manifest definitions so it calculates it automatically.

Thanks for the feedback,
Ryan

closed 2015-07-20 by ryanbinns

DirkBri, 2015-11-21:
Hi Ryan, 

I add 

{ 0x0001082e, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },

Is this a good idea?

Thank for you work - it's a great tool!
Dirk

dlenski, 2015-11-22:
@DirkBri, I added this as PR #52

ryanbinns, 2015-11-23:
Thanks Dan. I have integrated it.