ryanbinns/ttwatch #30: Compilation error on armhf (raspberry pi2)

weechatter, 2015-07-21:
Compiling libttwatch.o...                                                                                                                         libttwatch.cpp: In function ‘int ttwatch_open_file(TTWATCH*, uint32_t, int, TTWATCH_FILE**)’:
libttwatch.cpp:435:54: warning: missing braces around initializer for ‘uint32_t [2] {aka unsigned int [2]}’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_close_file(TTWATCH_FILE*)’:
libttwatch.cpp:459:54: warning: missing braces around initializer for ‘uint32_t [2] {aka unsigned int [2]}’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_delete_file(TTWATCH*, uint32_t)’:
libttwatch.cpp:476:54: warning: missing braces around initializer for ‘uint32_t [2] {aka unsigned int [2]}’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_read_file_data(TTWATCH_FILE*, void*, uint32_t)’:
libttwatch.cpp:507:47: warning: missing braces around initializer for ‘uint8_t [50] {aka unsigned char [50]}’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_write_file_data(TTWATCH_FILE*, const void*, uint32_t)’:
libttwatch.cpp:526:65: warning: missing braces around initializer for ‘uint8_t [54] {aka unsigned char [54]}’ [-Wmissing-braces]
libttwatch.cpp:527:48: warning: missing braces around initializer for ‘uint32_t [3] {aka unsigned int [3]}’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_find_first_file(TTWATCH*, uint32_t*, uint32_t*)’:
libttwatch.cpp:543:39: warning: missing braces around initializer for ‘uint32_t [2] {aka unsigned int [2]}’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_enumerate_files(TTWATCH*, uint32_t, TTWATCH_FILE_ENUMERATOR, void*)’:
libttwatch.cpp:589:5: error: invalid use of qualified-name ‘::iterator’
libttwatch.cpp:589:5: error: expected ‘;’ before ‘it’
libttwatch.cpp:589:5: error: ‘it’ was not declared in this scope
libttwatch.cpp:589:5: error: expected ‘)’ before ‘;’ token
libttwatch.cpp:589:5: error: ‘it’ was not declared in this scope
libttwatch.cpp:589:5: error: expected ‘;’ before ‘)’ token
libttwatch.cpp: In function ‘int ttwatch_reset_gps_processor(TTWATCH*)’:
libttwatch.cpp:675:38: warning: missing braces around initializer for ‘char [60]’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_get_watch_time(TTWATCH*, time_t*)’:
libttwatch.cpp:700:46: warning: missing braces around initializer for ‘uint32_t [4] {aka unsigned int [4]}’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_get_firmware_version(TTWATCH*, uint32_t*)’:
libttwatch.cpp:749:45: warning: missing braces around initializer for ‘char [60]’ [-Wmissing-braces]
libttwatch.cpp: In function ‘int ttwatch_clear_data(TTWATCH*)’:
libttwatch.cpp:823:5: error: invalid use of qualified-name ‘::iterator’
libttwatch.cpp:823:5: error: expected ‘;’ before ‘it’
libttwatch.cpp:823:5: error: ‘it’ was not declared in this scope
libttwatch.cpp:823:5: error: expected ‘)’ before ‘;’ token
libttwatch.cpp:823:5: error: ‘it’ was not declared in this scope
libttwatch.cpp:823:5: error: expected ‘;’ before ‘)’ token
libttwatch.cpp:1490:1: error: expected ‘}’ at end of input
libttwatch.cpp: At global scope:
libttwatch.cpp:1490:1: error: expected ‘}’ at end of input
libttwatch.cpp: In function ‘int ttwatch_clear_data(TTWATCH*)’:
libttwatch.cpp:1490:1: warning: control reaches end of non-void function [-Wreturn-type]
Makefile:23: recipe for target 'libttwatch.o' failed
make[1]: *** [libttwatch.o] Error 1
make[1]: Leaving directory '/home/pi/git/ttwatch/libttwatch'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2

ryanbinns, 2015-07-22:
What version of GCC are you using? It appears that your version does not support some valid uses of `decltype`. I don't know exactly when support for this was added, but 4.5 and earlier definitely don't work. I'm using 4.8 and it works fine. I believe it was scheduled to be fixed in 4.6, but I don't know if it was.

weechatter, 2015-07-22:
gcc version 4.6.3 (Debian 4.6.3-14+rpi1)

ryanbinns, 2015-07-23:
Can you try with a newer version of GCC? It's probably that 4.6.3 doesn't support `decltype` fully; it's unlikely that it's from compiling for raspberry pi.

weechatter, 2015-07-23:
Ok Ryan you can close issue. i installed gcc version 4.9.2 (Raspbian 4.9.2-10) and now i can compile ttwatch. thanks for the hint

closed 2015-07-24 by ryanbinns