ryanbinns/ttwatch #34: Demo Mode

dbrb2, 2015-08-01:
Any ideas why, every time I put the watch into the USB cradle, it reverts back to "Demo Mode"...? Seems to be a quirk with latest f/w. May be nothing to do with your code...

ryanbinns, 2015-08-04:
Hmmmm... strange. I haven't heard of it doing this, and I've got the latest firmware. Are you running the ttwatch software as a daemon, or are you running it manually from the command line?

If you're running the software as a daemon, try killing it then docking the watch and seeing if it still reverts to demo mode.

You could also try using the Windows software and seeing what happens, if you have access to a Windows machine you can try it on.

weechatter, 2015-08-09:
I am using the latest firmware and can not confirm this problem

dbrb2, 2015-08-09:
I think I know what is happening: 

The Demo mode seems to be enabled by the acts of docking the watch, independant of whether the ttwatch software is running or not. 

However, the firmware has been updated, using ttwatch, since the unti was last connected to a windows machine running the tomtom software

My guess is that for some reason TomTom have set their watch to go into "demo" mode when it is docued if it has never been paired with official windows software since the latest firmware update, on the basis that if it hasn't, it is likely still on display in a shop.

Without a windows machine I can't confirm this, and if true it seems a rather strange decision by TomTom devs - but I will try to get hold of a windows machine at work to test the theory

ryanbinns, 2015-08-10:
I'm not sure that's the problem. I haven't used the Windows software for well over 6 months, and the firmware's been updating using the ttwatch software ever since. The watch has never reverted to demo mode.

I don't know what could be the problem though. It sounds like a problem with the watch. Maybe a factory reset using the Windows software would fix it.

closed 2015-09-14 by dbrb2