ryanbinns/ttwatch #35: `ttwatch` doesn't support activity transfer over Bluetooth LE

dlenski, 2015-08-08:
Okay, this is not exactly an "issue" ... however I have reverse-engineered the TomTom devices' BLE protocol and have a working beta over at [`dlenski/ttblue`](https://github.com/dlenski/ttblue).

I would love to get more eyeballs on this code (and the [BLE protocol documentation I'm writing](https://github.com/dlenski/ttblue/blob/master/tt_bluetooth.md)), and perhaps someday it can be integrated into `ttwatch` if you're so inclined!

/cc @onedayfishsale and @oleg-umnik

ryanbinns, 2015-08-10:
Interesting! I haven't even looked at the BLE protocol since it doesn't really seem useful to me - it's really, really slow. I don't have a bluetooth adapter for my linux machine anyway.

I'll have a look when I get some time, but it may not be for a long time.

dlenski, 2015-08-10:
You're right that it is really, really slow! But this is mostly because the device firmware is *too stupid to ask for a faster connection to the host*, which it can actually support.

It is actually possible to increase the transfer speed to ~1800 B/s (around 100s to transfer a 1 hour activity) by forcing a shorter connection interval, and I've been trying to figure out the least-intrusive way to do this: http://thread.gmane.org/gmane.linux.bluez.kernel/63778

Although the actual packet structure for the BLE interface is quite different from the USB interface, the device basically presents the same interface (identically-numbered files which can be read/written/deleted), and I think it should be possible to replicate almost all of the USB functionality in this way—not sure about firmware updates though :-)

ryanbinns, 2015-08-10:
Yes, I read your post and your documentation. I suspect it doesn't request a faster speed because it uses more power, although surely that would be offset by the extended time it would be transmitting for. Maybe not, though.

I'm wondering if it would be useful to integrate your BLE protocol into `libttwatch` so that the same `ttwatch` program can use both interfaces. I have to admit, however, that I don't really see a lot of point in using the BLE protocol, as the watch has to be docked to recharge anyway.

dlenski, 2015-08-10:
Yeah, integration into `libttwatch` is what I was thinking about too. I'll give it a crack at some point.

Personally, I like the BLE uploading because it's a lot less "fiddly" when I'm on the go (although at this point I've spent ~100 hours "fiddling" with it so I guess it's a moot point!).

I believe that the TT devices don't request a faster speed from the host because they were only tested with iOS and Android which default to a faster connection speed. Also, from the history of [updates and poor reviews for the official TomTom app](//play.google.com/store/apps/details?id=com.tomtom.mysports) I get the impression that they don't really have a good handle on making the BLE work reliably themselves...