ryanbinns/ttwatch #39: Fails to build since commit 064fd09d5a

McNoggins, 2015-08-25:
I get build errors since commit 064fd09d5a

make -C ttbin
make[1]: Entering directory '/home/denis/software/ttwatch-git/src/ttwatch-git/ttbin'
Compiling export_tcx.o...
export_tcx.c: In function ‘export_tcx’:
export_tcx.c:112:32: error: ‘smooth_cadence_count’ undeclared (first use in this function)
                     float dt = smooth_cadence_count == 0 ? 1.0f : record->treadmill.timestamp - smooth_cadence_lt;
                                ^
export_tcx.c:112:32: note: each undeclared identifier is reported only once for each function it appears in
export_tcx.c:112:97: error: ‘smooth_cadence_lt’ undeclared (first use in this function)
                     float dt = smooth_cadence_count == 0 ? 1.0f : record->treadmill.timestamp - smooth_cadence_lt;
                                                                                                 ^
export_tcx.c:116:21: error: ‘smooth_cadence’ undeclared (first use in this function)
                     smooth_cadence += alpha * (coarse_cadence - smooth_cadence);
                     ^
Makefile:16: recipe for target 'export_tcx.o' failed
make[1]: *** [export_tcx.o] Error 1
make[1]: Leaving directory '/home/denis/software/ttwatch-git/src/ttwatch-git/ttbin'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2

ryanbinns, 2015-08-25:
Fixed... Thanks for letting us know.

closed 2015-08-25 by ryanbinns

dlenski, 2015-08-26:
Sorry 'bout that :-(. I gotta make sure that I'm not building off of partially-merged code which won't actually build as committed...

ryanbinns, 2015-08-26:
No worries. It happens sometimes. I'll hopefully get round to getting some interval data for testing on Friday. I'll go for a run with exaggerated step cadence changes between intervals, so we can test some alternative smoothing methods.