ryanbinns/ttwatch #42: Mysports account info

dbrb2, 2015-09-14:
Reading through the ttblue documentsation: 

https://github.com/dlenski/ttblue/blob/master/tt_bluetooth.md

It is mentioned that the reason the android app will, on occasion, fail to download activities from a watch is that although it does not need an  internet connection, it does check the contents of the file 0x00f20000

It would be very nice to be able to set this from linux, without having to a) use the windows app and b) have a mysports account

From the ttblue docs:
"
the official apps won't function without being logged into it [Mysports]. Making this even more tedious, the mobile app frequently gets confused about the synced account, but only the desktop app can write the small XML file on the watch (0x00f20000) which stores account information.
"

dlenski, 2015-09-14:
If you look at `<exporters>` tag in the XML text of file `0x00f20000` after linking to a MySports account, you'll see some non-ASCII mess, which by my educated guess is an obfuscated [OAuth](https://en.wikipedia.org/wiki/OAuth) token enabling TomTom's apps to authenticate with the web app.

There is no public documentation for how to generate OAuth tokens for the MySports web app. So it's not possible for `ttwatch` (or `ttblue`) to set up the MySports account by itself, at least not without documentation or reverse-engineering.

In contrast, other fitness-tracking sites provide documentation and an API for generating their authentication tokens (e.g. Strava: http://strava.github.io/api/v3/oauth).

Unforunately, as we were discussing [here](https://github.com/dlenski/ttblue/issues/3#issuecomment-139410473), the "official" TomTom apps don't work unless linked to a MySports account, even though there is no technical reason why this is necessary.

dbrb2, 2015-09-14:
That makes sense. However, I have noticed that my watch will complain about
not being registered in the absence of an internet connection. I wonder
then whether it actually needs valid login details to download data...or
just some details that look valid, but won't actually work...

dlenski, 2015-09-14:
>I wonder then whether it actually needs valid login details to download data...or just some details that look valid, but won't actually work...

@dbrb2, why don't you try pushing a preferences.xml to your watch with a made-up/random MySports auth token?

Presumably this will answer the question.

dbrb2, 2015-09-14:
Happy to - will need to work out how first :-) Think I need to read the
ttwatch and ttblue docs in a bit more detail :-)

ryanbinns, 2015-09-15:
The Windows and Android apps do need an internet connection. There are a few reasons. The elevation data is downloaded from TomTom's web server, as are the firmware (although this isn't used by the Android app) and GPSQuickFix data. The URLs of the elevation, firmware and GPSQuickFix data (among other things) are stored in a settings JSON file that is also downloaded from their website, and the URL of THIS file is stored in the preferences file in the watch. Basically, this means that TomTom can reorganise their website and only have to keep ONE file in the same place - the one that's pointed to by the URL stored in the preferences file in the watch. Everything else is determined when the settings file is downloaded.

The ttwatch program currently ignores this URL in the preferences file and the URLs are hardcoded in the source code. I do intend to fix this at some point.

The latest version of the ttwatch software will create the preferences XML file if there isn't one, although it can't create MySports login details. It may be possible to at some point create the login details given an existing MySports account, as it uses the MapMyFitness API behind the scenes and that is documented.

dlenski, 2015-09-15:
@ryanbinns, is this the MapMyFitness API you're referring to? http://www.mapmyfitness.com/api/3.1/

I didn't realize that's what MySports was using.

ryanbinns, 2015-09-15:
Yep, that's the one. MySports is run by MapMyFitness, so (from memory) if you've got a MySports account, you can also log into MapMyFitness.

codefolder, 2016-07-27:
Just tried to log in like that from browser, and MapMyFitness would not accept my MySports credentials.

heliumir, 2016-07-27:
This doesn't seem to be the case anymore:
https://en.discussions.tomtom.com/runner-multi-sport-184/syncing-to-map-my-fitness-994680/index1.html#post1051004

ryanbinns, 2016-07-28:
Yes, it appears to have changed. I don't use MySports (and never have), so I hadn't noticed.

codefolder, 2016-07-28:
A bonus for the TomTom Spark (77) model owners would be to push Activity monitoring data, e.g. steps, etc., to the MySports website.

The c program does not appear to have code for it. (Sorry about the digression.)

(Back to the script.) Ideal is not to use MS Windows at all for the workflow (even when we speak of uploading a Run-activity)...

What would be needed to tweak the MySports authentication script?

codefolder, 2016-07-28:
Maybe an aside... Anyone has a script that would automatically upload to Strava?

dlenski, 2016-07-29:
@codefolder, I wrote a simple shell script that uses `ttbincnv` and `curl` to upload to Strava as part of [ttblue](http://github.com/dlenski/ttblue): [`ttbin2strava.sh`](http://github.com/dlenski/ttblue/blob/master/ttbin2strava.sh).

Strava has a well-documented public API for such things; unfortunately it requires generating per-app authentication codes. I set up a simple web server which can generate one for you at: http://stravacli-dlenski.rhcloud.com/

codefolder, 2016-07-31:
@dlenski, thank you very much, it is useful and simple to use!