ryanbinns/ttwatch #43: TAG 23h structure information

jps1x2, 2015-09-14:
Hi!

On 2014-12-24 i have opened following issue on FluffyKaon/TomTom-ttbin-file-format about new TAG found in my investigation (0x3Fh):

https://github.com/FluffyKaon/TomTom-ttbin-file-format/issues/4

Now i am trying to write a .GPX to .TTBIN converter for testing purposes in python but i don't know structure of TAG 23h and can't Reverse engineer it.

You told us that you received documentation from Tomtom about .TTBIN file structure. Please, could you help me with this TAG structure?

Thanks in advance!

jps1x2

ryanbinns, 2015-09-15:
Tag 23h is not documented in the information I got from TomTom. My reverse engineering indicates that it contains information about which GPS satellites are being used and a confidence level (or signal strength, or something similar) for each one.

If you're writing TTBIN files, you don't need to include those records, and it won't make any difference. I doubt they'd be used by TomTom's software anyway.

dlenski, 2015-09-15:
@jps1x2 , if you want to print out the contents of unknown tags like 0x23, [ttbindec](https://github.com/dlenski/ttbindec) will print out the raw binary contents.

(I also thought that it had something to do with satellite connections, since its value changes most rapidly when the connection to the satellites is poor or the watch drops the connection.)

ryanbinns, 2015-11-24:
If you're still interested in the structure of the 23h record (although it's not necessary to insert it in the file) this is the information I've been able to decode:
```c
struct TTBIN_GPS_SATELLITE_RECORD
{
    uint8_t  tag;             /* always 0x23 */
    uint32_t dop;             /* dilution of precision, seems to be in metres * 10^7 */
    uint8_t  _unk1;
    uint8_t  satellite[4];    /* up to 4 satellite numbers, satellites currently being used */
    uint8_t  signal_strength; /* signal strength for each satellite listed above, not sure what units */
    uint8_t  _unk2[4];
    uint8_t  _unk3[2];
};
```
I'm not sure what the unknown fields are, but their values definitely change, seemingly randomly, so it's probably not something you could just fake up. The ttwatch software doesn't use this record at all. I'm not sure if TomTom's software does, probably only for determining if there were any GPS dropouts.

closed 2015-11-24 by ryanbinns

uded, 2015-11-24:
According to my notes and observation `0x23` record should be 20 bytes long. If I am not mistaken, based on the structure provided, you consider it to be 17 bytes long. 

I am using Spark with file version 8. Maybe here we do have a problem or something? Or am I missing something obvious...

ryanbinns, 2015-11-24:
Just a typo. signal_strength was meant to be a 4-element array as well. I fixed it.

uded, 2015-11-24:
OK, then it's fine.

Signal strength seems to be SNR value (signal-to-noise-ratio) which is usually provided by GPS receiver. The range of the value will be between 0 and 99, but practically it's between 1 and 55-60. It seems like those values are similar to what I got on my test GPS receiver...

Example:
`dop=16.0, snr=[26, 22, 17, 16]` - bad GPS reception, none or one bar on the watch
`dop=15.0, snr=[23, 22, 18, 17]` - bad GPS reception, none or one bar on the watch
`dop= 3.0, snr=[35, 33, 33, 32]` - good GPS signal, all bars on the watch
`dop= 5.0, snr=[33, 32, 32, 30]` - good GPS singal, all bars on the watch

`_unk3` is really odd. Normally it has the same value all over my files `[4, 2]`. Once or twice it was `[4, 0]`...

jps1x2, 2015-11-24:
Thank you very much for your help!

ryanbinns, 2015-11-25:
@uded If the values are similar to what you get on a GPS receiver, do you think we can assume that the values are SNR in dB then?

Yeah, _unk3 is almost always [4, 2] for me as well. No idea what it means.

_unk2 is always 4 bytes of 0 or 1 though, and which ones are which is always changing. It's obviously a 4-element boolean array associated with the satellites, but I have no idea what it represents.

_unk1 seems to be a satellite count, but I'd need to see a few more files with lost GPS signal and other variations to be sure. The coverage around my house is too good; I never get less than full bars. I've seen this number vary between 4 and 9, but usually sits between 4 and 6.

uded, 2015-11-25:
Now I can fully confirm - the values are SNR and they're almost identical to what I have recorder on my GPS receiver. So it seems correct.

_unk3 has two values for me `[4,2]` and `[4,0]`. I cannot find any logical explanation for them.

_unk2 - I am not certain. I see no correspondence in GPS navigation (boolean or no boolean) to be used here. But, obviously, I am missing something ;-)

_unk1 - Here I can help having access to some specialized equipment and controlled environment. Let me do some checking and I will try to figure this out...

uded, 2015-11-25:
Actually, no - `_unk1` does not reflect number of satellites tracked. It seems to be 5 when I had almost no signal at all with dop around 20 meters, 4 when it was great and full coverage with 3 meters DOP and it turned to 3 still having 3 meters DOP with full coverage.

ryanbinns, 2015-11-25:
It could be the number of visible satellites, rather than the number being tracked. If satellites are too close to the horizon they can't be used for tracking, even though they are technically visible. Although, 3 satellites aren't enough for a full 3D position fix so if it was showing 3 with good coverage then it's not that.

uded, 2015-11-25:
Well, not exactly. For proper tracking you need 4 satellites, hence the 4 satellites in the array. Normally you see much more, though. So I doubt that...