ryanbinns/ttwatch #45: Laps are missing in TCX export (Treadmil)

mipapo, 2015-09-24:
I'm happy to see that @ryanbinns added the treadmil export.

I had a look at https://github.com/dlenski/ttbindec
It's based on your repository and it seems the Laps can be read. Would be great if the rounds would be available in the tcx files too.

If I find time I will have a deeper look into the source code and try it by myself.

````
00013178 25 FILE_HEART_RATE_RECORD(heart_rate=134, _reserved=255, timestamp=1442857141)
0001317f 32 FILE_TREADMILL_RECORD(timestamp=1442857142, distance=7902.68212891, calories=645, steps=8021, step_length=0)
00013190 25 FILE_HEART_RATE_RECORD(heart_rate=134, _reserved=255, timestamp=1442857142)
00013197 3b FILE_INTERVAL_FINISH_RECORD(type=2, total_time=3240, total_distance=7903.86474609, total_calories=645)
000131a3 3a FILE_INTERVAL_START_RECORD(type=3)
000131a5 32 FILE_TREADMILL_RECORD(timestamp=1442857143, distance=7903.86474609, calories=645, steps=8023, step_length=0)
000131b6 25 FILE_HEART_RATE_RECORD(heart_rate=134, _reserved=255, timestamp=1442857143)
000131bd 32 FILE_TREADMILL_RECORD(timestamp=1442857144, distance=7905.04931641, calories=645, steps=8026, step_length=0)
000131ce 25 FILE_HEART_RATE_RECORD(heart_rate=134, _reserved=255, timestamp=1442857144)
````

dlenski, 2015-09-25:
Hi @mipapo, I actually wrote the treadmill-specific TCX export code (see #38).

I don't understand what you mean about "Laps" or "rounds" being missing...

* Laps (either automatic time/distance laps, or manual laps) should **not** be missing in the TCX output.
* I don't understand what you are highlighting in the ttbin file.

mipapo, 2015-09-25:
Hi @dlenski Please have a look at http://forum.runalyze.de/viewtopic.php?f=4&t=248 (Download the attachment of the first post). There is a ttbin file in there for which are no laps exported.
There is only the starting lap and end lap.

Rounds=Laps=Intervals
I just wanted to higlight this part:
```
00013197 3b FILE_INTERVAL_FINISH_RECORD(type=2, total_time=3240, total_distance=7903.86474609, total_calories=645)
000131a3 3a FILE_INTERVAL_START_RECORD(type=3)
```
These parts are missing.

dlenski, 2015-09-25:
For the purposes of the TomTom watches, a "Lap" and an "Interval" are **not the same thing** (there are two different ways of setting them up on the watch, and they're stored differently in the TTBIN file.

The TCX file format has no way of storing intervals *per se*, but it can store laps. So it seems like the solution you want is to divide up the TCX files into `<Lap>` tags according to the INTERVAL tags.

If that's what you want, it should be a fairly straightforward modification to [`export_tcx()`](//github.com/ryanbinns/ttwatch/blob/master/ttbin/export_tcx.c#L10) to add an option to split the track at the start/end of intervals.

```
000000c5 39 FILE_INTERVAL_SETUP_RECORD(warm_type=2, warm=0, work_type=0, work=240, rest_type=0, rest=60, cool_type=2, cool=0, sets=12)
...
0000175a 32 FILE_TREADMILL_RECORD(timestamp=1442854142, distance=616.064208984, calories=51, steps=623, step_length=0)
0000176b 25 FILE_HEART_RATE_RECORD(heart_rate=143, _reserved=255, timestamp=1442854142)
00001772 3b FILE_INTERVAL_FINISH_RECORD(type=2, total_time=240, total_distance=618.713928223, total_calories=51)

... end </Lap> tag here, and restart new <Lap> tag ...

0000177e 3a FILE_INTERVAL_START_RECORD(type=3)
00001780 32 FILE_TREADMILL_RECORD(timestamp=1442854143, distance=618.713928223, calories=51, steps=625, step_length=0)
00001791 25 FILE_HEART_RATE_RECORD(heart_rate=143, _reserved=255, timestamp=1442854143)
00001798 32 FILE_TREADMILL_RECORD(timestamp=1442854144, distance=621.302307129, calories=51, steps=628, step_length=0)
```

mipapo, 2015-09-25:
Good to know that Lap/Interval are not the same, but for the TCX Laps/Intervals are - just like you said - laps, which are starting with the <Lap> and ending with </Lap>.

````
        case TAG_INTERVAL_START:
                  fputs("<Lap>\r\n", file);
                break;  
        case TAG_INTERVAL_FINISH:
                fputs("</Lap>\r\n", file);
                break;
```
For the TAG_INTERVAL_START must be a check whether it's the first Lap or not, because there would be two Starting Laps in the beginning of the export.
So if `lap_state` = 0 there should be no new lap created? (The several states of `lap_state` are not that clear to me in the moment)

ryanbinns, 2015-09-29:
Yeah, `lap_state` is a little confusing. Basically, the start/end laps are only written to a file when a GPS or treadmill record is encountered in the TTBIN file.
- If `lap_state == 0`, then the next GPS/treadmill record is written to the TCX file with no extra lap processing.
- If `lap_state == 1`, then a lap is begun prior to writing out the next GPS/treadmill record, and `lap_state` is set to 0 ready for normal record processing.
- If `lap_state == 2`, then a lap is ended after the next GPS/treadmill record is written and `lap_state` is set to 1 ready to begin the next lap when another GPS/treadmill record is received.

It's done this way for a couple of reasons:
- To make sure that there are no GPS/treadmill records outside of a lap.
- To make sure that there is no empty lap at the end of the file - the next lap is only started if there is more GPS/treadmill data to follow.

I should really have used an `enum`. One of these days I'll get around to changing it...

If you simply write out the start and end tags for a lap as you posted above, you'll break the XML as the data is actually inside a `<Track>` record inside the `<Lap>` record. You'll also lose the summary information that is inserted in the TCX file at the end of each lap. I'd recommend using `TAG_INTERVAL_START` and `TAG_INTERVAL_FINISH` to set `lap_state` appropriately and let the existing code work to write the file. You'll have to experiment to see if this causes lost information though. It shouldn't, but will need checking.

closed 2016-05-18 by ryanbinns