ryanbinns/ttwatch #47: ttbincnv renames file (not just extension), stops ttbin2mysports.sh from working

mjsearle, 2015-10-02:
Had to modify line 62 in ttbin2mysports.sh to this
( cd "${dir}" && cat "${file}.ttbin" | ttbincnv -t >"${file}.tcx" )

as default behaviour of ttbincnv is to rename file to {activity type}_{code}.tcx

Otherwise good work. All working for me on Ubuntu 15.04

ryanbinns, 2015-10-02:
Thanks for the feedback. I don't use MySports and I didn't write that script, so I've never noticed. It should work if the ttbin files haven't been renamed though.

I'll make the change, unless you'd like to submit a pull request and I'll merge it.

mjsearle, 2015-10-03:
Happy for you to make the change. Thanks.  I've done plently of scripting
but are new to Git. The script worked fine once I made that change. The
workouts all appeared on my MapMyFittness dashboard after the script
uploaded them.

ryanbinns, 2015-10-04:
I've committed a fix for this. Can you verify it when you have a chance please? My watch has died and is in for repair :-(

mjsearle, 2015-10-05:
Made a fresh clone. Worked fine. Thanks.

closed 2015-10-05 by ryanbinns