ryanbinns/ttwatch #48: "Unable to read configured formats" on --get-activities

devendermishrajio, 2015-10-25:
My /etc/ttwatch.conf is as follows
GetActivites=true

~/.ttwatch
ActivityStore=/home/devender/ttwatch

When I run

ttwatch --get-activities

I get the following error:

"Unable to read configured formats"

I am unable to download the activities.

ryanbinns, 2015-10-26:
That error won't stop the activities downloading - it just means that the program was unable to get a list of offline file formats from the watch. It should still download the activity file; it just won't convert the file to any other file format. If you want to stop the error appearing, just add a file format to be converted, by running (for example)

```
ttwatch --set-formats csv
```
This will make a CSV file for every activity you download, but at least the error will disappear.

Are there actually any activities on the watch that have not been downloaded before? Are there any other errors being shown? Have you checked to see if the ttbin files are appearing in your activity store directory?

The `--get-activities` function is capable of handling corrupt files, so even there are errors with the data it gets from the watch, it should still write the file to disk. It will print an error if it cannot read the data from the watch at all, or if it cannot write the data to file, or if the written data does not verify properly (it reads it back from the file to make sure it was written). It will never silently fail, and will only do nothing if there are no activities on the watch to be downloaded, such as if an activity has already been downloaded previously.

devendermishrajio, 2015-10-26:
Where will it store the activity files? I am unable to find the ttbin files.

ryanbinns, 2015-10-28:
It should be putting them in /home/devender/ttwatch/&lt;watchname&gt;/&lt;date&gt;

closed 2015-11-02 by devendermishrajio