ryanbinns/ttwatch #49: Support for Spark / Runner2

lgrangeia, 2015-11-02:
Hi,

I just bought a TomTom Spark, and this tool doesn't support it yet. Ryan, I can start to add support for the Spark if you agree. The protocol seems to be very similar.

Let me know if you have any plans to add support. If not I can start working on a patch.

ryanbinns, 2015-11-03:
Feel free to add support for the Spark. I would if I could, but I don't have one and can't afford one, so I won't be able to. There may not be too many differences (maybe mainly the music side of things, although that may be implemented as a mass storage device, I don't know).

uded, 2015-11-03:
The music upload is implemented, as far as I have checked, using mass storage device interface. Simple and clean. Uploading MP3s requires a playlist files and folders are uploaded in sequence with naming as such - no custom names for them.

I can help you out with additional tag and data fields (at least 3) introduced by Spark (already started working on that, but with my own [Java implementation](https://bitbucket.org/uded/ttbin4java/overview)...

gsoyez, 2015-12-22:
I just discovered the software and have a Runner 2 at hand.Which does not seem to be recognised (ttwatch --devices does not return anything). Do you have any hint regarding where to get started if one wants to update your code (simply updating the PRODUCT_ID in libttwatch.cp does not seem to get the work done)?

Thanks in advance!

mipapo, 2015-12-25:
Seems csv output is working (seems I have an swim test file without coordinates), but everything else not.

ryanbinns, 2015-12-27:
If you've confident using gdb, I would recommend single-stepping through `ttwatch_open_device` and seeing where it errors out. This will tell you where the problem is.

Also, make sure the user you're running as has permission to access to the USB device. See `README.md` for information on how to set this up. You can also run using `sudo` if you want to check quickly if this is the problem.

krys-g, 2015-12-29:
Hi all,

I am also very interested with Runner 2 support, and as mentioned earlier, the device is not recognized due to a different protocol.

The product ID for that watch is 0x7475 (so a simple #define no longer works), but that is not enough sadly. If I recall correctly (I was having a brief look last week), when calling `ttwatch_send_startup_message_group`, the following line fails:

    RETURN_ERROR(send_packet(watch, MSG_UNKNOWN_0D, 0, 0, 20, 0));

which is to say the very first call to `send_packet`. I also tried as root, so this should not be a permission problem.

I guess the next logical step is generating USB traces, which I can send to you if you want & have time to have a look.

Cheers, and thanks for ttwatch!
k.

matthiasdallner, 2015-12-29:
Hi all,

I also have a Runner2 and already analyzed some USB traces of the original TomTom software. It appears, that the Runner2 is using a fixed 256 byte packet size for usb communication (filled with 0s). Adjusting the packet size accordingly made ttwatch communicate with the watch. I have created a pull request containing these adjustments. Strangely, however, the product id for my Runner2 is 0x7477 and not 0x7475, as @krys-g mentioned.

krys-g, 2015-12-29:
Hi Matthias,

Interesting. Do you have a TomTom Spark or a Runner2? (I see your changes mention SPARK). I double-checked, and my Runner 2 is 0x7475. The serial name starts with: HO6395G.

In any case, I checked out your pull request and... it worked! At least to the extent that it detected my watch, and could list the races. I will test further.

Thanks a lot! :+1:

matthiasdallner, 2015-12-30:
I have a Runner2 Cardio (no Music). The serial starts with HN2395G. Which version do you have? 

According to the TomTom support forum the Spark and Runner2 models are using the same head units (just a different name and strap) and therefore should also have equal product IDs. I simply used MULTISPORT and SPARK instead of RUNNER and RUNNER2 for the #defines to make it easier to differentiate the two when reading the code.

Maybe the other Runner2/Spark owners could also post the product IDs together with the exact version (Cardio, Music). Then we can add further #defines, if necessary, and adjust the checks.

krys-g, 2015-12-30:
Sorry, I should have said that my watch is a Runner2 GPS Music (no cardio).

I tested a bit more, and some features do not work sadly.

* :white_check_mark: `--update-gps`: worked
* :white_circle: `--update-fw`: possibly ok (no update was available)
* :white_check_mark: `--get-time`: worked
* :red_circle: `--set-time`: did not work (time unchanged, no error message)
* :white_check_mark: `--list-races`: worked
* :red_circle: `--get-activities`: did not work (no file created, no error message)
* :red_circle: `--list-history`: return non-sensical values, as shown below.
```
    1: 2015/12/21 14:31:48, 1241513984s,     0.00m, 1917238192 calories
    2: 2015/12/23 15:53:15, -452984832s, 166154172912495855359845427658620928.00m, 189189272 calories
```

matthiasdallner, 2016-01-02:
Did the `--set-time` command fail to set the correct time or did you test it with an intentionally wrong system time and that time didn't get set? As far as I know the command simply sets the timezone offset to UTC time and it is set correctly on my watch if I change the system timezone. Changing only the system time before issuing the command hence doesn't affect the time on the watch.

Regarding the `--list-history` command, TomTom added 3 additional (yet unknown) bytes between the `second` and `duration` fields of the history entry. Therefore, changing `uint32_t _unk3[2];` to `uint8_t _unk3[11];` in the `TTWATCH_HISTORY_ENTRY` struct solves this problem.

krys-g, 2016-01-04:
Happy New Year!

It turns out you were spot on, doubly so!

1) I did not think of changing the TZ, but after doing so, `--set-time` did change the time on my watch.
2) Updating the data structure to add padding also solved the absurd values returned by `--list-history`.

Once again, I am very grateful for your help. :bow:

Do you plan to productize these changes in your repository? If not, I can probably do it. Otherwise, I can focus on the music upload part.

ryanbinns, 2016-01-06:
I have made the data padding change in the repository. Let me know if it works properly as I don't have a Spark to test against.

derickr, 2016-01-06:
It does not make a difference for me, I still get:

```
Running:
1: 2016/01/04 12:25:58,    1s, 2199023255552.00m, 1409286144 calories
2: 2016/01/04 12:58:58,    2s,     0.00m, 805306368 calories
3: 2016/01/06 12:21:37,    3s,    -0.00m, -1778384896 calories
```

It looks like the struct has changed slightly. I'll try to find out what it is.

matthiasdallner, 2016-01-06:
Thanks for taking care of the changes to the history entry. Unfortunately it doesn't work correctly yet. As you have placed all the data fields of each watch in one union, all members access the same memory.

I think we need something like this instead:
```
typedef struct __attribute__((packed))
{
    ...
    union
    {
        struct __attribute__((packed))
        {
           ...
        } multisport;
        struct __attribute__((packed))
        {
           ...
        } spark;
    }
} TTWATCH_HISTORY_ENTRY;
```

Furthermore, the `index + 1` arguments to the `write_log()` functions in ttwatch.c lines 1261 and 1268 need to be removed (the index is already contained in the first `write_log()` call in line 1257).

With these changes it works for the Spark models.

derickr, 2016-01-06:
Not quite @matthiasdallner , I still have an issue with duration - continuing to look:

```
Running:
1: 2016/01/04 12:25:58,    0s,  5145.58m, 1620 calories
2: 2016/01/04 12:58:58,    1s,  1011.35m,  304 calories
3: 2016/01/06 12:21:37,    2s,  6693.93m, 2198 calories
```

I've made PR #64 with your code suggestions.

matthiasdallner, 2016-01-06:
You need to remove the `index + 1` completely (not just the `+ 1`) in ttwatch.c, then it should work.

derickr, 2016-01-06:
Correct, will push that fix.

ryanbinns, 2016-01-06:
Ugh. I'm sure I tested it, but obviously not. Sorry about that, but thanks for the fix!

gsoyez, 2016-02-21:
Sorry for the slow reply. It's actually the first time since Christmas since I manage to grab hold of the watch (which is my mother's). 
Anyway, at first sight it seems to be working now (though I have only tested basic things like retreiving the activities)!
Thanks to all of you for the quick fix.

closed 2016-05-18 by ryanbinns