ryanbinns/ttwatch #50: Upload of pre-defined races

weechatter, 2015-11-07:
Hello ryan,

i want to create a race with: http://www.pacecreator.com/index.php

It would be fine to upload the created race to the watch

ryanbinns, 2015-11-07:
I'll have a look into this when I get some history data on my watch and see what information is actually stored on the watch. My watch died so I just got a new one (fortunately under warranty) and I haven't had a chance to get some data on it yet.

weechatter, 2015-11-07:
If you need some data from the watch let me know.

ryanbinns, 2015-11-10:
Ok, I went for a run this afternoon and looked at the data the watch generates for the "race myself" function. Essentially, this should be fairly easy to implement, although to make it simpler and easier, I think I'll implement the race generation completely, rather than loading a file generated by that website. Something like:
```bash
ttwatch --create-continuous-race r,"3km@4:00"
ttwatch --create-continuous-race t,"3km@240s"
ttwatch --create-continuous-race c,"12km/1:00:00"
ttwatch --create-continuous-race f,"2km@3:00,10km@4:00"
```
The first letter of each option value would be the activity type. The first two commands would create the same race (but for different activities), creating a race of 3km at a pace of 4min/km. The third would create a cycling race of 12km in 1hr. The fourth would create a freestyle race of 2km at a pace of 3min/km followed by 10km at 4min/km. How does that sound?

I can't think of a better option name at the moment, but if you have any ideas...

ryanbinns, 2015-11-11:
I have pushed an update that implements this. Let me know if it works for you. It's basically exactly as I specified above.

weechatter, 2015-11-11:
Wow! That sounds great. Is there a way  to send you an paypal donation, for a beer for example, ryan? :-)

ryanbinns, 2015-11-11:
Thanks for the thought, but no, that's ok. I don't drink anyway ;-)

closed 2016-01-05 by ryanbinns

weechatter, 2016-01-05:
Only one question Ryan. Is there any limitation in this function for <distance/time> / <distance>@<pace>
or is it possible to create a race with 21 or even 42 values?

ryanbinns, 2016-01-05:
No, there's no limitation. It will keep on extending the race for as long as there are values left in the string.

weechatter, 2016-01-06:
Thanks. So the only limitation is the watch itself and i guess only TomTom knows how many values are possible.