ryanbinns/ttwatch #53: Support for tag 0x41

uded, 2015-11-24:
It seems that this tag is connected to gym activity. Sorry, translating from Java here, so I hope it will be correct:

```c
struct TTBIN_GYM_RECORD
{
    uint8_t  tag;                  /* always 0x41 */
    uint32_t timestamp;            /* well, timestamp */
    uint8_t  _unk1[4];             /* not sure here, connected with movement somehow */
    uint16_t  calories;            /* calories burned since beginning of the exercise */
};
```

My Java implementation can be found [here](https://bitbucket.org/uded/ttbin4java/src/742d954318f21cfe2064b94a6db08c2e4df3b220/src/main/java/pl/radical/ttbin/records/GymActivityRecord.java?at=master&fileviewer=file-view-default)

This can be easily added here as a new tag...

ryanbinns, 2015-11-25:
Thanks. I'll have a look at the test data you've got in your repository to see if I can get anything meaningful out of the unknown field. I doubt it, but another pair of eyes is always good.

Is the TomTom software able to make any offline file types (csv, tcx etc...) for gym activities? If so, I can look at adding support. They might provide clues about that unknown field too.

uded, 2015-11-25:
OK, got it. It's number of _cycles_, whatever that is. This is how it's described in TomTom exports. And I've messed up order of fields, fixed above and below:

```c
struct TTBIN_GYM_RECORD
{
    uint8_t  tag;                  /* always 0x41 */
    uint32_t timestamp;            /* well, timestamp */
    uint16_t  calories;            /* calories burned since beginning of the exercise */
    uint32_t  cycles;             /* number of "cycles" since beginning of the exercise */
};
```

uded, 2015-11-25:
Would be happy to attack the rest of unknowns, though: `0x30`, `0x31`, `0x37`, `0x40`, `0x42`

ryanbinns, 2015-11-25:
Cool. I'll add that. I've just worked out `0x42`, race status (1st position or 2nd position). I've just checked it in. I haven't yet worked out the others.

Cycles generally means steps, strokes or cycles (!) depending on whether the activity it running, swimming or cycling respectively. Not sure how to interpret it for a gym activity, since I don't exactly know what the activity is intended to record. I presume it uses the accelerometer to measure how many times your arm changes direction when lifting weights or something.

uded, 2015-11-25:
`0x42` - OK, are you sure about that? It's kind of strange, I got both values all over the files in all types of activities :grey_question: 

For cycles - completely agreed. I have checked that against CSV file exported by TomTom MySports, so it's right on spot. Although CSV file seems to export diff values (1, 2...n) and not one value incremented over the time. Nevertheless, the value is correct.

Can we take the discussion to email or something? I have issues with some other fields, we can discuss and move the implementation forward...

ryanbinns, 2015-11-25:
I was sure until I looked at the file I recorded today. I went for a run with training set to 1km laps and I got `0x42`s every second. They all had the value 1 though. It matched perfectly with the race I ran against last week though, including the changes for in-front and behind, which is why I thought it was right.

I'd prefer to keep the discussion on here, so that other people can chip in with their observations if they have any. Feel free to raise another issue. We can close this one since I've added your structure and activity type to the code. The code doesn't yet do anything with them, but they're there.

uded, 2015-11-25:
Usually I do free runs, interval and zone trainings and those values do change all over the files. It's always 1 or 2 in my case as well, but I do not see the link between race and this entry.

OK, we can keep it here - but the subject of the ticket might get irrelevant during the discussion and I do not see the point of setting a new issue every time ;-)

closed 2015-11-25 by ryanbinns