ryanbinns/ttwatch #55: Support for tag 0x31 and 0x40

uded, 2015-11-25:
This one seems to be interesting. First one is 11 bytes while second is 12 bytes in length. I have no clue what those one are about. If anyone can provide me with an example file I will be happy to investigate...

ryanbinns, 2015-11-25:
`0x31` is cycling cadence. For some reason I missed it when writing the software. It has this structure:
```c
typedef struct
{
    uint32_t wheel_revolutions;
    uint16_t wheel_revolutions_time;
    uint16_t crank_revolutions;
    uint16_t crank_revolutions_time;
} CYCLING_CADENCE_RECORD;
```
I don't have any information for `0x40` yet.

uded, 2015-11-25:
Great! Added `0x31` in my implementation.

`0x40` seems to arrive with Spark, so it might be connected to one of the new modes...

uded, 2015-11-26:
`0x40` is connected to Indoor Cycling. No idea yet what it is really...

uded, 2015-12-09:
Any chance that one of You guys will have a file with proper indoor cycling tracking? I would love to figure this one out...

mschaaf, 2016-09-22:
Is there a chance that the cycling cadence will be exported to a gpx or tcx file?

mipapo, 2016-12-28:
@uded If you're still interested. In #108 is a indoor cycling file

closed 2017-01-09 by ryanbinns

uded, 2017-01-09:
Thanks, already covered that and will commit ASAP

PR #110 (2017-01-09): Implement exporting cycling cadence data from compatible bluetooth sensors
Tested to work with runner2 & Tomtom - 9UJ0.001.01 speed&cadence sensor
Both GPX and TCX files are recognized by strava.

No cadence points smoothing/filtering for the moment, it seems to work fine as-is.

Wheel speed does not work for me: ttbin files do not seem to include the wheel_size packet, maybe it needs to be added artificially (but out of the scope of this pull request)

Fixes #55