ryanbinns/ttwatch #56: Altitude in normal training file

uded, 2015-11-26:
Got a question - in all of my exported files the altitude information seems to be derived from somewhere. And it seems to be quite accurate. But I cannot find that value in any of my files. Any idea?

ryanbinns, 2015-11-26:
The altitude information is downloaded from TomTom's website. It is generated from a digital elevation model by sending them all the lat/long points. It's reasonably accurate, but not perfect. There are plenty of reports of wildly inaccurate information. Unfortunately, the watch doesn't store GPS altitude in the files.

uded, 2015-11-26:
Interesting, good to know. Thanks!

closed 2015-11-26 by uded

dlenski, 2015-11-27:
See the [`download_elevation_data()` function](https://github.com/ryanbinns/ttwatch/blob/d02ebd64a83705dde3797dfdd5ca2b401c7c2246/ttbin/ttbin.c#L933).

If you are planning to upload your GPS tracks to fitness tracking sites like Strava or Garmin Connect, they typically ignore the elevation data from the source file (since standard GPS receivers are not very accurate for elevation) and re-calculate it using a similar database to what TomTom uses, unless explicitly told that the GPS device includes a more reliable altimeter.

The biggest problem of these databases is that bridges and overpasses get reliably screwed up ... I actually looked for a better one but have not found a publicly accessible elevation database that includes bridges!