ryanbinns/ttwatch #57: Unable to set format and export activities

mbrouer, 2015-12-02:
I'm not sure if I'm doing something wrong but I'm not able to set the file format or export activities
Installed libusb from the sourceforge link and openssl and curl libs from Ubuntu's repos (15.10)

root@T450s:~# /usr/local/bin/ttwatch -a
Unable to read configured formats
Downloading GPSQuickFix data file...
Writing file to watch...
GPSQuickFix data updated
Current BLE firmware is already at latest version
Current firmware is already at latest version

root@T450s:~# /usr/local/bin/ttwatch --activity-store=/tmp/ --get-activities --set-formats=tcx
Unable to read configured formats
Unable to clear format list

root@T450s:~# /usr/local/bin/ttwatch --activity-store=/tmp/ --get-activities
Unable to read configured formats

root@T450s:~# /usr/local/bin/ttwatch --set-formats=tcx
Unable to clear format list

root@T450s:~# /usr/local/bin/ttwatch --list-history
Running:
1: 2015/12/02 07:54:11,  933s, 15820.30m,  193 calories

ryanbinns, 2015-12-02:
Have you ever connected the watch using the Windows software? If not, try using the `--initial-setup` command. That will create a default preferences file and you should be able to set a list of formats after that.

mbrouer, 2015-12-02:
Hi Ryan

Thanks for the reply :)
No, I have not had the watch connected to a Windows machine.  The
--initial-setup helped on the errors I was getting.

*sudo /usr/local/bin/ttwatch --get-formats        *
gpx tcx

Now I'm not getting any errors, but following command does not write any
files in /tmp (add the packet option)

*sudo /usr/local/bin/ttwatch --activity-store=/tmp/ --get-activities
 --packets*
09 02 00 0D
01 16 00 0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
09 02 01 0A
01 02 01 0A
09 02 02 22
01 03 02 22 00
09 02 03 22
01 03 03 22 00
09 02 04 20
01 06 04 20 E9 03 00 00
09 02 05 0A
01 02 05 0A
09 02 06 28
01 06 06 28 00 00 00 18
09 02 07 28
01 06 07 28 00 00 00 18
09 02 08 1F
01 06 08 1F 00 00 00 00
09 06 09 06 00 F2 00 00
01 16 09 06 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 00 00 00 00 00 00
09 06 0A 05 00 F2 00 00
01 16 0A 05 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 01 FB 00 00 00 00
09 0A 0B 07 00 F2 00 00 00 00 00 32
01 3C 0B 09 00 F2 00 00 00 00 00 32 3C 3F 78 6D 6C 20 76 65 72 73 69 6F 6E
3D 22 31 2E 30 22 20 65 6E 63 6F 64 69 6E 67 3D 22 55 54 46 2D 38 22 3F 3E
0D 0A 3C 70 72 65 66 65 72 65 6E 63
09 0A 0C 07 00 F2 00 00 00 00 00 32
01 3C 0C 09 00 F2 00 00 00 00 00 32 65 73 20 76 65 72 73 69 6F 6E 3D 22 31
22 20 6D 6F 64 69 66 69 65 64 3D 22 57 65 64 20 32 2E 20 44 65 63 20 31 34
3A 35 38 3A 31 39 20 32 30 31 35 22
09 0A 0D 07 00 F2 00 00 00 00 00 32
01 3C 0D 09 00 F2 00 00 00 00 00 32 3E 0D 0A 20 20 20 20 3C 77 61 74 63 68
4E 61 6D 65 3E 74 6F 6D 74 6F 6D 3C 2F 77 61 74 63 68 4E 61 6D 65 3E 0D 0A
20 20 20 20 3C 53 79 6E 63 54 69 6D
09 0A 0E 07 00 F2 00 00 00 00 00 32
01 3C 0E 09 00 F2 00 00 00 00 00 32 65 54 6F 50 43 3E 31 3C 2F 53 79 6E 63
54 69 6D 65 54 6F 50 43 3E 0D 0A 20 20 20 20 3C 53 65 6E 64 41 6E 6F 6E 79
6D 6F 75 73 44 61 74 61 3E 30 3C 2F
09 0A 0F 07 00 F2 00 00 00 00 00 32
01 3C 0F 09 00 F2 00 00 00 00 00 32 53 65 6E 64 41 6E 6F 6E 79 6D 6F 75 73
44 61 74 61 3E 0D 0A 20 20 20 20 3C 57 61 74 63 68 57 69 6E 64 6F 77 4D 69
6E 69 6D 69 7A 65 64 3E 30 3C 2F 57
09 0A 10 07 00 F2 00 00 00 00 00 32
01 3C 10 09 00 F2 00 00 00 00 00 32 61 74 63 68 57 69 6E 64 6F 77 4D 69 6E
69 6D 69 7A 65 64 3E 0D 0A 20 20 20 20 3C 43 6F 6E 66 69 67 55 52 4C 3E 68
74 74 70 73 3A 2F 2F 6D 79 73 70 6F
09 0A 11 07 00 F2 00 00 00 00 00 32
01 3C 11 09 00 F2 00 00 00 00 00 32 72 74 73 2E 74 6F 6D 74 6F 6D 2E 63 6F
6D 2F 73 65 72 76 69 63 65 2F 63 6F 6E 66 69 67 2F 63 6F 6E 66 69 67 2E 6A
73 6F 6E 3C 2F 43 6F 6E 66 69 67 55
09 0A 12 07 00 F2 00 00 00 00 00 32
01 3C 12 09 00 F2 00 00 00 00 00 32 52 4C 3E 0D 0A 20 20 20 20 3C 65 78 70
6F 72 74 65 72 73 3E 0D 0A 20 20 20 20 20 20 20 20 3C 6F 66 66 6C 69 6E 65
3E 0D 0A 20 20 20 20 20 20 20 20 3C
09 0A 13 07 00 F2 00 00 00 00 00 32
01 3C 13 09 00 F2 00 00 00 00 00 32 65 78 70 6F 72 74 20 69 64 3D 22 67 70
[... 110 lines cut ...]
09 02 49 12
01 16 49 12 00 00 00 00 00 01 01 01 00 00 00 00 00 00 76 98 00 00 00 00
09 02 4A 12
01 16 4A 12 00 00 00 00 00 85 00 00 00 00 00 00 00 00 07 DE 00 00 00 00
09 02 4B 12
01 16 4B 12 00 00 00 00 00 01 01 00 00 00 00 00 00 00 7D 96 00 00 00 00
09 02 4C 12
01 16 4C 12 00 00 00 00 00 02 00 01 00 00 00 00 00 00 00 48 00 00 00 00
09 02 4D 12
01 16 4D 12 00 00 00 00 00 F2 00 00 00 00 00 00 00 00 01 FB 00 00 00 00
09 02 4E 12
01 16 4E 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01
09 02 4F 1A
01 06 4F 1A 00 00 10 8B
09 02 50 0A
01 02 50 0A
09 02 51 23
01 05 51 23 69 00 00
09 02 52 23
01 05 52 23 67 00 00

>
> Have you ever connected the watch using the Windows software? If not, try
using the --initial-setup command. That will create a default preferences
file and you should be able to set a list of formats after that.

ryanbinns, 2015-12-02:
That packet list shows that there are no activities on your watch. Perhaps it was downloaded previously? The first command of your first message for this issue would have downloaded activities and put them in `~/ttwatch/<watchname>/<date>`. Activities can only be downloaded once - they are deleted from the watch after being downloaded, but only if downloaded and saved (and verified) successfully.

mbrouer, 2015-12-03:
ahh, ok.. so I guess that the activity I can see with --list-history is not
valid because it was created before the initial setup.
I will try to wipe out the watch and start over :)

sudo /usr/local/bin/ttwatch  --list-history
Running:
1: 2015/12/02 07:54:11,  933s, 15820.30m,  193 calories

Thank you very much for all your help :)

ryanbinns, 2015-12-03:
No, you don't need to wipe it. The watch keeps history separate to the activity data. What you can see now is only a summary. It is used so you can look back through your watch at previous activities to compare them, and for supporting the Race feature, where you can race against a previous activity. The detailed activity data, however, is deleted after it is downloaded to save storage space on the watch.

The activity you can see is still valid, but it's only a summary. It tells you date/time, duration, distance, calories, lap count (if you were doing laps), race position (if you were doing a race) and not much more.

mbrouer, 2015-12-03:
Awesome..
I have it working now, thank you very much for all your help :)

closed 2015-12-14 by ryanbinns