ryanbinns/ttwatch #58: Incorrect time update

bastoon06, 2015-12-13:
ttwatch --set-time wrongly update tomtom watch time. In my case, it updated it to 2010/01/04 16hXX while time was 2015/12/13 13HXX.

This occurs since firmware update. Using now firmware 1.8.46.

dlenski, 2015-12-13:
Try connecting to the GPS satellites and then doing `ttwatch --set-time` again.

The TomTom watch doesn't actually store the current clock time: it only stores the offset between UTC time (which it gets from the GPS satellites) and the local time zone. For example, `utc_offset=-18000` if you're in the Eastern USA (UTC-5 hours).

If the watch loses the correct UTC time, then it won't be able to set the correct local time either...

ryanbinns, 2015-12-14:
I saw this a few weeks ago when I got a new watch (my last one died) and connected it to the computer before doing a GPS activity. It set the time back to 1st January 2010. All I had to do was go outside and let the watch sync to GPS (by setting up an activity - I didn't need to actually start it), then reconnect it to the computer and set the time again. The ttwatch and TomTom software just sets the offset from UTC, so would behave identically (IIRC) in this circumstance.

DirkBri, 2015-12-26:
Hi Ryhan, I observe this here:

dirk@PC1:~$ date
Sa 26. Dez 20:03:00 CET 2015
dirk@PC1:~$ ttwatch --get-time
UTC time:   2010-01-01 03:07:13
Local time: 2015-12-26 20:03:13 (UTC+52456.9)

It's new. I am sure to have not seen this in November.

KR Dirk

DirkBri, 2015-12-27:
OK, after GPS-Sync everythink is fine

dirk@PC1:~$ date
So 27. Dez 09:38:13 CET 2015
dirk@PC1:~$ ttwatch --get-time
UTC time:   2015-12-27 08:38:13
Local time: 2015-12-27 09:38:13 (UTC+1)

ryanbinns, 2015-12-27:
Ok, glad it's working now. The watch automatically resets the time to 2010-01-01 00:00:00 when it is factory-reset or starts up for the first time, and apparently when it crashes too. Unfortunately, there is no way (that I know of) of telling the watch what the UTC time is to fix this - a GPS sync must be done.

closed 2015-12-27 by ryanbinns