ryanbinns/ttwatch #60: Activities only downloaded once

derickr, 2016-01-01:
For some reason (could be my fault), after I first ran --get-activities as root, it put it nicely in /root/ttwatch/<name>/<time>..., but a subsequent run with my normal user, did not create /home/username/ttwatch with the activities files? Is there a way to reset that?

dlenski, 2016-01-01:
ttwatch deletes the activity files from the watch after downloading them (as do the official TomTom clients). So they won't be re-fetched on subsequent invocation.

ryanbinns, 2016-01-01:
As Dan said, activities are deleted after being downloaded so they don't take up storage space on the watch, otherwise the watch storage will fill up and there wouldn't be space for new activities.

closed 2016-01-01 by ryanbinns