ryanbinns/ttwatch #62: Allow swim sport to be exported as tcx

mipapo, 2016-01-05:
Not working in my case
Do you need example files?

dlenski, 2016-01-05:
I did this for treadmill activities in #38 + #40. Adding swimming support to `export_tcx` will likely be similar, but I only have a Runner so couldn't test this myself.

mipapo, 2016-01-05:
I will try it in a moment - Thanks for the hint

mipapo, 2016-01-05:
Example file. At the moment I don't get the converter running. Just getting an empty file with my changes.
[00910000_20151222_114839.ttbin.zip](https://github.com/ryanbinns/ttwatch/files/78760/00910000_20151222_114839.ttbin.zip)

mipapo, 2016-09-03:
Would be so great if anyone has an idea how to add that to the converter

misdoro, 2017-01-22:
@mipapo do you have any example TCX files of swim activities recorded with other watches?
Btw, Tomtom watches do not support measuring heart rate while swimming, according to their specification.

mipapo, 2017-01-22:
[indoor_swim.tcx.zip](https://github.com/ryanbinns/ttwatch/files/722189/indoor_swim.tcx.zip)
Created with garmin 920XT - Exported tcx from Garmin Connect

misdoro, 2017-01-22:
@mipapo I've produced an export file from your ttbin from the message of 05 january 2016, does it look reasonable to you?
[Pool_swim_10-51-32.tcx.zip](https://github.com/ryanbinns/ttwatch/files/722279/Pool_swim_10-51-32.tcx.zip)

I'll prepare a merge request, but it depends on changes for other indoor activities, so it will go the last.
If you have a few other ttbin files, especially some recorded with an external heart rate monitor, it will be cool.

mipapo, 2017-01-22:
@misdoro File looks good

mipapo, 2017-02-08:
@misdoro I'm not sure if TomTom can record swim sessions with heart rate? 
This is just another PoolSwimming file (don't think that it's with heartrate)
[PoolSwimming_08-54-33.ttbin.zip](https://github.com/ryanbinns/ttwatch/files/760048/PoolSwimming_08-54-33.ttbin.zip)

weechatter, 2017-03-01:
TomTom watch can not record the heart rate during swim activity:
https://de.discussions.tomtom.com/spark-runner-2-spark-3-runner-3-475/gps-puls-beim-schwimmen-spark-cardio-977673/index1.html#post1019901

mipapo, 2017-04-21:
@misdoro Just asking: Any updates? :)

mipapo, 2018-02-01:
works now! Thanks (471889afe3174e352108b9b212ab02840f5b90c0)

closed 2018-02-01 by mipapo