ryanbinns/ttwatch #69: Cannot --get-activites from TomTom Runner Cardio

stathisportaritis, 2016-02-07:
Hi there, 

downloaded and compiled your fantastic ttwatch project on RPi B running Raspbian Jessie.

Plugging a TomTom Runner Cardio watch (firmware v1.8.46), and executing `ttwatch --get-activities` doesn't do anything - no files are saved on the default activity path (no folders created either). The terminal just freezes for a split second, then it returns to the prompt without outputting any messages.

Other commands function properly (i.e., --get-name, --get-time etc)

All the best

St

dlenski, 2016-02-08:
If you run `ttwatch --list`, does it show any files numberd in the `0x0091****` range? If not, there are no activity files on the watch.

The official TomTom software, and `ttwatch`, delete activity files automatically after they are transferred from the watch. This behavior may be unexpected if you're used to most Garmin devices.

stathisportaritis, 2016-02-08:
Hmm, can't seem to find a `ttwatch --list` command - is this available only after compiling with the `--with-unsafe` option? If so, I'll try recompiling in danger mode.

The watch itself contains a few recent activities - these can be seen when running the `ttwatch --list-history`. The `ttwatch --get-formats` returns a `csv gpx` result.

dlenski, 2016-02-08:
The activities which are kept in the history file are just brief summaries. It is _normal_ for the watch to have activities listed in the history, despite the fact that the detailed GPS logs for those activities (`.ttbin` files) are no longer on the watch:
```
      --list-history         Lists the activity history as reported by the watch
```

What happens if you do `ttwatch -r 0x00910000 > 00910000.ttbin`? Does this actually save an activity file?

stathisportaritis, 2016-02-08:
Can confirm that `ttwatch -r 0x00910000 > 00910000.ttbin` does save a ttbin file.

Earlier I ran `ttwatch --get-activities -runas=pi` (note the single '-' before the command runas=pi) and it dumped lots of bin files on ~

stathisportaritis, 2016-02-08:
OK, performing a `ttbincnv 00910000.ttbin --csv` (on the previously saved .ttbin file) converts successfully.

ryanbinns, 2016-02-09:
The `--runas` option is ignored unless the `--daemon` option is specified as well. What actually happened when you entered `-runas=pi` is that it actually interpreted this as the `-r` command with a parameter (which is required to be numeric) of `unas=pi`. Since it couldn't convert this to an integer, it used a default file ID of 0, which caused the program to dump all the files on the watch to the current directory. You could run `ttwatch -r0` to have the same effect.

As I understand, currently running `ttwatch --list` shows at least one file with an ID of 0x0091****, but running `ttwatch --get-activities` does not download them. Also, running `ttwatch -r 0x00910000` does download a file. Also, you are not getting any error messages when running with `--get-activities`. Is this correct?

If so, this is odd, since the program will always report an error, including if it fails to write (or verify) a file. Is it possible for you to run the program under gdb and step through the `do_get_activities_callback` function to see where it is failing?

stathisportaritis, 2016-02-09:
Yep, I kind of figured out the silent conversion of -runas to -r.

To summarise:
`ttwatch --list` outputs about 40 lines - here's the first few:
0x00720701: 11
0x00730701: 25
0x0001009a: 136
0x00910003: 4520
0x00720704: 27
0x00730704: 45
0x00830006: 357
0x00910007: 17237

`ttwatch --get-activities` freezes the terminal for 1 sec or so, then returns with no errors and without exporting any files from the watch - like this:
```
pi@raspberrypi:~ $ ttwatch --get-activities
pi@raspberrypi:~ $
```
`ttwatch -r 0x00910007` dumps lots of data on the terminal.
`ttwatch -r 0x00910007 > 00910007.ttbin` saves the data from the watch to the filename specified.
Then, `ttbincnv 00910007.ttbin --csv` converts the file just fine.

I'll do my best with gdb, and report back within the next few days :)

dlenski, 2016-02-09:
Are you running a big-endian or little-endian Linux distro on your Raspberry Pi? (`uname -a` should show either `arm` or `armel`, I think)

If it's big-endian, then has anyone successfully run ttwatch on a big-endian system before? There could be some subtle endian bugs.

stathisportaritis, 2016-02-09:
Raspbian appears to be little-endian. `uname -a` returns:
`Linux raspberrypi 4.1.17-v7+ #834 SMP Mon Feb 1 15:17:54 GMT 2016 armv7l GNU/Linux`

Also ran two commands I found [here](https://serverfault.com/questions/163487/how-to-tell-if-a-linux-system-is-big-endian-or-little-endian) - all results confirm it's little endian
```
pi@raspberrypi:~ $ echo -n I | od -to2 | head -n1 | cut -f2 -d" " | cut -c6
1
pi@raspberrypi:~ $ python -c "import sys; print( 0 if sys.byteorder=='big' else 1 )"
1
```

dlenski, 2016-02-11:
Yeah, looks like nearly all [Debian ARM builds](https://wiki.debian.org/ArmPorts) are LE. Well, I for one am stumped, especially given that `--list` and `--read` do the right thing here!

stathisportaritis, 2016-02-14:
`--list` and `--read` work fine here too, I was mostly wondering about `--get-activities`.

So, debugging ttwatch with the --get-activities option, the debugger steps to:
`main()` (in ttwatch.c) > `do_get_activities()` > `ttwatch_enumerate_files()`. 

From the latter, it calls `RETURN_ERROR(enum_files(watch, type, files));`; inside `enum_files()` nothing is pushed_back to the `files` vector (its capacity equals 0). Btw, the `length` variable changes with each do/while iteration. 

Not sure whether this should be happening, but presumably as the `files` vector stays empty, the program doesn't have any files to process, so it terminates quietly. Any thoughts?

Hope this helps.

As a side query: are there any plans/thoughts to switch the project to C++?

ryanbinns, 2016-02-15:
Yes, the `length` and `id` values should change with each iteration. Can you check each iteration to see if `id` every contains an activity file (0x0091xxxx)? If you do see a file that matches this, check the value of the `type` parameter and make sure it equals 0x00910000. It should, but if there's some problem somewhere, it may have got corrupted I guess.

I have no plans to switch to C++ at the moment. It would be nice, but I just don't have the time for that big an effort. I have a 5 week old baby, moved house 2 weeks ago and are getting ready to start building a house, so I'm pretty flat out.

stathisportaritis, 2016-02-15:
Nice, congratulations! :) 

Debugging the `enum_files()` for every iteration of do/while, and:
- `id` always equals 0
- `type` always equals 9502720
- `length` changes

ryanbinns, 2016-03-07:
Bugger. On closer inspection, I've found a subtle but deadly bug. Or more specifically, 4 of them. The `if` statements on lines 575, 576, 593 and 594 (in `ttwatch_find_first_file` and `ttwatch_find_next_file`) are meant to check if the `file_id` and `length` pointers are null, and if not, store the file ID and length in them. However, all four lines are checking the *value* pointed at by the pointers, not the pointers themselves. If the pointers are valid and happen to point to the value 0, then they will not be changed, so the file ID and length will never be updated. This may explain a few strange occurrences over the last few months.

To fix, change:
```
if (*file_id) *file_id = be32toh(response.id);
if (*length) *length   = be32toh(response.file_size);
```
to
```
if (file_id) *file_id = be32toh(response.id);
if (length) *length   = be32toh(response.file_size);
```
in both functions. I'll check in a fix.

ryanbinns, 2016-05-18:
Have you had any luck downloading activities yet?

stathisportaritis, 2016-05-18:
Hi R, sorry for disappearing. Just today I downloaded the latest version of your project, installed it on a Raspberry Pi 3. Connected the watch and ran `sudo ttwatch --get-activities` - the terminal prompted with "Downloading elevation data" a few times, and also some lines with "Unable to write file formats: gpx". The watch's storage indicator is empty so that's all working fine and good.

I think I messed up and should have specified the "--runas:username"; I can't find the files stored on the Pi (logged in as 'pi', which is the default user). Any thoughts on this?

ryanbinns, 2016-05-19:
You don't need to specify `--runas` unless you're running the program as a daemon (`--daemon`). It has no effect when running from the command line.

The activity files should have been put in `~/ttwatch`, so probably `/home/pi/ttwatch`.

If you can find the activity files, you can run `ttbincnv -a` to try to convert them again. If this still fails, you can send one to me if you like and I can have a look why it's failing.

stathisportaritis, 2016-05-28:
Hi R, sorry for the delay again.

It works perfectly. Just git pulled today the latest snapshot and went through building, installing and setting up privileges for user 'pi'. 

Running `ttwatch --get-activities` creates all necessary folders in `home/pi/` and pulls all the data form the watch. 

Nice one, thanks a massive bunch for all the work you've put to this!! :)

closed 2016-05-28 by stathisportaritis

ryanbinns, 2016-05-30:
No worries. Glad it's working now.