ryanbinns/ttwatch #70: usblib-1.0-16 or higher

wv1, 2016-03-04:
Hello,

I tried to make & compile this application but to no avail....

I use Ubuntu 14.04.4 LTS.  libusb that is installable from software center is 1.0-0

When configuring I get the following message:

```
checking libusb-1.0/libusb.h usability... no
checking libusb-1.0/libusb.h presence... no
checking for libusb-1.0/libusb.h... no
checking for libusb_init in -lusb-1.0... no
configure: error: **The libusb version must be 1.0.16 or higher**
```

So I tried to download a more recent package from the libusb site ( http://sourceforge.net/projects/libusb/ ) and follow all instructions rigorously, but I get cascaded further with all kinds of other dependencies. Is there an easy way to get libusb-1.0-16 installed in Ubuntu?

I looked on the internet for solutions but none were applicable. Though this seems an issue that lots of people must be struggling with. I have a straight forward setup.

Thank you very much. I am just a Linux user, not a developer. And this is the only anchor I have to get my TomTom watch to work for me.  So again, thank you very much!

kingosticks, 2016-03-04:
Ubuntu 14.0.4 has [version 1.0.17](http://packages.ubuntu.com/trusty/libusb-1.0-0-dev)   You probably just need the dev package. Do you have libusb-1.0-0-dev package installed?  You'll also need a libcurl4-*-dev package. This could be clearer in the readme.

ryanbinns, 2016-03-07:
Hi guys. I compiled and installed the libraries from source, so I don't know which packages you need to install. It should work if you install the -dev packages though, as the headers aren't usually installed for non-dev packages.

kingosticks, 2016-03-07:
Hi @ryanbinns (great project). My point was just that the readme could be a little clearer that in order to compile the source you do actually need the -dev packages. Some people might not realise that.

ryanbinns, 2016-03-07:
I didn't realise I had to install the -dev package when I installed from source. Maybe I did, but it was quite a while ago, so I don't remember.

derickr, 2016-03-08:
You should not have to install the -dev package when you compile (libusb) from *source*. It's very likely the headers that this -dev package installs are from the older version - the one that the non -dev package would install. If Ubuntu has 1.0.17 as @kingosticks says, then you need to install the -dev package for that version of libusb too. The distribution makes sure that is the case, as is correctly has dependencies for libusb-1.0-0-dev to libusb-1.0-0 as per http://packages.ubuntu.com/trusty/libusb-1.0-0-dev

kingosticks, 2016-03-08:
Ahhh OK, I misunderstood @ryanbinns's original comment. And as it turns out, the README also.

I think the point I should have made is that, it could be a little clearer that if you are want to satisfy the ttwatch build decencies via your system package manager (rather than building *everything* from source) you must install the -dev packges of the listed dependencies.  This probably isn't obvious to everyone, particularly those that might prefer to let their package manager help them with this. And it'll be the easiest way for some people to get up and running with ttwatch. So maybe it's worth including a section in the README that explicitly mentions this.

ryanbinns, 2016-05-04:
I have added a section to the `README.md` file about using prebuilt packages.

closed 2016-05-04 by ryanbinns