ryanbinns/ttwatch #73: What does set-time actually do?

marcust, 2016-04-05:
I kind of expected that 
```
  --set-time             Updates the time on the watch
```
means that the clock time will be set.
But I get this output:
```
$ date; ttwatch --set-time; ttwatch --get-time
Di 5. Apr 21:23:15 CEST 2016
UTC time:   2010-02-01 03:27:17
Local time: 2010-02-01 05:27:17 (UTC+2)
```

So this does not work. I had a look at the sources (https://github.com/ryanbinns/ttwatch/blob/master/ttwatch/ttwatch.c#L309-L326) and it looks to me as if it just updates the GMT offset. So --set-time actually just sets the offset to GMT?

I found no functionality to actually set the time in libttwatch, so I figure that is just not implemented?

dlenski, 2016-04-05:
See #58.

It seems that it is not actually possibly to set "the time" on the TomTom watches. They only store the displayed time as an offset relative to the UTC time obtained by the GPS.

That means that if the watch doesn't have the correct UTC time (normally only when it is brand new or has been factory reset!), it won't be set correctly. So you have to try the GPS before you can set the time correctly.

marcust, 2016-04-05:
Oh, sorry for the duplicate, thanks for the answer. 

That's true, getting a GPS fix sets the time. 

The option is still confusing though. Even more confusing is the fact that it asks for the time after factory reset.

closed 2016-04-05 by marcust

ryanbinns, 2016-04-06:
Yes, the watch asks for the time after a factory reset, but all they do is adjust the UTC offset so that the internal time (which is not user-settable but automatically updates from GPS) matches what you'd like to see. It sounds a little bizarre, but it's much simpler to manage than having two internal clocks and keep them synchronised somehow.

The downside is that a new watch with a wrong UTC time shows strange times when you do a set-time, and then enable the GPS. It works well if you do it the other way around however.