ryanbinns/ttwatch #75: Porting to freebsd

polle58, 2016-04-10:
I use ttwatch under linuc and it works very good. However I prefer to use freebsd, tried to install ttwatch but it fails according to find libusb. Would it be posible to port ttwtch to freebsd?

ryanbinns, 2016-04-11:
To be perfectly honest, I have no idea. I have never used or written software for FreeBSD, and I don't know the differences between it and linux. I suspect that it is close enough that the software should work with minor changes. Have you downloaded libusb ([www.libusb.org](url)) and compiled/installed it? It probably isn't available on the system by default so you'd need to either install a package (or whatever the equivalent is for FreeBSD or compile from source.

polle58, 2016-04-11:
yan, thanks for your answer, from freebsd 8 (current freebsd 10),  libusb is included .FreeBSD 8 and above include a FreeBSD-specific reimplementation of the libusb-1.0 API, so your applications will probably work there too. The source code for this library can be found ​here: http://svnweb.freebsd.org/base/head/lib/libusb/
I think it's rather similar because I have an acr38 id  usb card reader and after installing the mac-linux driver it works.( I installed linux compability on freebsd). When I try to install ttwatch-master it fails by not finding libusb.h or similar (don t remember the exact error). I would like to try more, but my knowledge of programming is almost zero. Thanks for your nice work and if I can do some more test to find the problem let me know, thanks again

ryanbinns, 2016-04-12:
Could you attach the `config.log` file that `configure` produced to this issue? That may help to determine exactly what failed.

polle58, 2016-04-12:
[config.txt](https://github.com/ryanbinns/ttwatch/files/214907/config.txt)

polle58, 2016-04-12:
just changed the extension to txt  if not I could not upload it

ryanbinns, 2016-04-12:
I see. Any installation of libusb on linux (whether from a package or doing a make install from source) puts the libusb.h file in a subdirectory. Your FreeBSD system hasn't, so configure can't find it. I'd have to add a FreeBSD specific configuration flag or get configure to try looking in multiple places. I'll see how I go, but it may take a little while.

polle58, 2016-04-12:
Thanks , I still have a working linux on other harddisk, so I am not in a hurry. Thanks for your effort

ryanbinns, 2016-05-04:
I have uploaded a change that will search for `libusb.h` in a few locations. Can you give it a go and let me know if it works for FreeBSD?

polle58, 2016-05-05:
I tried and some progress:
checking libusb.h usability... yes
checking libusb.h presence... yes
checking for libusb.h... yes
checking for libusb_init in -lusb-1.0... no
configure: error: The libusb version must be 1.0.16 or higher
[config.txt](https://github.com/ryanbinns/ttwatch/files/250077/config.txt)

polle58, 2016-05-05:
I found something else that doesn't sound good: "The library is also compliant with	LibUSB version 0.1.12"

[https://www.freebsd.org/cgi/man.cgi?query=libusb&sektion=3](url)

ryanbinns, 2016-05-05:
It looks like FreeBSD uses a different library name too, not just a different include location.

Can you look in your libusb.h and find the value of LIBUSB_API_VERSION?

polle58, 2016-05-05:
Ryan , I tried to find that value, sure you mean the libusb.h in /usr/ include. this is the file:
[libusb.h.txt](https://github.com/ryanbinns/ttwatch/files/250962/libusb.h.txt)
Found also this ( but hata's about frreebsd 11, I have 10.3)

[https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=199256](url)

polle58, 2016-05-06:
For testing purposes I installed OpenBSD 5.9 om a usb stick; it is different from freebsd as it uses libusb. I installed libusb-1.0.20 (same file as I installed on my linuxmint)> After that I tried to compile ttwatch. It fails:`checking libusb.h usability... no
checking libusb.h presence... no
checking for libusb.h... no
checking /usr/include/libusb.h usability... no
checking /usr/include/libusb.h presence... no
checking for /usr/include/libusb.h... no
checking /usr/include/libusb-1.0/libusb.h usability... no
checking /usr/include/libusb-1.0/libusb.h presence... no
checking for /usr/include/libusb-1.0/libusb.h... no
checking /usr/local/include/libusb.h usability... no
checking /usr/local/include/libusb.h presence... no
checking for /usr/local/include/libusb.h... no
checking /usr/local/include/libusb-1.0/libusb.h usability... yes
checking /usr/local/include/libusb-1.0/libusb.h presence... yes
checking for /usr/local/include/libusb-1.0/libusb.h... yes
checking for libusb_init in -lusb-1.0... no
checking curl/curl.h usability... no
checking curl/curl.h presence... no
checking for curl/curl.h... no
checking for curl_easy_init in -lcurl... no
configure: Errors prevent Makefile being generated
`
This is the config log:
[config.log.txt](https://github.com/ryanbinns/ttwatch/files/252079/config.log.txt)
Her I found th libusb>h in /usr/local/include/libusb-1.0
`/** \def LIBUSB_API_VERSION
 * \ingroup misc
 * libusb's API version.
 *
 * Since version 1.0.13, to help with feature detection, libusb defines
 * a LIBUSB_API_VERSION macro that gets increased every time there is a
 * significant change to the API, such as the introduction of a new call,
 * the definition of a new macro/enum member, or any other element that
 * libusb applications may want to detect at compilation time.
 *
 * The macro is typically used in an application as follows:
 * \code
 * #if defined(LIBUSB_API_VERSION) && (LIBUSB_API_VERSION >= 0x01001234)
 * // Use one of the newer features from the libusb API
 * #endif
 * \endcode
 *
 * Internally, LIBUSB_API_VERSION is defined as follows:
 * (libusb major << 24) | (libusb minor << 16) | (16 bit incremental)
 */
#define LIBUSB_API_VERSION 0x01000104

/* The following is kept for compatibility, but will be deprecated in the future */
#define LIBUSBX_API_VERSION LIBUSB_API_VERSION`

[libusb.h.txt](https://github.com/ryanbinns/ttwatch/files/252081/libusb.h.txt)

 Openbsd is not realy usable as desktop, but maybe it helps to find the issues

ryanbinns, 2016-05-06:
Ok, second attempt. I have modified the configure code to skip the version check and change the expected library name for FreeBSD. Specify the `--with-freebsd` command line option when running `configure`. Hopefully it will work this time!

polle58, 2016-05-06:
The config file created without errors, but now I get this with make:

root@freebsd:/home/eddy/Downloads/ttwatch-master # make
make -C ttbin
cc -O2 -pipe   -g -O0 -c ttbin.c -o ttbin.o
ttbin.c:15:10: fatal error: 'curl/curl.h' file not found
#include <curl/curl.h>
         ^
1 error generated.
*** Error code 1

Stop.
make[1]: stopped in /usr/home/eddy/Downloads/ttwatch-master/ttbin
*** Error code 1

Stop.
make: stopped in /usr/home/eddy/Downloads/ttwatch-master
root@freebsd:/home/eddy/Downloads/ttwatch-master #

polle58, 2016-05-07:
Ryan I modified the make file  with CFLAGS="-I/usr/local/include"
now I get other error:
[Makefile.txt](https://github.com/ryanbinns/ttwatch/files/253523/Makefile.txt)
[outputmake.txt](https://github.com/ryanbinns/ttwatch/files/253524/outputmake.txt)

polle58, 2016-05-08:
Further reading , I found that linux use gnu make en freebsd his own make, but freebsd has gmake, so i tried this:

$ cd /home/eddy/Downloads/ttwatch-master
$ gmake
gmake -C ttbin
gmake[1]: Entering directory '/usr/home/eddy/Downloads/ttwatch-master/ttbin'
gmake[1]: 'libttbin.a' is up to date.
gmake[1]: Leaving directory '/usr/home/eddy/Downloads/ttwatch-master/ttbin'
gmake -C libttwatch
gmake[1]: Entering directory '/usr/home/eddy/Downloads/ttwatch-master/libttwatch'
gmake[1]: 'libttwatch.a' is up to date.
gmake[1]: Leaving directory '/usr/home/eddy/Downloads/ttwatch-master/libttwatch'
gmake -C ttwatch
gmake[1]: Entering directory '/usr/home/eddy/Downloads/ttwatch-master/ttwatch'
Compiling ttwatch.o...
ttwatch.c:90:19: error: 'MANIFEST_DEFINITION_00010819_COUNT' undeclared here (not in a function)
     { 0x00010819, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                   ^
ttwatch.c:90:55: error: 'MANIFEST_DEFINITIONS_00010819' undeclared here (not in a function)
     { 0x00010819, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                                                       ^
ttwatch.c:94:19: error: 'MANIFEST_DEFINITION_0001082e_COUNT' undeclared here (not in a function)
     { 0x0001082e, MANIFEST_DEFINITION_0001082e_COUNT, MANIFEST_DEFINITIONS_0001082e },
                   ^
ttwatch.c:94:55: error: 'MANIFEST_DEFINITIONS_0001082e' undeclared here (not in a function)
     { 0x0001082e, MANIFEST_DEFINITION_0001082e_COUNT, MANIFEST_DEFINITIONS_0001082e },
                                                       ^
ttwatch.c:95:19: error: 'MANIFEST_DEFINITION_00010113_COUNT' undeclared here (not in a function)
     { 0x00010113, MANIFEST_DEFINITION_00010113_COUNT, MANIFEST_DEFINITIONS_00010113 },
                   ^
ttwatch.c:95:55: error: 'MANIFEST_DEFINITIONS_00010113' undeclared here (not in a function)
     { 0x00010113, MANIFEST_DEFINITION_00010113_COUNT, MANIFEST_DEFINITIONS_00010113 },
                                                       ^
ttwatch.c:1812:5: error: unknown type name 'libusb_hotplug_event'
     libusb_hotplug_event event, void *user_data)
     ^
ttwatch.c: In function 'main':
ttwatch.c:2346:35: error: 'LIBUSB_CAP_HAS_HOTPLUG' undeclared (first use in this function)
         if (libusb_has_capability(LIBUSB_CAP_HAS_HOTPLUG))
                                   ^
ttwatch.c:2346:35: note: each undeclared identifier is reported only once for each function it appears in
ttwatch.c:2349:66: error: 'LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED' undeclared (first use in this function)
             if ((result = libusb_hotplug_register_callback(NULL, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED,
                                                                  ^
ttwatch.c:2350:17: error: 'LIBUSB_HOTPLUG_ENUMERATE' undeclared (first use in this function)
                 LIBUSB_HOTPLUG_ENUMERATE, TOMTOM_VENDOR_ID, TOMTOM_MULTISPORT_PRODUCT_ID,
                 ^
ttwatch.c:2351:17: error: 'LIBUSB_HOTPLUG_MATCH_ANY' undeclared (first use in this function)
                 LIBUSB_HOTPLUG_MATCH_ANY, hotplug_attach_callback, options, NULL)) != 0)
                 ^
ttwatch.c:2351:43: error: 'hotplug_attach_callback' undeclared (first use in this function)
                 LIBUSB_HOTPLUG_MATCH_ANY, hotplug_attach_callback, options, NULL)) != 0)
                                           ^
Makefile:17: recipe for target 'ttwatch.o' failed
gmake[1]: *** [ttwatch.o] Error 1
gmake[1]: Leaving directory '/usr/home/eddy/Downloads/ttwatch-master/ttwatch'
Makefile:12: recipe for target 'all' failed
gmake: *** [all] Error 2

ryanbinns, 2016-05-17:
It would seem that the FreeBSD libusb library doesn't support hot-plug detection, which `ttwatch` uses. I could add an option to remove daemon support (which is the only part that requires hot-plug detection), but that would take a fair bit of effort. I'll think about it, but I just don't get a lot of time to work on this these days.

polle58, 2016-05-18:
Ryan, Already a big thanks for your effort,. would be nice if you could get it to work, but no hurry, for the moment I will boot  my copy of linuxmint to use ttwatch .

ryanbinns, 2016-05-18:
Ok, I've pushed a fix that adds a flag to disable daemon mode, so it will hopefully compile under FreeBSD now. Note that it's moved from a `configure` based build to `cmake`, so when you download the new version, run:
```
cmake -Ddaemon=off .
make
```
instead of the old build method. If you don't have `cmake` installed, you'll have to install it. There should be a package available for FreeBSD.

polle58, 2016-05-18:
Still no succes:
$ cd /home/eddy/Downloads/ttwatch-master
$  mkdir build
$ cd build && cmake .. -Ddaemon=off
-- The C compiler identification is Clang 3.4.1
-- The CXX compiler identification is Clang 3.4.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/CC
-- Check for working CXX compiler: /usr/bin/CC -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/local/lib/libcurl.so (found version "7.48.0") 
-- Found OpenSSL: /usr/lib/libssl.so;/usr/lib/libcrypto.so (found version "1.0.1s") 
-- Found libusb-1.0:
--  - Includes: /usr/include
--  - Libraries: /usr/lib/libusb.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eddy/Downloads/ttwatch-master/build
$ make
Scanning dependencies of target libttbin
[  5%] Building C object CMakeFiles/libttbin.dir/src/export_csv.c.o
[ 11%] Building C object CMakeFiles/libttbin.dir/src/export_gpx.c.o
[ 16%] Building C object CMakeFiles/libttbin.dir/src/export_kml.c.o
[ 22%] Building C object CMakeFiles/libttbin.dir/src/export_tcx.c.o
[ 27%] Building C object CMakeFiles/libttbin.dir/src/ttbin.c.o
[ 33%] Linking C static library libttbin.a
[ 33%] Built target libttbin
Scanning dependencies of target ttbincnv
[ 38%] Building C object CMakeFiles/ttbincnv.dir/src/ttbincnv.c.o
[ 44%] Linking C executable ttbincnv
[ 44%] Built target ttbincnv
Scanning dependencies of target manifest
./make_manifest.pl: not found
*** Error code 127

Stop.
make[2]: stopped in /usr/home/eddy/Downloads/ttwatch-master/build
*** Error code 1

Stop.
make[1]: stopped in /usr/home/eddy/Downloads/ttwatch-master/build
*** Error code 1

Stop.
make: stopped in /usr/home/eddy/Downloads/ttwatch-master/build

polle58, 2016-05-18:
When I try with gmake instead of make:
$ cd /home/eddy/Downloads/ttwatch-master
$  mkdir build
$ cd build && cmake .. -Ddaemon=off
-- The C compiler identification is Clang 3.4.1
-- The CXX compiler identification is Clang 3.4.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/CC
-- Check for working CXX compiler: /usr/bin/CC -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/local/lib/libcurl.so (found version "7.48.0") 
-- Found OpenSSL: /usr/lib/libssl.so;/usr/lib/libcrypto.so (found version "1.0.1s") 
-- Found libusb-1.0:
--  - Includes: /usr/include
--  - Libraries: /usr/lib/libusb.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eddy/Downloads/ttwatch-master/build
$ gmake
Scanning dependencies of target libttbin
[  5%] Building C object CMakeFiles/libttbin.dir/src/export_csv.c.o
[ 11%] Building C object CMakeFiles/libttbin.dir/src/export_gpx.c.o
[ 16%] Building C object CMakeFiles/libttbin.dir/src/export_kml.c.o
[ 22%] Building C object CMakeFiles/libttbin.dir/src/export_tcx.c.o
[ 27%] Building C object CMakeFiles/libttbin.dir/src/ttbin.c.o
[ 33%] Linking C static library libttbin.a
[ 33%] Built target libttbin
Scanning dependencies of target ttbincnv
[ 38%] Building C object CMakeFiles/ttbincnv.dir/src/ttbincnv.c.o
[ 44%] Linking C executable ttbincnv
[ 44%] Built target ttbincnv
Scanning dependencies of target libttwatch
[ 50%] Building CXX object CMakeFiles/libttwatch.dir/src/libttwatch.cpp.o
/home/eddy/Downloads/ttwatch-master/src/libttwatch.cpp:14:10: fatal error: 'endian.h' file
      not found
#include "endian.h"
         ^
1 error generated.
CMakeFiles/libttwatch.dir/build.make:62: recipe for target 'CMakeFiles/libttwatch.dir/src/libttwatch.cpp.o' failed
gmake[2]: *** [CMakeFiles/libttwatch.dir/src/libttwatch.cpp.o] Error 1
CMakeFiles/Makefile2:249: recipe for target 'CMakeFiles/libttwatch.dir/all' failed
gmake[1]: *** [CMakeFiles/libttwatch.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
gmake: *** [all] Error 2

polle58, 2016-05-18:
If  I change in libttwatch.cpp the line #include "endian.h'   with this:      #include "sys/endian.h"

then it fails with same error:
$ gmake
[ 33%] Built target libttbin
[ 44%] Built target ttbincnv
Scanning dependencies of target libttwatch
[ 50%] Building CXX object CMakeFiles/libttwatch.dir/src/libttwatch.cpp.o
[ 55%] Building CXX object CMakeFiles/libttwatch.dir/src/libttwatch_cpp.cpp.o
[ 61%] Linking CXX static library libttwatch.a
[ 61%] Built target libttwatch
Scanning dependencies of target manifest
./make_manifest.pl: not found
CMakeFiles/manifest.dir/build.make:57: recipe for target 'CMakeFiles/manifest' failed
gmake[2]: *** [CMakeFiles/manifest] Error 127
CMakeFiles/Makefile2:180: recipe for target 'CMakeFiles/manifest.dir/all' failed
gmake[1]: *** [CMakeFiles/manifest.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
gmake: *** [all] Error 2

hoheinzollern, 2016-05-18:
The problem is that endian.h is Linux specific, hence it won't compile on FreeBSD. I solved this problem on OSX on commit 220da8e247f6d159e1b49ad2a39685b640836d3e by adding the following on libttwatch.cpp:

```
#ifndef __APPLE__
#include "endian.h"
#else
#include <libkern/OSByteOrder.h>

#define htobe16(x) OSSwapHostToBigInt16(x)
#define htole16(x) OSSwapHostToLittleInt16(x)
#define be16toh(x) OSSwapBigToHostInt16(x)
#define le16toh(x) OSSwapLittleToHostInt16(x)

#define htobe32(x) OSSwapHostToBigInt32(x)
#define htole32(x) OSSwapHostToLittleInt32(x)
#define be32toh(x) OSSwapBigToHostInt32(x)
#define le32toh(x) OSSwapLittleToHostInt32(x)

#define htobe64(x) OSSwapHostToBigInt64(x)
#define htole64(x) OSSwapHostToLittleInt64(x)
#define be64toh(x) OSSwapBigToHostInt64(x)
#define le64toh(x) OSSwapLittleToHostInt64(x)
#endif
```

Basically converting linux specific calls into the corresponding OSX library. It would be nice to find an OS independent way of making these calls, to support FreeBSD as well.

hoheinzollern, 2016-05-18:
This might be relevant: http://svn.apache.org/repos/asf/openoffice/trunk/main/sal/inc/osl/endian.h looks like each OS has their own incompatible version of `endian.h` so it would be good to follow some big project on this issue.

ryanbinns, 2016-05-19:
`cmake` supports determining system endianness, so I could edit the build system to determine at compile time whether or not byte swapping was required and implement the two macros required by hand. It won't be difficult.

polle58, 2016-05-20:
Still trying. now I don't get the warning about the endian.h but still make fails complaining that it can not find manifest.pl:
 $ make
[ 28%] Built target libttbin
[ 38%] Built target ttbincnv
[ 42%] Generating manifest_definitions_0001082e.h
../manifest/make_manifest.pl: not found
*** Error code 127

I checked but in the directory  manifest there was  a file manifest.pl present. Just for a try a tried to rename that file ( copy paste from the error message) and suddenly I got further but still no luck (als strange because the file name for me looked identical) Now I get this:
$ make
[ 28%] Built target libttbin
[ 38%] Built target ttbincnv
[ 52%] Built target manifest
[ 66%] Built target libttwatch
[ 71%] Building C object CMakeFiles/ttwatch.dir/src/ttwatch.c.o
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:91:19: error: use of undeclared identifier 'MANIFEST_DEFINITION_00010819_COUNT'
    { 0x00010819, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:91:55: error: use of undeclared identifier 'MANIFEST_DEFINITIONS_00010819'; did
      you mean 'MANIFEST_DEFINITIONS'?
    { 0x00010819, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                      MANIFEST_DEFINITIONS
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:89:3: note: 'MANIFEST_DEFINITIONS' declared here
} MANIFEST_DEFINITIONS[] =
  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:92:19: error: use of undeclared identifier 'MANIFEST_DEFINITION_00010819_COUNT'
    { 0x00010822, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:92:55: error: use of undeclared identifier 'MANIFEST_DEFINITIONS_00010819'; did
      you mean 'MANIFEST_DEFINITIONS'?
    { 0x00010822, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                      MANIFEST_DEFINITIONS
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:89:3: note: 'MANIFEST_DEFINITIONS' declared here
} MANIFEST_DEFINITIONS[] =
  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:93:19: error: use of undeclared identifier 'MANIFEST_DEFINITION_00010819_COUNT'
    { 0x00010823, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:93:55: error: use of undeclared identifier 'MANIFEST_DEFINITIONS_00010819'; did
      you mean 'MANIFEST_DEFINITIONS'?
    { 0x00010823, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                      MANIFEST_DEFINITIONS
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:89:3: note: 'MANIFEST_DEFINITIONS' declared here
} MANIFEST_DEFINITIONS[] =
  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:94:19: error: use of undeclared identifier 'MANIFEST_DEFINITION_00010819_COUNT'
    { 0x0001082a, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:94:55: error: use of undeclared identifier 'MANIFEST_DEFINITIONS_00010819'; did
      you mean 'MANIFEST_DEFINITIONS'?
    { 0x0001082a, MANIFEST_DEFINITION_00010819_COUNT, MANIFEST_DEFINITIONS_00010819 },
                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                      MANIFEST_DEFINITIONS
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:89:3: note: 'MANIFEST_DEFINITIONS' declared here
} MANIFEST_DEFINITIONS[] =
  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:95:19: error: use of undeclared identifier 'MANIFEST_DEFINITION_0001082e_COUNT'
    { 0x0001082e, MANIFEST_DEFINITION_0001082e_COUNT, MANIFEST_DEFINITIONS_0001082e },
                  ^
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:95:55: error: use of undeclared identifier 'MANIFEST_DEFINITIONS_0001082e'; did
      you mean 'MANIFEST_DEFINITIONS'?
    { 0x0001082e, MANIFEST_DEFINITION_0001082e_COUNT, MANIFEST_DEFINITIONS_0001082e },
                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                      MANIFEST_DEFINITIONS
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:89:3: note: 'MANIFEST_DEFINITIONS' declared here
} MANIFEST_DEFINITIONS[] =
  ^
[... 50 lines cut ...]
    ~~~~~~~~~~~~~~~~~~~                                              ^~~~~~~~
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:1873:70: warning: implicit conversion from 'float' to 'uint32_t'
      (aka 'unsigned int') changes value from 40233.602 to 40233 [-Wliteral-conversion]
    ttwatch_create_race(watch, TTWATCH_Cycling,   2, "25MI 1HR",     40233.6f,  3600, 16, FILE_0x00710102);
    ~~~~~~~~~~~~~~~~~~~                                              ^~~~~~~~
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:1878:71: warning: implicit conversion from 'float' to 'uint32_t'
      (aka 'unsigned int') changes value from 9656.0996 to 9656 [-Wliteral-conversion]
    ttwatch_create_race(watch, TTWATCH_Treadmill, 2, "6MI 50M",       9656.1f,  3000, 12, FILE_0x00710702);
[... cut ...]
    ~~~~~~~~~~~~~~~~~~~                                               ^~~~~~~
/home/eddy/Downloads/ttwatch-master/src/ttwatch.c:1885:70: warning: implicit conversion from 'float' to 'uint32_t'
      (aka 'unsigned int') changes value from 48280.301 to 48280 [-Wliteral-conversion]
    ttwatch_create_race(watch, TTWATCH_Freestyle, 4, "30MI 2HR",     48280.3f,  7200, 30, FILE_0x00710804);
    ~~~~~~~~~~~~~~~~~~~                                              ^~~~~~~~
8 warnings and 16 errors generated.
*** Error code 1

Stop.

ryanbinns, 2016-05-20:
Just a hunch, but where does your `perl` binary live? What does `which perl` output?

ryanbinns, 2016-05-20:
My hunch was right. I installed FreeBSD in a VM and tried to compile ttwatch. `make_manifest.pl` assumed `perl` was in `/usr/bin`, but on FreeBSD it is in `/usr/local/bin`. I have added support to cmake to automatically find where `perl` is located so now it compiles properly on FreeBSD (or at least it does on my 10.3 install). I haven't tried using it since I don't want to muck around with sharing USB devices with the VM; I've never had a lot of success. Hopefully it will work. At least it compiles now...

polle58, 2016-05-20:
Ryan, thanks again for the effort, now indeed the compiling went fine, installing also. I put a small test file on my watch and was able to retrieve the ttbin with:. ttwatch --get-activities. For the moment only as root but that must be a permission problem, freebsd uses devfs instead of udev, I will look later at it and also I will test more. Thanks again, nice job.

polle58, 2016-05-20:
Don't know if it is completely the right way, as I am not an expert but I got ttwatch in freebsd 10.3 working as regular user in this way:

add two lines  to the file 
etc/devfs.rules

[usb_devices=10]
add path 'usb/*' mode 0660 group usb

 Add a line to :
/etc/rc.conf:

# Set the default devfs ruleset.
devfs_system_ruleset="usb_devices"
	    
create group usb:

# pw groupadd usb

add user to group usb

# pw groupmod usb -m username

polle58, 2016-05-20:
I tried to disconnect and reconnect the watch, after a few seconds ttwatch has again access to the tomtomcardio.
Also tried this:
$ ttwatch --update-gps
Downloading GPSQuickFix data file...
Writing file to watch...
GPSQuickFix data updated
converting the ttbin to tcx also works
For me the same features i use under linux now also work under freebsd, so realy nice!

ryanbinns, 2016-05-21:
Excellent, I'm glad it finally works. Could you add instructions to the `README.md` file saying how to get non-root USB access on FreeBSD and submit a pull request please? It will help others to get it working.

closed 2016-05-26 by ryanbinns