ryanbinns/ttwatch #76: New firmware don't seem to be found for the TomTom Spark Cardio

derickr, 2016-05-15:
Hi!

It seems that ttwatch does not find the latest firmware (1.3.27) for this watch. Mine is still stuck at 1.2.0. Let me know how I can help debugging this.

cheers,
Derick

ryanbinns, 2016-05-17:
Can you check your preferences XML files (run `ttwatch --read 0x00f20000`) and let me know what the `ConfigURL` value is? The `ttwatch` software currently uses hard-coded URLs, but if the Spark has uses a different URL, then we'll need to change that...

derickr, 2016-05-17:
After having compiled ttwatch with ``--with-unsafe``, it shows this:

```
    <ConfigURL>https://mysports.tomtom.com/service/config/config.json</ConfigURL>
```

ryanbinns, 2016-05-17:
Ok, so that's the same. Can you let me know the Product ID of the watch? `ttwatch -v` should show it.

derickr, 2016-05-17:
```
$ install/ryanbinns-ttwatch/ttwatch/ttwatch -v
Product ID:       0xd8070000
BLE Version:      0
Firmware Version: 1.2.0
...
```

ryanbinns, 2016-05-17:
Right. TomTom have changed their website layout and the location of the firmware files has changed for all watches. I'll have to add support for determining the download URLs dynamically, just in case it happens again. Which means parsing JSON (ugh).

derickr, 2016-05-17:
Is there a quick "hack" I can apply in the meanwhile?

ryanbinns, 2016-05-17:
Sure. Replace the three instances of `"http://download.tomtom.com/sweet/fitness/Firmware/"` with `"https://mysports-preprod.tomtom-static.com/downloads/firmware/"` in `ttwatch.c`

derickr, 2016-05-17:
That worked - cheers!

On a side note, I noticed that after the firmware upgrade, and the subsequent restart, the timezone was set wrong (also really, it had forgotten all the other config settings too). Another ``ttwatch -a`` fixed that. Perhaps a good idea to check for the firmware upgrades *first* with ``-a``? Or otherwise, rewrite the timezone to the watch again? (I can make a new issue for this if you'd like).

ryanbinns, 2016-05-17:
Interesting, I haven't noticed that happening before. The reason I do the firmware update last is because the firmware update resets the watch, which disconnects from the USB, and I didn't want to go to the hassle of adding code to reconnect to the watch after the reset is complete. It's probably mostly laziness, but it would be a major refactor of the code.

derickr, 2016-05-17:
Lazy is good :-) Perhaps a different exit code then? So that a wrapper script can just restart ttwatch if the exist code is something that indicates a firmware upgrade has happened?

ryanbinns, 2016-05-18:
I've uploaded a fix to make the firmware update work now. It dynamically finds the correct URL based on the watch's preferences file.

I'll have a look at a different exit code. There are already a few...

ryanbinns, 2016-06-10:
I've pushed a fix to return a new exit code (2) when the firmware is updated. If seems there wasn't a few anyway. So it returns 0 for no error, 1 for an error in input, 2 if firmware is updated. Reasonably simple.

closed 2016-06-10 by ryanbinns