ryanbinns/ttwatch #78: Segfault when doing factory-reset

Calimerorulez, 2016-05-19:
Hi,

I'm getting a segfault when factory-resetting the watch with the latest commit:

[24065.151348] ttwatch[19812]: segfault at 0 ip 00007f89c1cb563e sp 00007ffe2052ac70 error 4 in libc-2.23.so[7f89c1c28000+1c0000]
[24266.768666] ttwatch[19832]: segfault at 0 ip 00007fb5bc61963e sp 00007ffdf0d0cd40 error 4 in libc-2.23.so[7fb5bc58c000+1c0000]

Ubuntu 16.04

ryanbinns, 2016-05-19:
Interesting. I've just tried it on my watch and it worked fine. Are you able to run it under `gdb` and post a backtrace of where it segfaults?

Could it possibly be a package configuration issue? http://ubuntuforums.org/showthread.php?t=2010223 (older thread, but probably still relevant)

ahoogesteger, 2016-05-25:
This may actually be caused by the issue I fixed today in the `replace()` function. I suggest getting the latest version and see if you run into the same problem again.

ryanbinns, 2016-06-10:
Hi Calimerorulez. Have you been able to get this to work now?

johnzwarich, 2016-06-11:
I just downloaded and built today and am probably having the same problem:

john@scabbers:~/ttwatch-master$ ./ttwatch -v
Product ID:       0xed030000
BLE Version:      19
Firmware Version: 1.7.22
Unable to read watch name
john@scabbers:~/ttwatch-master$ ./ttwatch --factory-reset --factory-reset
Formatting watch ... Done
Segmentation fault (core dumped)

The watch has a picture of a battery on it and says tomtom.com/123 on the bottom line. When I unplug it, it shows me the picture of it connected to a computer with tomtom.com/reset underneath. I guess I have to go find a friend with a windows computer to try the reset with the official software.

ryanbinns, 2016-06-11:
Interesting. I'm glad you got it working but I'd like to know why it segfaulted. Have you tried doing a factory reset again?

johnzwarich, 2016-06-12:
Since everything is working at the moment, I'd rather not try it at this time.

closed 2016-10-05 by ryanbinns