ryanbinns/ttwatch #83: Debian Jessie install 3.16.0-4-amd64 device not attached

ramblingbarney, 2016-05-26:
Hi, I installed ttwatch but ttwatch --devices doesnt list the watch, my 99-tomtom.rules are below

ATTRS{idVendor}=="1390", ATTRS{idProduct}=="7474", SYMLINK+="tomtom", GROUP="plugdev", MODE="660"
ATTRS{idVendor}=="1390", ATTRS{idProduct}=="7475", SYMLINK+="tomtom", GROUP="plugdev", MODE="660"
ATTRS{idVendor}=="1390", ATTRS{idProduct}=="7477", SYMLINK+="tomtom", GROUP="plugdev", MODE="660"

I have asked the question on superuser here https://superuser.com/questions/1080920/tomtom-multisport-gps-watch-https-github-com-ryanbinns-ttwatch and posted the results of dmesg

ryanbinns, 2016-05-27:
Can you try running it using `sudo` and see if it works? That should tell you if your setup is correct. I noticed that your udedv rules file here and on the other site are different.

ramblingbarney, 2016-05-27:
Thanks, it works using sudo

ryanbinns, 2016-05-30:
Ok, if it works with `sudo` then it's probably a setup issue. I see that your rules above are missing the `SUBSYSTEM` argument. It should look something like this:
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1390", ATTRS{idProduct}=="7474", SYMLINK+="tomtom", GROUP="plugdev", MODE="660"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1390", ATTRS{idProduct}=="7475", SYMLINK+="tomtom", GROUP="plugdev", MODE="660"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1390", ATTRS{idProduct}=="7477", SYMLINK+="tomtom", GROUP="plugdev", MODE="660"
```
You'll need to tell udev to reload the rules using `udevadm control --reload-rules`. Also make sure that your user is part of the `plugdev` group:
```
sudo addgroup plugdev
sudo usermod -a -Gplugdev <user_name>
```
You'll then need to either log out/in again, or restart the machine to make the group change take effect. After this it should work ok.

Let me know how you go.

ramblingbarney, 2016-06-07:
adding the subsytems fixed the problem, watch identifed without sudo

ryanbinns, 2016-06-10:
Excellent, glad to know it worked.

closed 2016-06-10 by ryanbinns