ryanbinns/ttwatch #84: Firmware version not supported

adzane, 2016-05-31:
Hi, with my Runner 2, firmware 1.3.27, ttwatch --settings returns 'Firmware version not supported'. Any way to fix? 
Many thanks for this very useful work!

ryanbinns, 2016-06-01:
You can edit the `MANIFEST_DEFINITIONS` array at the top of `ttwatch.c` to add the new firmware version using the same details as the old one. You'll have to check to make sure that the values for all the settings make sense to be sure that the two versions are compatible though. I don't have a Runner 2, so I won't be able to do this check.

adzane, 2016-06-04:
I have added:

`{ 0x0001031b, MANIFEST_DEFINITION_00010113_COUNT, 
MANIFEST_DEFINITIONS_00010113 },`

and checked some random settings. It works, though the manifest file lacks new options ('splits', custom races for all activities, new languages) and should be updated.

ryanbinns, 2016-06-06:
I don't have a Runner 2 to be able to find the new settings. The way I found the existing settings was to change one setting at a time on the watch and check what changed in the manifest file (0x85000000). If you can spend some time doing that, it would be very helpful. If not, that's fine.

adzane, 2016-06-08:
I would be glad to help, Ryan, but could you explain me better how to do 
this task, as I'm not so skilled...

ryanbinns, 2016-06-10:
Ok. It's a little complication, but hopefully I can explain well enough. You will need to compile with the unsafe functions enabled (`cmake -Dunsafe=on`)

1. Save a copy of the manifest file using `ttwatch --read 0x00850000 manifest1.bin`.
2. Change one (and only one) setting on the watch.
3. Save a new copy of the manifest file using `ttwatch --read 0x00850000 manifest2.bin`.
4. Use a hex editor to find the difference between the two files. Each entry in the file is four-bytes long (a single 32-bit value), preceded by a 16-bit entry number.
5. If the setting has multiple values that can be chosen from a list, repeat steps 2 to 4 to find out which value is stored in the file for every possible value of that setting. So we should have an entry number and a list of valid values.
6. If the setting has a arbitrary value you can enter (such as metres per lap), then I need to know how the value is encoded. So is it in metres, feet, seconds, hours etc... And is it an integer or a 32-bit floating point number. A mapping between the value you input and what is put in the file is necessary, since some values are input in km/h but stored in the file in m/s.
7. Repeat all the steps for every new setting.

It's all a bit complicated, but once you get the hang of it, it's fairly easy to repeat. I use a hex editor (010 Editor) that can show me differences graphically and also has user-defined templates that decode the file structure which makes it a LOT easier to see the internal file structure. But it's definitely possible to do without. If you don't feel confident to do it, that's fine.

adzane, 2016-06-11:
OK, I will try. Thanks for your support!

closed 2016-08-03 by adzane