ryanbinns/ttwatch #85: Runner watch firmware not supported

sjquinney, 2016-06-08:
I've been using ttwatch successfully for about 6 months now with my Runner watch (Product ID 0xe9030000) but after having upgraded the firmware to the latest new version - 1.8.52 - I now just get "Firmware version not supported".

fake-fur, 2016-06-09:
I'm also having this issue but cannot find enough information from the source files to figure out the fix

ryanbinns, 2016-06-10:
Hi guys,
I have pushed a fix to add support for this firmware version. I hadn't yet had a chance to see if the settings changed between the previous version and this one. They don't seem to have done. If you want to update it yourself in future, the definitions are at the top of `ttwatch.c`, in the `MANIFEST_DEFINITIONS` array.

closed 2016-06-10 by ryanbinns