ryanbinns/ttwatch #86: Update GPS aborted, update firmware Segmentation fault

benlou, 2016-06-10:
Hi,
When running --update-gps I get the below error
> *** Error in `ttwatch': free(): invalid next size (fast): 0x0000000001434b10 ***
> Aborted

When running --update-fw I get
> Segmentation fault
Wondering where the issue could be...

Just got my watch yesterday, still on firmware 1.1.19 so badly needs a fw update, but cannot get it to work...

Thanks for your answers!

ryanbinns, 2016-06-10:
If you've never plugged your watch in to a Windows machine, you'll need to run `ttwatch` with the `--initial-setup` option first. This will create the default preferences file and other data files on the watch, which the other options rely on.

Before you do that though, I'd appreciate it if you could run the program using `gdb` and when it aborts give me the stack trace. That way I can see where it's failing and put in more checks.

benlou, 2016-06-10:
I have already run the `--initial-setup` command before posting above...
gdb output is as below

> (gdb) run -a
> Starting program: /usr/local/bin/ttwatch -a
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> [New Thread 0x7ffff1363700 (LWP 16911)]
> [New Thread 0x7ffff0b62700 (LWP 16915)]
> [Thread 0x7ffff0b62700 (LWP 16915) exited]
> *** Error in `/usr/local/bin/ttwatch': free(): invalid next size (fast): 0x0000000000668b30 ***
> 
> Program received signal SIGABRT, Aborted.
> 0x00007ffff6dadc37 in __GI_raise (sig=sig@entry=6)
>     at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
> 56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
> (gdb) backtrace
> #0  0x00007ffff6dadc37 in __GI_raise (sig=sig@entry=6)
>     at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
> #1  0x00007ffff6db1028 in __GI_abort () at abort.c:89
> #2  0x00007ffff6dea2a4 in __libc_message (do_abort=do_abort@entry=1, 
>     fmt=fmt@entry=0x7ffff6ef86b0 "*** Error in `%s': %s: 0x%s ***\n")
>     at ../sysdeps/posix/libc_fatal.c:175
> #3  0x00007ffff6df655e in malloc_printerr (ptr=<optimized out>, 
>     str=0x7ffff6ef8850 "free(): invalid next size (fast)", action=1)
>     at malloc.c:4996
> #4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0)
>     at malloc.c:3840
> #5  0x000000000040e909 in replace ()
> #6  0x000000000040f490 in do_update_gps ()
> #7  0x000000000040a2ae in main ()
>
Thanks!

ryanbinns, 2016-06-10:
A (hopefully) solved mystery! There was a serious bug in the `replace()` function that was fixed a few weeks ago, but somehow it became broken again. I may have pushed a new commit without pulling the latest update that included the fix (submitted by someone else). Anyway, I have re-fixed it and hopefully it will work for you now.

benlou, 2016-06-10:
Excellent!
Downloaded the latest version and all worked very well. Firmware updated and GPS fix downloaded.
Thanks for your prompt support!

ryanbinns, 2016-06-10:
No worries. Glad it works for you!

closed 2016-06-10 by ryanbinns