ryanbinns/ttwatch #87: Building with gcc 5.4

sjquinney, 2016-06-10:
I've just tried to build the latest ttwatch on Debian unstable which has gcc 5.4, it appears that it fails to build due to missing declarations for some functions. 

I've attached a patch which adds all the missing includes for the various headers. I've only tested this on Debian so I can't guarantee it is all correct for other platforms.

[0001-Fix-various-warnings-about-implicit-declaration-of-f.patch.txt](https://github.com/ryanbinns/ttwatch/files/309018/0001-Fix-various-warnings-about-implicit-declaration-of-f.patch.txt)

ryanbinns, 2016-06-11:
Thanks for the patch. I have fixed the code. It would be easier if you submitted a pull request next time though.

closed 2016-06-11 by ryanbinns

sjquinney, 2016-06-11:
Thanks, I've not really used git or github that much so I'm still getting
my head around how it all works. It's a shame that github doesn't make it
easy to submit small patch files.

Stephen

ryanbinns, 2016-06-11:
It's actually not too difficult. If you checkout the code then make whatever changes you like and check in again, you can click on "New Pull Request" and select the branches you want to compare and GitHub will compare the two and create a diff in a way that makes it simple to merge. [See here](https://help.github.com/articles/creating-a-pull-request/)