ryanbinns/ttwatch #88: elevation gain

stbland, 2016-06-13:
Display (or calculate) the elevation gain for each recorded track will be a nice enhancement.

ryanbinns, 2016-06-14:
It would, except the elevation data is not particularly accurate. It only has a resolution of 1m, and doesn't know anything about bridges or tunnels or anything man-made. The elevation data is downloaded from TomTom's web server, and is provided from a digital elevation model service after uploading the latitude/longitude pairs. The watch itself does not log elevation data. It doesn't cover regions close to the poles (several people in Scandinavia have report that there is no coverage for their location), and one person that I know of has reported changes in elevation of tens of meters while kayaking on a flat river.

I don't think there is much point in adding this without having an accurate source of elevation data.

stbland, 2016-06-14:
My idea was not to upload any personal data to TomTom's web servers.
I was thinking that all triplets (latitude, longitude, altitude) was stored in the watch.
It could be possible to calculate the elevation gain by using a web service like Google Elevation ( https://developers.google.com/maps/documentation/elevation/ ).

ryanbinns, 2016-06-15:
Yeah, it would be nice to not have to upload it. Unfortunately, there are no offline digital elevation models available that I know of. You may be able to get around it by creating a KML file without elevation data, importing into Google Earth, then exporting again; it may add elevation data, but I don't know - and it would probably query the elevation from Google's servers anyway. Personally, I would trust TomTom more than Google with my data, but that's just me. I thought about using Google's elevation model, but they require you to use an API key and I wasn't willing to expose my API key in an open-source project for anyone to use, so everyone who used it would need to request their own API key, which didn't seem very useful to me.

kingosticks, 2016-06-15:
Would something like https://github.com/tkrajina/srtm.py be any use?

ryanbinns, 2016-06-17:
Possibly. It's still only got a 1m resolution, so there would need to be some filtering involved. It's got less coverage than TomTom's web service, so I think sticking to that would be better. If someone wants to put in the effort to include this, I'm happy to include it.

closed 2016-10-05 by ryanbinns