ryanbinns/ttwatch #91: "Unable to open watch"

jamestombs, 2016-07-05:
Followed the whole README and got ttwatch set up.

Running ttwatch --devices brings up:
```
ttwatch --devices
HC6186G01815:
```
Trying to subsequently do anything with the watch results in a "Unable to open watch"

```
ttwatch --device=HC6186G01815 -a
Unable to open watch
```

lsusb results in `Bus 002 Device 006: ID 1390:7474 TOMTOM B.V. GPS Sport Watch [Runner, Multi-Sport]`

Doesn't work with the user or root.

lsusb -v results in:

```
Bus 002 Device 006: ID 1390:7474 TOMTOM B.V. GPS Sport Watch [Runner, Multi-Sport]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1390 TOMTOM B.V.
  idProduct          0x7474 GPS Sport Watch [Runner, Multi-Sport]
  bcdDevice            2.00
  iManufacturer           1 TomTom
  iProduct                2 TomTom GPS Watch
  iSerial                 3 HC6186G01815
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 Workout Drive
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              4 Workout Drive
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      36
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

ryanbinns, 2016-07-06:
Have you tried running `ttwatch -a` (without specifying a specific device)? It will try to open the first device it finds. It won't list the device unless it is able to open it properly, so it may be a problem with the device filtering code.

jamestombs, 2016-07-06:
Yes tried without specifying the device and got the same message.

ryanbinns, 2016-07-06:
That's bizarre. The `ttwatch_enumerate_devices` (used by `--devices`) and `ttwatch_open` (used by everything else) functions both delegate to the `ttwatch_open_device` function, so the watch opening code is identical for each. If one works, the other should too.

Are you able to step through the `ttwatch_open_device` function using `gdb` and see where it returns a failure?

jamestombs, 2016-07-06:
Don't have access to the PC at the moment, can you give instruction on how I'd do that?

ryanbinns, 2016-07-06:
If you haven't used `gdb` before it can be a little trickyto understand, but I'll try. Make sure you've got the latest version of the code so the line numbers match up.

Compile in debug mode (assuming you're building in a `build` subdirectory of the source)
```
cd build
cmake -DCMAKE_BUILD_TYPE=Debug ..
make
```
Start `gdb` and set a breakpoint for the right function
```
gdb ttwatch
break libttwatch.cpp:361
```
Start the `ttwatch` program running with the `-a` option
```
run -a
```
When `gdb` stops at the breakpoint, continually type `n` and (Enter) to tell `gdb` to execute the next line of code. When you get to a line with a return statement note the line number (left-hand side of the screen). That will at least tell us where in the code the problem is.

You can quit gdb using the `quit` command.

Hopefully this is clear enough. If you can post the full output of your `gdb` session that will help too.

jamestombs, 2016-07-06:
Output attached. Note that this is a different watch to before. The other watch was a brand new one, this is an old one although they're the same make and model.

```
(gdb) break libttwatch.cpp:361
Breakpoint 1 at 0x40fcdd: file /home/lyn/tomtom/src/libttwatch.cpp, line 361.
(gdb) run -a
Starting program: /home/lyn/tomtom/build/ttwatch -a
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1363700 (LWP 5977)]

Breakpoint 1, ttwatch_open_device (device=0x637720, 
    serial_or_name=0x6343c0 "HC6186G01815", watch=0x7fffffffdae8)
    at /home/lyn/tomtom/src/libttwatch.cpp:361
361	    if (libusb_open(device, &handle))
(gdb) n
367	    *watch = (TTWATCH*)malloc(sizeof(TTWATCH));
(gdb) n
368	    memset(*watch, 0, sizeof(TTWATCH));
(gdb) n
369	    (*watch)->device = handle;
(gdb) n
370	    (*watch)->usb_product_id = desc.idProduct;
(gdb) n
374	    while (attempts++ < 60)
(gdb) n
377	        if (libusb_kernel_driver_active(handle, 0))
(gdb) n
379	            if (!libusb_detach_kernel_driver(handle, 0))
(gdb) n
380	                (*watch)->attach_kernel_driver = 1;
(gdb) n
382	        if ((result = libusb_claim_interface(handle, 0)) != 0)
(gdb) n
394	            break;
(gdb) n
397	    if (result)
(gdb) n
406	        (unsigned char*)(*watch)->serial_number, sizeof((*watch)->serial_number));
(gdb) n
407	    if (count > 0)
(gdb) n
408	        ((char*)(*watch)->serial_number)[count] = 0;
(gdb) n
410	    RETURN_ERROR(ttwatch_send_startup_message_group(*watch));
(gdb) n
413	    if (ttwatch_get_watch_name(*watch, name, sizeof(name)) != TTWATCH_NoError)
(gdb) n
417	    if (!serial_or_name ||
(gdb) n
418	        (strcasecmp(serial_or_name, serial) == 0) ||
(gdb) n
417	    if (!serial_or_name ||
(gdb) n
419	        (strcasecmp(serial_or_name, name) == 0))
(gdb) n
418	        (strcasecmp(serial_or_name, serial) == 0) ||
(gdb) n
429	        ttwatch_close(*watch);
(gdb) n
430	        *watch = 0;
(gdb) n
431	        return TTWATCH_NoMatchingWatch;
(gdb) n
433	}
(gdb) 
ttwatch_open (serial_or_name=0x6343c0 "HC6186G01815", watch=0x7fffffffdae8)
    at /home/lyn/tomtom/src/libttwatch.cpp:326
326	    for (ssize_t i = 0; i < count; ++i)
(gdb) n
328	        if (ttwatch_open_device(list[i], serial_or_name, watch) == TTWATCH_NoError)
(gdb) n
326	    for (ssize_t i = 0; i < count; ++i)
(gdb) n
328	        if (ttwatch_open_device(list[i], serial_or_name, watch) == TTWATCH_NoError)
(gdb) n
326	    for (ssize_t i = 0; i < count; ++i)
(gdb) n
328	        if (ttwatch_open_device(list[i], serial_or_name, watch) == TTWATCH_NoError)
(gdb) n
326	    for (ssize_t i = 0; i < count; ++i)
(gdb) n
328	        if (ttwatch_open_device(list[i], serial_or_name, watch) == TTWATCH_NoError)
(gdb) n
326	    for (ssize_t i = 0; i < count; ++i)
(gdb) n
332	    libusb_free_device_list(list, 1);
(gdb) n
333	    return *watch ? TTWATCH_NoError : TTWATCH_NoMatchingWatch;
(gdb) n
334	}
(gdb) n
main (argc=2, argv=0x7fffffffdfd8) at /home/lyn/tomtom/src/ttwatch.c:1616
1616	        write_log(1, "Unable to open watch\n");
(gdb) n
Unable to open watch
1617	        free_options(options);
(gdb) n
1618	        return 1;
(gdb) n
1785	}
(gdb) n
__libc_start_main (main=0x4094bc <main>, argc=2, argv=0x7fffffffdfd8, 
    init=<optimised out>, fini=<optimised out>, rtld_fini=<optimised out>, 
    stack_end=0x7fffffffdfc8) at libc-start.c:321
321	libc-start.c: No such file or directory.
```

ryanbinns, 2016-07-07:
Interesting, the program is looking for a specific watch even though you're not specifying it on the command line. Do you have a config file in `/etc/ttwatch.conf` or `~/.ttwatch` that includes a `Device` line?

jamestombs, 2016-07-07:
I did still have that, I've commented it out and I now no longer get the "Unable to open watch" error but get a "Segmentation fault (core dumped)" for any commands. I've rebuilt without using the debug line and get the same issue.

ryanbinns, 2016-07-07:
Ok, well at least it's a different error. I fixed a bug yesterday that will cause a segmentation fault for a new watch or a watch with the XML preferences file missing. If you haven't updated your code tree since then, I suggest you try that.

For a new watch, to be able to do anything, you will need to run `ttwatch --inital-setup` before you can run any other commands. This command will create a default XML preferences file and create the default races for all the different activity types. You should then set a watch name using `ttwatch --set-name <watch_name>`.

jamestombs, 2016-07-07:
All appears to be working now, thanks very much.

Might be worth adding to the README about new watches having to be set up, I missed that from the help options as it was buried with all the other options.

The other thing was the ttbin2mysports script looks for a folder structure which isn't created by default. It's looking for ~/ttwatch/{DEVICE_ID}/* but the default output of the ttwatch program is to put everything directly in to ~/ttwatch/*.

Nothing has appear on MapMyFitness though, I'm hoping that this is just a delay on their end as the script says it's imported.

```
./ttbin2mysports -v
Treadmill_14-26-17.tcx has already been uploaded from 2016-07-02
Treadmill_01-11-24.tcx has already been uploaded from 2016-07-03
Running_07-16-10.tcx has already been uploaded from 2016-07-06
Running_17-52-24.tcx has already been uploaded from 2016-07-06
Running_18-02-25.tcx has already been uploaded from 2016-07-06
Running_18-03-20.tcx has already been uploaded from 2016-07-06
Running_18-04-19.tcx has already been uploaded from 2016-07-06
```

closed 2016-07-07 by jamestombs

ryanbinns, 2016-07-08:
The program has a folder structure of `~/ttwatch/<watch name>/<date>/*.ttbin`. If the watch name is not set, then that part of the path doesn't exist. The `ttbin2mysports` script was written by someone else and I've never used it, and probably never will. I've never heard of anyone else having issues with it, but I don't know how many people actually use it. It assumes that the watch name has been set, which would be true for most people and it therefore works for multiple watches stored in the same activity store location.

ryanbinns, 2016-07-08:
I'll add a comment in the `README` about setting up watches for the first time.