ryanbinns/ttwatch #92: No activities downloaded from Runner 2

Unicorn-rzl, 2016-07-12:
I now own a Runner 2 Cardio and want to use ttwatch instead of the TomTom bloatware. The watch is recognized and I was able to set it up correctly (initial setup, setting a name for the watch, etc).

The only thing currently not working is getting the activities from the watch, which would be very nice....

Here is what happens:
<pre>ttwatch -v
Product ID:       0xd8070000
BLE Version:      0
Firmware Version: 1.3.27
Watch Name:       Drecksding
Serial Number:    HNxxxxxxxxx</pre>

<pre>ttwatch --list-history
Treadmill:
1: 2016/07/12 14:45:55,  244s,    76.76m,   14 calories
Cycling:
1: 2016/07/12 12:04:31,  499s,  1441.18m,   72 calories
</pre>

<pre>ttwatch -a
Downloading GPSQuickFix data file...
Writing file to watch...
GPSQuickFix data updated
Current firmware is already at latest version</pre>

<pre>
ttwatch --get-activities -s /home/unicorn/ttwatch/foo/
ls /home/unicorn/ttwatch/foo/
(empty)
</pre>

<pre>
ttwatch --list-history
Treadmill:
1: 2016/07/12 14:45:55,  244s,    76.76m,   14 calories
Cycling:
1: 2016/07/12 12:04:31,  499s,  1441.18m,   72 calories
</pre>

I'm using Linux Mint 17.3
<pre>uname -a
Linux 4.4.0-28-generic #47~14.04.1-Ubuntu SMP Fri Jun 24 16:30:35 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
lsb_release -a
No LSB modules are available.
Distributor ID:	LinuxMint
Description:	Linux Mint 17.3 Rosa
Release:	17.3
Codename:	rosa
</pre>

Before I dig into your code, I wanted to ask whether there is any way to get more Information why the activities are not downloaded. Can somebody help me with this?

Please let me know if there are relevant information missing. Thank you.

Here is a package dump, I don't know whether this gives useful information:
<pre>
ttwatch --get-activities --packets -s /home/unicorn/ttwatch/foo/

01 16 00 0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

01 02 01 0A 

01 03 02 22 00 

01 03 03 22 00 

01 06 04 20 D8 07 00 00 

01 02 05 0A 

[... 272 lines cut ...]
01 16 8E 12 00 00 00 00 00 83 00 02 00 00 00 00 00 00 00 54 00 00 00 00 

01 16 8F 12 00 00 00 00 00 B1 00 03 00 00 00 00 00 00 06 30 00 00 00 00 

01 16 90 12 00 00 00 00 00 B1 00 04 00 00 00 00 00 00 00 90 00 00 00 00 

01 16 91 12 00 00 00 00 00 B1 00 05 00 00 00 00 00 00 01 36 00 00 00 00 

01 16 92 12 00 00 00 00 00 B1 00 08 00 00 00 00 00 00 00 95 00 00 00 00 

01 16 93 12 00 00 00 00 00 72 07 01 00 00 00 00 00 00 00 26 00 00 00 00 

01 16 94 12 00 00 00 00 00 73 07 01 00 00 00 00 00 00 00 2D 00 00 00 00 

01 16 95 12 00 00 00 00 00 B1 00 07 00 00 00 00 00 00 01 C8 00 00 00 00 

01 16 96 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 

01 06 97 1A 00 00 11 18 

01 02 98 0A 

01 05 99 23 64 01 00 

01 05 9A 23 64 01 00 </pre>

ryanbinns, 2016-07-12:
Based on that packet dump, your watch doesn't actually have any activities on it. They may have been downloaded previously. The history items are separate from the activity data. The history items only store enough information to allow the race function to work and show history information on the watch itself. They do not contain the entire logged data.

Once the activities are downloaded, they are deleted from the watch to save storage space, so they can only be downloaded once. If your watch was attached to the computer using the TomTom software after these activities were created, then the activity data is no longer on the watch and cannot be downloaded again.

Unicorn-rzl, 2016-07-12:
Thank you for the really fast answer!
I've created a new activity on the watch and did:
<pre>
ttwatch --get-activities -s /home/unicorn/ttwatch/foo/
tree /home/unicorn/ttwatch/foo/
/home/unicorn/ttwatch/foo/
└── Drecksding
    └── 2016-07-12
        ├── Treadmill_16-03-20.tcx
        └── Treadmill_16-03-20.ttbin

2 directories, 2 files
</pre>
Hooray! It works!

Then I did another "threadmill" session on my desk and tried the -a option:
<pre>
strace -e trace=write ttwatch -a
write(7, "\1", 1)                       = 1
write(12, " \t\0\1\3\33\330\7\23\26\205W\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0005xp_"..., 1170) = 1170
write(12, "<?xml version=\"1.0\" encoding=\"UT"..., 4096) = 4096
write(12, "        </Extensions>\r\n         "..., 4096) = 4096
write(12, "ion/v2\">\r\n                      "..., 4096) = 4096
write(12, "                        <Extensi"..., 4096) = 4096
write(12, "                   <Time>2016-07"..., 3056) = 3056
write(12, "\26\3\1\1(\1\0\1$\3\3^\0231\2735\245\361E\5JQ\26D\20\235\245\242\334IG\20"..., 301) = 301
write(12, "\26\3\3\1\6\20\0\1\2\1\0R\325\210\372\31\320Y\304\0\366\21\36\362\36&\364\256{\20\21\312"..., 342) = 342
write(12, "\27\3\3\0\240R\221b\226?\35w\274:\330'iK\201E\376\376\252\37\330\26\260\222G\300[x"..., 165) = 165
write(12, "\25\3\3\0000\20V\36\r\222dgt\3120z\1\0161\22\333E~j\315r@\21\331(\240\366"..., 53) = 53
write(1, "Downloading GPSQuickFix data fil"..., 37Downloading GPSQuickFix data file...
) = 37
write(12, "\26\3\1\0014\1\0\0010\3\3\30\263\366\211\35\241vY\212\v\30\217>-\326\305\f\256\370^\244"..., 313) = 313
write(12, "\26\3\3\1\6\20\0\1\2\1\0\207\320/4K\36\4[\30\322\313\240\320\26lQ_\355 \234\216"..., 358) = 358
write(12, "\27\3\3\0\260j\237O\"\223Ou\231`k\16\243{\346+\304Cj\270\357\275\207\327\4\177\374\255"..., 181) = 181
write(12, "\25\3\3\0@\"\26\200\0177\271\355\340\\J\327.m\251IC\353\226\201\266V\253\277\257v\334G"..., 69) = 69
write(1, "Writing file to watch...\n", 25Writing file to watch...
) = 25
write(1, "GPSQuickFix data updated\n", 25GPSQuickFix data updated
) = 25
write(12, "\26\3\1\1(\1\0\1$\3\3Xh\351$:[ \27\332\365\306\32eS\3\247 \202%i,"..., 301) = 301
write(12, "\26\3\3\1\6\20\0\1\2\1\0\314\\W\22#\236\256\302N\245O/\24\374\233L\f\372\313\21\334"..., 342) = 342
write(12, "\27\3\3\0\240\264J[e\271\234F\2\221\303\20&T\320?\31#\364\200E\357\212\270\352R\375\255"..., 165) = 165
write(12, "\25\3\3\0000yTa\225+\26\206\4\242\325\220\250\32x}H\20a\350\354\251|\25\207\352\230\35"..., 53) = 53
write(12, "\26\3\1\1/\1\0\1+\3\3\240\306\232r\253\35\16\206\200&c\270\376\247\225\226\343\16\333+\334"..., 308) = 308
write(12, "\26\3\3\0F\20\0\0BA\4Bi\337\242\200\36\227\20\367CH\16\324\276\214\336_./\305\315"..., 126) = 126
write(12, "\27\3\3\0\251%\221\375\341\266^\25\341?RM\26\363\10\2105e\2 \200\333\356\242\"F\242\375"..., 174) = 174
write(12, "\25\3\3\0\32%\221\375\341\266^\25\342T\0164\340e\354\316\234\362\363\276(\312\260\357\ro~", 31) = 31
write(2, "Current firmware is already at l"..., 46Current firmware is already at latest version
) = 46
write(7, "\1", 1)                       = 1
write(5, "\1", 1)                       = 1
+++ exited with 0 +++
</pre>
<pre>
ttwatch --get-activities -s /home/unicorn/ttwatch/foo/
UniPad unicorn # tree /home/unicorn/ttwatch/foo/
/home/unicorn/ttwatch/foo/
└── Drecksding
    └── 2016-07-12
        ├── Treadmill_16-03-20.tcx
        └── Treadmill_16-03-20.ttbin

2 directories, 2 files
</pre>

So, the -a seems to delete the activities from my watch without saving them?

Unicorn-rzl, 2016-07-12:
Ok, I was able to find my mistake. ttwatch did download the data, but since I've used a root session, the data got stored in my /root and I failed to look there.
Sorry for the noise, hopefully I can act as a bad example for others who make the same mistake.

closed 2016-07-12 by Unicorn-rzl

ryanbinns, 2016-07-12:
No worries, I'm glad it's all working now.