ryanbinns/ttwatch #93: make fails debian jessie AMD64  (in-source & out-of-source)

ramblingbarney, 2016-07-14:
Any help is appreciated to further diagnose the problem 

```
make
Scanning dependencies of target libttbin
[  3%] Building C object CMakeFiles/libttbin.dir/src/export_csv.c.o
[  6%] Building C object CMakeFiles/libttbin.dir/src/export_gpx.c.o
[  9%] Building C object CMakeFiles/libttbin.dir/src/export_kml.c.o
[ 12%] Building C object CMakeFiles/libttbin.dir/src/export_tcx.c.o
[ 15%] Building C object CMakeFiles/libttbin.dir/src/ttbin.c.o
Linking C static library libttbin.a
[ 15%] Built target libttbin
Scanning dependencies of target libttwatch
[ 18%] Building CXX object CMakeFiles/libttwatch.dir/src/libttwatch.cpp.o
[ 21%] Building CXX object CMakeFiles/libttwatch.dir/src/libttwatch_cpp.cpp.o
Linking CXX static library libttwatch.a
[ 21%] Built target libttwatch
Scanning dependencies of target ttbincnv
[ 25%] Building C object CMakeFiles/ttbincnv.dir/src/ttbincnv.c.o
Linking C executable ttbincnv
[ 25%] Built target ttbincnv
Scanning dependencies of target ttbinmod
[ 28%] Building C object CMakeFiles/ttbinmod.dir/src/ttbinmod.c.o
Linking C executable ttbinmod
[ 28%] Built target ttbinmod
Scanning dependencies of target manifest
[ 31%] Generating manifest_definitions.h
given is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 39.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 41.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 54.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 67.
[ 34%] Generating manifest_definitions_00010113.h
given is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 39.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 41.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 54.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 67.
[ 37%] Generating manifest_definitions_0001082e.h
given is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 39.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 41.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 54.
when is experimental at /home/conor/ttwatch/manifest/make_manifest.pl line 67.
[ 37%] Built target manifest
Scanning dependencies of target ttwatch
[ 40%] Building C object CMakeFiles/ttwatch.dir/src/ttwatch.c.o
[ 43%] Building C object CMakeFiles/ttwatch.dir/src/log.c.o
[ 46%] Building C object CMakeFiles/ttwatch.dir/src/options.c.o
[ 50%] Building C object CMakeFiles/ttwatch.dir/src/json.c.o
[ 53%] Building C object CMakeFiles/ttwatch.dir/src/download.c.o
[ 56%] Building C object CMakeFiles/ttwatch.dir/src/firmware.c.o
[ 59%] Building C object CMakeFiles/ttwatch.dir/src/misc.c.o
[ 62%] Building C object CMakeFiles/ttwatch.dir/src/get_activities.c.o
[ 65%] Building C object CMakeFiles/ttwatch.dir/src/update_gps.c.o
[ 68%] Building C object CMakeFiles/ttwatch.dir/src/set_time.c.o
Linking CXX executable ttwatch
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_CTX_use_certificate_chain_file@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_CTX_set_verify@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `ASN1_STRING_type@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `CRYPTO_free@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_get_shutdown@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_get_session@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_get_verify_result@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `d2i_PKCS12_fp@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `UI_method_set_opener@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `EVP_cleanup@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_write@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_peek@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `BIO_new@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_set_fd@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_get_privatekey@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `BN_num_bits@OPENSSL_1.0.0'
[... 118 lines cut ...]
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `ASN1_STRING_data@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_CTX_set_srp_username@OPENSSL_1.0.1'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `RAND_load_file@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_CTX_load_verify_locations@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `i2t_ASN1_OBJECT@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `ERR_remove_thread_state@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `RAND_bytes@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `MD5_Update@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `UI_method_get_writer@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `SSL_CIPHER_get_name@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `X509_EXTENSION_get_critical@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `PKCS12_free@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `sk_num@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `ERR_error_string@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `X509_NAME_print_ex@OPENSSL_1.0.0'
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `UI_method_set_writer@OPENSSL_1.0.0'
collect2: error: ld returned 1 exit status
CMakeFiles/ttwatch.dir/build.make:316: recipe for target 'ttwatch' failed
make[2]: *** [ttwatch] Error 1
CMakeFiles/Makefile2:233: recipe for target 'CMakeFiles/ttwatch.dir/all' failed
make[1]: *** [CMakeFiles/ttwatch.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

```

ryanbinns, 2016-07-14:
It looks like you don't have OpenSSL installed. Try running "apt-get install openssl openssl-dev".

ramblingbarney, 2016-07-14:
thanks for your comment, openssl, libssl-dev, libcurl4-openssl-dev are installed, I ran this to find missing libraries but the below output indicates no missing libraries

```
sudo ldd /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so
	linux-vdso.so.1 (0x00007fffc778c000)
	libidn.so.11 => /usr/lib/x86_64-linux-gnu/libidn.so.11 (0x00007f2dd5c83000)
	librtmp.so.1 => /usr/lib/x86_64-linux-gnu/librtmp.so.1 (0x00007f2dd5a65000)
	libssh2.so.1 => /usr/lib/x86_64-linux-gnu/libssh2.so.1 (0x00007f2dd583c000)
	libssl.so.1.0.0 => /usr/lib/x86_64-linux-gnu/libssl.so.1.0.0 (0x00007f2dd55db000)
	libcrypto.so.1.0.0 => /usr/lib/x86_64-linux-gnu/libcrypto.so.1.0.0 (0x00007f2dd51df000)
	libgssapi_krb5.so.2 => /usr/lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007f2dd4f94000)
	libkrb5.so.3 => /usr/lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007f2dd4cc0000)
	libk5crypto.so.3 => /usr/lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f2dd4a8f000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f2dd488b000)
	liblber-2.4.so.2 => /usr/lib/x86_64-linux-gnu/liblber-2.4.so.2 (0x00007f2dd467c000)
	libldap_r-2.4.so.2 => /usr/lib/x86_64-linux-gnu/libldap_r-2.4.so.2 (0x00007f2dd442a000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f2dd420f000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2dd3ff2000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2dd3c47000)
	libgnutls-deb0.so.28 => /usr/lib/x86_64-linux-gnu/libgnutls-deb0.so.28 (0x00007f2dd3928000)
	libhogweed.so.2 => /usr/lib/x86_64-linux-gnu/libhogweed.so.2 (0x00007f2dd36f9000)
	libnettle.so.4 => /usr/lib/x86_64-linux-gnu/libnettle.so.4 (0x00007f2dd34c7000)
	libgmp.so.10 => /usr/lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f2dd3244000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f2dd2f62000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2dd2d5e000)
	libkrb5support.so.0 => /usr/lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f2dd2b52000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f2dd294e000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f2dd2737000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2dd612b000)
	libsasl2.so.2 => /usr/lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007f2dd251b000)
	libp11-kit.so.0 => /usr/lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007f2dd22d5000)
	libtasn1.so.6 => /usr/lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007f2dd20c1000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f2dd1eaf000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f2dd1ca7000)

```

ryanbinns, 2016-07-15:
Oh dear. That does look like openssl is installed properly. Every undefined reference was in the openssl library though, so it still looks like an installation issue with openssl.

Can you post the contents of CMakeCache.txt from your build location? It lists the found locations of all the libraries, which may give a clue.

ramblingbarney, 2016-07-15:
File attached, your help is appreciated, the openssl library found is a local version installed via linuxbrew

[CMakeCache.txt](https://github.com/ryanbinns/ttwatch/files/365755/CMakeCache.txt)

ryanbinns, 2016-07-15:
Interesting. It may have something to do with Linuxbrew. But it may not. Enabling verbose linker output might give more info. Add the following line after line 74 in CMakeLists.txt then rerun cmake and make:
```
set_target_properties(ttwatch LINK_FLAGS -Wl,-v)
```
You'll need to run `make VERBOSE=1` otherwise cmake will hide the output. It will print a LOT of data...

ramblingbarney, 2016-07-15:
Thanks, with your above line at line 77 in CMakeLists.txt I get this error

```
CMake Error at CMakeLists.txt:77 (set_target_properties):
  set_target_properties called with illegal arguments, maybe missing a
  PROPERTIES specifier?

```

ryanbinns, 2016-07-15:
Oh bugger. Yes. The word PROPERTIES needs to be inserted before LINK_FLAGS. Sorry about that.

ryanbinns, 2016-07-16:
Oh. `-Wl,-v` only enables part of the debug output (the exact command line passed to `ld`). To get the library information, it will need to be changed to `-Wl,--verbose`. So the entire line should be:
```
set_target_properties(ttwatch PROPERTIES LINK_FLAGS -Wl,--verbose)
```
I've just tested it, and it did what it was supposed to do.

ramblingbarney, 2016-07-18:
thanks, output below

```
make VERBOSE=1
/usr/bin/cmake -H/home/conor/ttwatch -B/home/conor/ttwatch/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/conor/ttwatch/build/CMakeFiles /home/conor/ttwatch/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/conor/ttwatch/build'
make -f CMakeFiles/libttbin.dir/build.make CMakeFiles/libttbin.dir/depend
make[2]: Entering directory '/home/conor/ttwatch/build'
cd /home/conor/ttwatch/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/conor/ttwatch /home/conor/ttwatch /home/conor/ttwatch/build /home/conor/ttwatch/build /home/conor/ttwatch/build/CMakeFiles/libttbin.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/conor/ttwatch/build'
make -f CMakeFiles/libttbin.dir/build.make CMakeFiles/libttbin.dir/build
make[2]: Entering directory '/home/conor/ttwatch/build'
make[2]: Nothing to be done for 'CMakeFiles/libttbin.dir/build'.
make[2]: Leaving directory '/home/conor/ttwatch/build'
/usr/bin/cmake -E cmake_progress_report /home/conor/ttwatch/build/CMakeFiles  1 2 3 4 5
[ 15%] Built target libttbin
make -f CMakeFiles/libttwatch.dir/build.make CMakeFiles/libttwatch.dir/depend
make[2]: Entering directory '/home/conor/ttwatch/build'
cd /home/conor/ttwatch/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/conor/ttwatch /home/conor/ttwatch /home/conor/ttwatch/build /home/conor/ttwatch/build /home/conor/ttwatch/build/CMakeFiles/libttwatch.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/conor/ttwatch/build'
make -f CMakeFiles/libttwatch.dir/build.make CMakeFiles/libttwatch.dir/build
make[2]: Entering directory '/home/conor/ttwatch/build'
make[2]: Nothing to be done for 'CMakeFiles/libttwatch.dir/build'.
make[2]: Leaving directory '/home/conor/ttwatch/build'
/usr/bin/cmake -E cmake_progress_report /home/conor/ttwatch/build/CMakeFiles  6 7
[ 21%] Built target libttwatch
make -f CMakeFiles/ttbincnv.dir/build.make CMakeFiles/ttbincnv.dir/depend
make[2]: Entering directory '/home/conor/ttwatch/build'
cd /home/conor/ttwatch/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/conor/ttwatch /home/conor/ttwatch /home/conor/ttwatch/build /home/conor/ttwatch/build /home/conor/ttwatch/build/CMakeFiles/ttbincnv.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/conor/ttwatch/build'
make -f CMakeFiles/ttbincnv.dir/build.make CMakeFiles/ttbincnv.dir/build
make[2]: Entering directory '/home/conor/ttwatch/build'
make[2]: Nothing to be done for 'CMakeFiles/ttbincnv.dir/build'.
make[2]: Leaving directory '/home/conor/ttwatch/build'
/usr/bin/cmake -E cmake_progress_report /home/conor/ttwatch/build/CMakeFiles  11
[ 25%] Built target ttbincnv
make -f CMakeFiles/ttbinmod.dir/build.make CMakeFiles/ttbinmod.dir/depend
make[2]: Entering directory '/home/conor/ttwatch/build'
cd /home/conor/ttwatch/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/conor/ttwatch /home/conor/ttwatch /home/conor/ttwatch/build /home/conor/ttwatch/build /home/conor/ttwatch/build/CMakeFiles/ttbinmod.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/conor/ttwatch/build'
make -f CMakeFiles/ttbinmod.dir/build.make CMakeFiles/ttbinmod.dir/build
make[2]: Entering directory '/home/conor/ttwatch/build'
make[2]: Nothing to be done for 'CMakeFiles/ttbinmod.dir/build'.
make[2]: Leaving directory '/home/conor/ttwatch/build'
/usr/bin/cmake -E cmake_progress_report /home/conor/ttwatch/build/CMakeFiles  12
[ 28%] Built target ttbinmod
make -f CMakeFiles/manifest.dir/build.make CMakeFiles/manifest.dir/depend
make[2]: Entering directory '/home/conor/ttwatch/build'
cd /home/conor/ttwatch/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/conor/ttwatch /home/conor/ttwatch /home/conor/ttwatch/build /home/conor/ttwatch/build /home/conor/ttwatch/build/CMakeFiles/manifest.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/conor/ttwatch/build'
make -f CMakeFiles/manifest.dir/build.make CMakeFiles/manifest.dir/build
make[2]: Entering directory '/home/conor/ttwatch/build'
make[2]: Nothing to be done for 'CMakeFiles/manifest.dir/build'.
make[2]: Leaving directory '/home/conor/ttwatch/build'
/usr/bin/cmake -E cmake_progress_report /home/conor/ttwatch/build/CMakeFiles  8 9 10
[ 37%] Built target manifest
make -f CMakeFiles/ttwatch.dir/build.make CMakeFiles/ttwatch.dir/depend
make[2]: Entering directory '/home/conor/ttwatch/build'
cd /home/conor/ttwatch/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/conor/ttwatch /home/conor/ttwatch /home/conor/ttwatch/build /home/conor/ttwatch/build /home/conor/ttwatch/build/CMakeFiles/ttwatch.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/conor/ttwatch/build'
make -f CMakeFiles/ttwatch.dir/build.make CMakeFiles/ttwatch.dir/build
make[2]: Entering directory '/home/conor/ttwatch/build'
Linking CXX executable ttwatch
/usr/bin/cmake -E cmake_link_script CMakeFiles/ttwatch.dir/link.txt --verbose=1
/usr/bin/c++   -g    -Wl,--verbose CMakeFiles/ttwatch.dir/src/ttwatch.c.o CMakeFiles/ttwatch.dir/src/log.c.o CMakeFiles/ttwatch.dir/src/options.c.o CMakeFiles/ttwatch.dir/src/json.c.o CMakeFiles/ttwatch.dir/src/download.c.o CMakeFiles/ttwatch.dir/src/firmware.c.o CMakeFiles/ttwatch.dir/src/misc.c.o CMakeFiles/ttwatch.dir/src/get_activities.c.o CMakeFiles/ttwatch.dir/src/update_gps.c.o CMakeFiles/ttwatch.dir/src/set_time.c.o  -o ttwatch -rdynamic libttwatch.a libtt [... cut ...]
/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcurl.so: undefined reference to `UI_method_set_writer@OPENSSL_1.0.0'
collect2: error: ld returned 1 exit status
CMakeFiles/ttwatch.dir/build.make:316: recipe for target 'ttwatch' failed
make[2]: *** [ttwatch] Error 1
make[2]: Leaving directory '/home/conor/ttwatch/build'
CMakeFiles/Makefile2:233: recipe for target 'CMakeFiles/ttwatch.dir/all' failed
make[1]: *** [CMakeFiles/ttwatch.dir/all] Error 2
make[1]: Leaving directory '/home/conor/ttwatch/build'
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

```

ryanbinns, 2016-07-19:
I suspect that the issue is with the linuxbrew version of `libssl`. `cmake` is specifying the exact location of `libssl.so` and `libcrypto.so` from the linuxbrew area on the command line, but `libcurl` is expecting the ones located in `/usr/lib`. I'm not sure how to test this without removing the linuxbrew version of OpenSSL. Is that possible to do?

ramblingbarney, 2016-07-19:
Thanks, your right, I removed the linuxbrew OpenSSL but it still failed to find the system OpenSSL, the stackoverflow options `apt-get install openssl-dev  libssl-dev` didn't solve the issue.

Setting the CMAKE_PREFIX_PATH worked, I used the linuxbrew version for no specific reason.

`cmake -D CMAKE_PREFIX_PATH=/home/conor/.linuxbrew/opt/openssl/bin/openssl ..`

Thanks for your help again

ryanbinns, 2016-07-19:
That's a strange one. If there is no linuxbrew library installed, `cmake` *should* be able to find and use the system one. Still, I'm glad it's working now.

closed 2016-07-19 by ryanbinns