ryanbinns/ttwatch #94: ttwatchd refuses to process my watch

bioxz, 2016-07-17:
Starting ttwatchd with ` sudo ttwatchd --update-gps --set-time --runas=<user>` starts the daemon. Connecting my watch shows up as `Watch connected...` in the log, but then fails with `Watch not processed - does not match user selection`. Maybe i'm missing something in the docs? Can't make sense of that error message or the line of code that leads to that message.

The same command without the daemon `ttwatch --update-gps --set-time` works like a charm.

I tried to run it without the runas parameter, with a specific device name or id, but none worked. My device is a Runner 2 with firmware 1.3.27.

ryanbinns, 2016-07-18:
The most likely cause of that message is that the watch doesn't match a device specification, which is why the message says that, but there are a few others. If the watch is open by another program, it will wait for up to 60 seconds for it to become available then fail if it's still busy. If for some other reason the program is unable to open the watch (possibly a permissions error) then it will fail. If the device that was plugged in is not actually a watch it will fail (not relevant in this case since the `ttwatch` program works). If the watch fails to respond correctly to a series of startup messages it will fail (again, not relevant here since the `ttwatch` program works).

Do you have a config file in `/etc/ttwatch.conf` or `~/.ttwatch`?

bioxz, 2016-07-18:
I don't have a config file (looked in my ~, /etc and /root).

There is no delay between the connection message and the error message in the log, they're both within one second.  I've also tried to run ttwatch as root (to make sure i haven't made any mistake in the udev rules and removed access for root), that's also working.

So `ttwatch` as user or root does work, `ttwatchd` giving the error message with and without the `--runas=user` option.

ryanbinns, 2016-07-18:
Ok. That sounds like a bug. Can you modify the `hotplug_attach_callback` function in `ttwatchd.c` to the following:
```
int hotplug_attach_callback(struct libusb_context *ctx, struct libusb_device *dev,
    libusb_hotplug_event event, void *user_data)
{
    OPTIONS *options = (OPTIONS*)user_data;
    TTWATCH *watch = 0;
    int error = 0;

    write_log(0, "Watch connected...\n");

    error = ttwatch_open_device(dev, options->select_device ? options->device : 0, &watch);
    if (error == TTWATCH_NoError)
    {
        daemon_watch_operations(watch, options);
        write_log(0, "Finished watch operations\n");
        ttwatch_close(watch);
    }
    else
        write_log(0, "Watch not processed - does not match user selection (%d)\n", error);
    return 0;
}
```
This will print out the error code that `ttwatch_open_device returns`, which will hopefully give a bit more insight.

Alternatively, you could try stepping through `ttwatch_open_device` in `gdb` to see where it fails. You may have to rerun `cmake` to make a debug build first (`cmake CMAKE_BUILD_TYPE=Debug <src_dir>`).

bioxz, 2016-07-18:
`Watch not processed - does not match user selection (9)`

I get a 9 from that function, that is `TTWATCH_UnableToOpenDevice`. I changed the return values so i can see which section causes the issue, get a 30, so its this section:

```
 360     // open the device so we can read the serial number
 361     if (libusb_open(device, &handle))
 362     {
 363         *watch = 0;
 364         return 30;
 365     }
```

ryanbinns, 2016-07-19:
Ok. Can you modify that section to return the return value from `libusb_open`:
```
 360     // open the device so we can read the serial number
         int error;
 361     if (error = libusb_open(device, &handle))
 362     {
 363         *watch = 0;
 364         return error;
 365     }
```
If we get the libusb error code, we can find out exactly why it's failing to open the device.

bioxz, 2016-07-19:
The function returns `-3` for `LIBUSB_ERROR_ACCESS`.

Maybe this wasn't a bug at all and i'm just to stupid to use this program. Running the ttwatchd without the `--runas` option does work now. Running it with `--runas <user>:tomtom` (which is the group that owns the device) does work, too. Just running it with only the user as parameter and no group does fail now. I'll try to figure out if i have changed something....

ryanbinns, 2016-07-19:
Yes, you'll need to specify the `tomtom` group if it's not the default group for your user. The daemon, when it drops privileges, always has to set a user id *and* a group id. If you don't specify a group on the command line, it uses the default group for the user, so if that group doesn't have access to the USB device, then it will fail.

The `README.md` file does not mention this, so I'll update it.

bioxz, 2016-07-19:
Ok, thank you for the help. I'll try to reproduce the permission problem as root, but i fear that that was also my fault.

closed 2016-07-19 by bioxz