ryanbinns/ttwatch #95: Daemon seems not working

Evanedyr, 2016-07-18:
Hi,
i'm trying to put ttwatch to work, but i don't seem able to manage it...
First, i built the ttwatch-git package for arch linux's aur and gave it a try, and later i downloaded the whole git repo and built it as it was suggested in the README.md here. In both cases, i have to work as root, but it may not be the main problem. In both cases, i never seem able to get reactions from ttwatchd: 
launching _sudo ttwatch --initial-setup_ from the terminal, it hangs for a while and then finishes without printing anything.
If i run _ttwatch --update-fw_ i get 
> Current BLE firmware is already at latest version
> Current firmware is already at latest version
whilst _ttwatchd --update-fw_ returns nothing

If i run _sudo ttwatch -s /home/stefano/ttwatch_ i get the help screen (maybe i misstyped the command?), so i guess it didn't work.

_sudo ttwatch --get-activities returns nothing, same if i run ttwatchd instead.

However _sudo ttwatch --list-history_ returns the correct data indeed:

> Treadmill:
> 1: 2015/02/25 11:34:29,   53s,    10.06m,    0 calories
> Running:
> 1: 2023/02/02 04:14:39,  311s,   717.21m,   49 calories
> 2: 2023/02/02 05:51:51, 1281s,  6119.30m,  360 calories

(the strange data are because i didn't set the watch's clock when i first tried it)
So the watch is actually connected to the computer, and it looks like ttwatch is able to look inside, but somehow i can't get the activities from the watch, and ttwatchd doesn't seem to be working: the only times i get an answer from it is if i type a command wrong. Unfortunately i can't find any kind of journal or log to understand where the problem is....

Sorry for any error or stupid questions, unfortunately i still have to learn a lot about linux!

ryanbinns, 2016-07-19:
Ok. It actually all seems to be working properly, based on your descriptions. To allow you to run as a non-root user, you'll need to run `make install_udev` to install the `udev` rule that creates a `usb` user group then adds your user to it and enables access to the watch devices for that group. You'll then need to log out and back in for the group change to take effect.

Running `ttwatch --initial-setup` won't print anything unless there is an error, so no output means it worked fine.

`ttwatchd` is a daemon process - it's designed to detach from the terminal and run in the background, so if you run `sudo ttwatchd` then run `ps -ef | grep ttwatch` you'll see it in the process listing. When you run it from the command line, it won't print anything, and won't actually do anything until a watch is plugged in. It logs to `/var/log/ttwatch/ttwatch.log` so you can see if it's actually doing anything.

Running `ttwatch -s /home/stefano/ttwatch` doesn't actually do anything because the only option you've specified is to tell it where to store downloaded activity files; you haven't actually told it to download anything. To tell it to download activity files, run `ttwatch -s /home/stefano/ttwatch --get-activities`. If your home directory is `/home/stefano` then you don't need the `-s` option as the program uses `~/ttwatch` as the default location anyway.

The history data and the activity data are NOT the same. This is a common misunderstanding (I may have to add it to the `README.md` file). The history data is an abbreviated summary of the activity data, which is stored on the watch so it can be displayed and used for the race function. The activity data is logged every second and is quite large, so as soon as it is downloaded, it is deleted from the watch to make space available for new activities. 

To use it successfully in the same fashion as the TomTom software, run `ttwatch -a`. The `-a` switch tells it to use auto mode. It will download activity files, synchronise the watch time to the computer, update the GPSQuickFix data and update the watch firmware (if there's new firmware available). The activity files will be put in `~/ttwatch/<watch_name>/<date>/*.ttbin`. To set the watch name, run `ttwatch --set-name <watch_name>`.

To use the daemon, run `sudo ttwatchd -a` to do the same things every time the watch is plugged in. I would strongly recommend running as a different user, rather than root. This is done using the `--runas` option like so: `sudo ttwatchd -a --runas <user>:usb`.

That should give you some basic information about how to use it. There are other options, but they're not as frequently used.

Evanedyr, 2016-07-19:
Thank you for your response! 
It made quite a few things crystal clear!
I did actually look for the daemon in the task manager, but i did miss to specify °all processes°, and therefore i missed those...
Anyway, you're right, looking at the log everything looks fine! Does --get-activities write something on it? Because still, using the commands you suggested, i don't get any file regarding the activities in the target folder.... And the watch still has the data regarding old activities on it (which, as far as i understood, should have been wiped, right?).

ryanbinns, 2016-07-19:
No, the `--get-activities` option does not log anything except on failure.

The watch will still have history data (`ttwatch --list-history` will have valid data), but that doesn't mean that the activity data is still there. They are completely separate. If there are no activity files being written, then there is no activity data on the watch. Try creating a new activity (treadmill and shake your arm a few times is the easiest) and then docking the watch. If the daemon is running, you should get an activity file created in `~/ttwatch`. If the daemon is not running, then `ttwatch -a` should also create the activity file.

ryanbinns, 2016-07-19:
Oh, just in case it's not obvious, make sure you only have one daemon process running at a time. `kill` all the others. You can actually run multiple daemon processes for multiple watches and/or multiple users if you need to, but for just one watch there is no benefit, and it will just complicate things.

closed 2016-10-05 by ryanbinns