ryanbinns/ttwatch #100: No elevation data fetched

KlausKhe, 2016-09-21:
Hi Ryan,
first of all thanks for the great program.

I just got the TomTom Runner3. I downloaded an activity via 'ttwatch --get-activities' and converted the file to csv with 'ttbincnv -c File.ttbin'.

I did some investigation and I am sure that the statement 'download_elevation_data(ttbin);' is executed (I inserted a fprint before that line and the comment was printed to the screen).

Further, the headerline of the csv file is slightly inconsistent with the table. E.g. the last column shall be local time, but second last column is the local time and the last column is duration is MM:SS, the first column is duration in seconds. Heart rate is missing on purpose in the attached file.

Best regards,
Klaus

[Running_17-50-57.csv.txt](https://github.com/ryanbinns/ttwatch/files/486138/Running_17-50-57.csv.txt)
[Running_17-50-57.ttbin.txt](https://github.com/ryanbinns/ttwatch/files/486141/Running_17-50-57.ttbin.txt)

ryanbinns, 2016-10-05:
I hadn't noticed that their elevation model has been taken down, but yes, it's not available at that location any more. When I get a chance, I'll check to see if they've moved it, or if they've discontinued it completely.

KlausKhe, 2016-10-05:
Ryans, I will reach out to the owner of http://www.gpsvisualizer.com/elevation and ask whether he is willing to support batch operations so that this web page might be used instead of the TomTom's one

siegbewi, 2017-01-23:
Hello Ryan,

thank you for your great work.

Can this issue be reopened, since it is not solved?

Greetings,
Siegfried.

ryanbinns, 2017-01-30:
Sure. I didn't realise it had been closed; it wasn't closed by me.

runningt, 2017-05-30:
While waiting on this issue to be fixed I created simple [project](https://github.com/runningt/TrainingEnhancer) to enhance TCXs exported from TomTom Runner with elevation data from [Mapzen Elevation Service](https://mapzen.com/documentation/elevation/elevation-service/).  It is very draft version yet but maybe it could be useful for someone.

@ryanbinns: 
I noticed that the issue is in `ttbin.c` file in `download_elevation_data()` method.
The problem is that tomtom fixmodel API you are using `https://mysports.tomtom.com/tyne/dem/fixmodel` seems not to be working any more.
Perhaps you could also use external elevation API like mapzen instead.

TheAkki, 2019-09-07:
Hello @ryanbinns ,
I have forked this nice Project and have implemented to use an open-elevation server to get the elevation data.
See TTWatch: https://github.com/TheAkki/ttwatch
See Open-Elevation: https://github.com/TheAkki/open-elevation (Use 30m SRTM data)