ryanbinns/ttwatch #101: Does it works for the runner 3

icefinger, 2016-10-03:
Dear developpers,
pretty everything in the title. Iḿ planning to but a tomtom runner 3, but I have only linux and I would like to know first if ttwatch works correctly for linux.

KlausKhe, 2016-10-03:
Hi icefinger,
I am not the developer but use runner 3 with linux using ttwatch. Actions used so far are downloading activities and setting GPSFix. Works. I initialized the watch via Win10 in a vmware. The only 'issue' I detected so far is that the incorporation of altitude data doesn't work anymore,  see my post before yours. But this is rather due to an out of service side of TomTom and has nothing to do with runner 3.
Cheers,
Klaus

icefinger, 2016-10-04:
Thanks a lot, it answers my question.

closed 2016-10-04 by icefinger