ryanbinns/ttwatch #102: Failing to read ttbin file - run after a defined route

mipapo, 2016-10-18:
Hi,
I don't own a TomTom 3, but some of our users using our software (Runalyze/Runalyze) do. 
A user complained that a ttbin file couldn't be read. 

- It's a running activity
- There is a new feature in TomTom3, where you can run after a defined route. You upload a gpx file to the watch and can just see that track (I think during the run). The user who wrote us did this.

Would be great if anyone could give us a hint. I'm not that deep in that ttbin /TomTom topic

[00910000_20161018_203136.ttbin.zip](https://github.com/ryanbinns/ttwatch/files/537185/00910000_20161018_203136.ttbin.zip)

ryanbinns, 2016-10-20:
Hi Michael,
That file seems to be corrupt. It's pretty much perfect except there are two extra bytes on the end of the file. If you remove them, the file should load perfectly. I don't know whether this will always happen, but I guess if you ignore a truncated record at the very end of the file, then it should be ok.

The file is describing an 8.1km run with laps every 1000m. There are a few records at the beginning of the file that I don't know the structure of, but they are described adequately in the file header so they can be skipped without issue. I presume these records hold information about the new feature you mentioned.

The `ttbincnv` program reads and converts this file without problem (after I fixed a minor bug) after removing the last two bytes.

mipapo, 2016-10-20:
Hi Ryan,
I really appreciate your support and this tools!
With your last commit everything works fine (and without removing those two bytes at the end!)

So from my side I just can say - It works!

closed 2016-10-20 by mipapo