ryanbinns/ttwatch #104: Upload of Trails/Routes available with Runner/Spark 3

machinaeXlasse, 2016-10-25:
With Runner 3 (I dont know about the golf and the hiking watch) it's possible to upload GPX Routes to the watch, giving you a reduced map, showing the way.

As with the windows software you upload a gpx file on the web interface. When the tool is running it uploads it to the watch and makes it available in the "trails" option that you can select for most activities.

I have no clue how this works internally but if there is any way I can find sth. out to help, please tell me!

Please feel free to close this issue whenever its too complicated to solve.

(FTR: I love your piece of software and that you put your time into this! Made the watch twice as useful for me! Thanks a lot!)

_Also: I can't find a donate button anywhere!_

ryanbinns, 2017-03-07:
I've only just seen this issue. I must have missed the email or something. Sorry about that.

It doesn't sound too difficult to add, but that's just from assuming how it works. Most likely it just writes a file to the watch with a file ID that the watch recognises.

If you're willing to help, that's great. If I'm correct, it shouldn't be too difficult.
1. Recompile the ttwatch software with the unsafe option enabled (`cmake -Dunsafe=on`).
2. Run `ttwatch --list` to get a list of all the files on the watch. Save the list.
3. Add a GPS route to the watch using the standard TomTom software.
3. Run `ttwatch --list` again to get a new list of files. Compare the two.
4. For any new files in the list, run `ttwatch --read <id> > <id>.bin` to download the file from the watch.
5. Ditto for any files with a different length (`ttwatch --list` reports file IDs and lengths).
6. Zip up the files (.tar.gz or .tar.bz2) and attach them so I can have a look.

If the new files are simply GPX files, you don't need to send them to me, but let me know their ID (0x....) as that will be important.

machinaeXlasse, 2017-08-03:
Now I too recognised your comment just now but perhabs I'm lucky and you are still up for the task.

I uploaded a track and attached all files that changed afterwards. The ones with a new ID are in the _new_ folder. The ones that only changed in length are in _difflen_.

Though if I'm not mistaken you are already ahead over here: #130 

Perhabs the files are still of some use.

[changed_files.tar.gz](https://github.com/ryanbinns/ttwatch/files/1197103/changed_files.tar.gz)

ryanbinns, 2017-08-03:
Thanks. I'll have a look when I get some time. Don't hold your breath though, it may be a while...

ABelliqueux, 2018-07-24:
@machinaeXlasse : Sorry to kind of necropost, but this functionality is available in @scubajorgen 's software available here : http://blog.studioblueplanet.net/?page_id=566

I would rather have ttwatch include this functionality, but it's nice to have a few different softwares, especially since Tomtom annouced they're killing their wearables branche. At least uour watches won't end as sophisticate bricks !