ryanbinns/ttwatch #106: unable to open watch ttwatch-git 20161128-1

ladywithhat, 2016-12-02:
With the new build of the ttwatch-git 20161128-1 i am unable to get a connection with my spark runner 2.
FW-version: 1.3.250
Linux: archlinux

`ttwatch -v
Unable to open watch
`
Former version 20160527-1 worked perfectly.

adzane, 2016-12-02:
Have you checked the connection? Is the watch *firmly* snapped into the cable cradle?

ladywithhat, 2016-12-04:
@adzane lsusb shows watch and watch is charging. so I assume the watch is detected
Bus 002 Device 008: ID 1390:7475 TOMTOM B.V. 
Linux: 4.8.11-1-ARCH

Other ways to find out why ttwatch is not working?

ryanbinns, 2016-12-05:
That's a strange situation. The latest update only revision information for use with the `--setting(s)` commands.

If you run `ttwatch --devices`, does it list your watch?

ladywithhat, 2016-12-05:
I had to kill ttwatchd than ttwatch worked as expected. But now ttwatchd does not work anymore.

closed 2016-12-25 by ladywithhat