ryanbinns/ttwatch #108: TCX conversion fails on a spinning file

mipapo, 2016-12-21:
The file was produced with a tomtom runner 2, while spinning/cycling.

This issue was submited by one of ours RUNALYZE users. Sadly I cannot help. It would be great if you /anyone would have a short look at that file.
Is this file defect or something missing in ttwatch?

ryanbinns, 2016-12-22:
The file has an activity type code of 11. The ttwatch programs only support activity types of 0 (Running), 1 (Cycling), 2 (Swimming), 7 (Treadmill), 8 (Freestyle) and 9 (Gym).

This must be a new activity type that I don't know anything about. I don't have a Runner 2, so I can't really help much with adding support for this.

The software can currently only interpret the heart-rate information in the file; although there is lots of other data in the file, I don't know what it means and the software can't interpret it. If someone wants to add support for this, that would be great, but I don't think I'll be able to at the moment.

ladywithhat, 2016-12-25:
Than 11 must be the "indoor" activity with the runner 2.

mipapo, 2016-12-28:
I'm not sure about the following:
Another record:(I just named it INDOOR_CYCLING here
 `#define TAG_INDOOR_CYCLING              (0x33)`

I played around with  https://github.com/dlenski/ttbindec and fixed some things so that I was able to read this file (version type 9)

```
# don't know what's in there and how many fields ...
class FILE_UNKNOWN1_RECORD(StructReprMixin, LittleEndianStructure):
    _pack_ = 1
    _fields_ = (
        ('timestamp',c_uint32), # local time
        ('unkown1',c_uint32), #
        ('frequency',c_uint16),
...
    )
```
I got the following:
`00002366 33 FILE_UNKNOWN1_RECORD(timestamp=1482091366, unkown1=1459189568, frequency=88, ...)
`
I'm really sure about the frequency (looks good), but I don't know what is in the other fields. Don't know if it could help to find a solution

I'm really not good at reverse-engineering proprietary files

[Unknown_20-55-46.ttbin.dump.zip](https://github.com/ryanbinns/ttwatch/files/675738/Unknown_20-55-46.ttbin.dump.zip)

misdoro, 2017-01-17:
@mipapo indoor cycling activities may be exported into CSV files now.
Do you have any example TCX file containing indoor activity?

mipapo, 2017-01-18:
@misdoro See attachment - recorded with an Garmin 920XT
[activity_1491283845 (2).tcx.zip](https://github.com/ryanbinns/ttwatch/files/713732/activity_1491283845.2.tcx.zip)

misdoro, 2017-01-18:
Thanks, I'll have a look!

PR #113 (2017-01-22): Export indoor cycling activity into TCX files
Fixes #108 
Exported file uploaded OK to Strava.
Heart rate, cadence info & distance are fully supported

Example output file:
[Indoor_22-29-42.tcx.zip](https://github.com/ryanbinns/ttwatch/files/722135/Indoor_22-29-42.tcx.zip)

Fixes #68 
Example input & output files:
[Gym_11-41-14.ttbin.zip](https://github.com/ryanbinns/ttwatch/files/722172/Gym_11-41-14.ttbin.zip)
[gym_export.zip](https://github.com/ryanbinns/ttwatch/files/722171/gym_export.zip)
closed 2017-01-22 by ryanbinns